smartavionics/Cura #65: Create complete top layer

kvnper, 2020-07-07:
I'm writing this from the github app on my phone and has far less functionality than the website does so forgive me if I've put something wrong. 

This is a feature request for this problem
 https://www.reddit.com/r/3dprinting/comments/hmu4pr

This is a reddit thread I stumbled across tonight and it isn't the first time.

Basically, if the original poster is hard to understand, when, for example, printing signs with embossed lettering, the top layer before the lettering begins will print in sections partitioned by the lettering. This causes the top layer before the lettering to be inconsistent and ugly. 
I'm not sure if it's simple but the proposed solution would be allowing that top layer before the lettering to be printed in a single sweep.

smartavionics, 2020-07-07:
Have you tried increasing the skin expand distance to cover the area below the lettering?

kvnper, 2020-07-07:
I'll give that a go and report back.