smartavionics/Cura #66: Cura Master crashes on file->open Windows 10 on my machine

andymozza, 2020-07-08:
**Application Version**
Cura Master build from your dropbox. Tried with a few builds and latest I tried was
Cura-mb-master-win64-20200610

**Platform**
WIndows 10 1909 and now 2004

**Printer**
Seems not related to chosen printer profile

**Steps to Reproduce**
Go to File-> Open Files

**Actual Results**
Cura just closes completely, no error or warning it just disappears from screen.

**Expected results**
Cura 4.6.1 build from Ultimaker actually shows the file browser and chooses and lets me pick the stl file.

**Additional Information**
I have tested the current build from Ultimaker and this does not show the same issue.
I have cleaned out the Master install location and re-installed fresh to ensure there are no left overs.

Hi, if this is something its possible to investigate, then what files / logs should I attach - please let me know.
Are there differences between runtimes / libraries used in your build that might affect this ?
Appreciate any thoughts on where to go next with diagnosing.

smartavionics, 2020-07-08:
Hmm, don't know what is happening there  but there could be something useful in the log files which will be under AppData, can you find the cura.log and stderr.log  files and zip up and attach to this issue, thanks.

smartavionics, 2020-07-08:
I don't know anything about Windows as I only use it for building Cura releases! How do you find out the OS version?

andymozza, 2020-07-08:
settings app -> updates and security -> right side of the update screen has a link 'os build info' 
[image]

andymozza, 2020-07-08:
[cura.log](https://github.com/smartavionics/Cura/files/4892353/cura.log)

[stderr.log](https://github.com/smartavionics/Cura/files/4892373/stderr.log)

I found these in Appdata->Roaming->Cura->master, but there is nothing recorded about the exit when I open the file browser. Last thing is simply the error because it can't upload the backup.

andymozza, 2020-07-08:
Also notice that I have this folder somehow and it has crash dumps in it . There is one for cura at the time it crashed.
[image]

andymozza, 2020-07-08:
[Cura.exe.38508.dmp.zip](https://github.com/smartavionics/Cura/files/4892394/Cura.exe.38508.dmp.zip)

Not sure if this is of any use. I don't know how to read / view it.

smartavionics, 2020-07-08:
I just tried my latest build and it doesn't crash. The Windows version is Windows 10 home 1903.

andymozza, 2020-07-08:
The stored exception information can be accessed via .ecxr.
(966c.6364): Security check failure or stack buffer overrun - code c0000409 (first/second chance not available)
Subcode: 0x7 FAST_FAIL_FATAL_APP_EXIT 
For analysis of this file, run !analyze -v
Qt5Core!QCoreApplicationPrivate::qmljsDebugArgumentsString+0xb8:
00007ff9`e7392b08 cd29            int     29h

andymozza, 2020-07-08:
Here is the full output of the rather impressive new dmp tool from Microsoft... it actually goes and grabs all the dlls,symbols etc automatically.... Not sure if this means anything to you:

EXCEPTION_RECORD:  (.exr -1)
ExceptionAddress: 00007ff9e7392b08 (Qt5Core!QCoreApplicationPrivate::qmljsDebugArgumentsString+0x00000000000000b8)
   ExceptionCode: c0000409 (Security check failure or stack buffer overrun)
  ExceptionFlags: 00000001
NumberParameters: 1
   Parameter[0]: 0000000000000007
Subcode: 0x7 FAST_FAIL_FATAL_APP_EXIT 

PROCESS_NAME:  Cura.exe

ERROR_CODE: (NTSTATUS) 0xc0000409 - The system detected an overrun of a stack-based buffer in this application. This overrun could potentially allow a malicious user to gain control of this application.

EXCEPTION_CODE_STR:  c0000409

EXCEPTION_PARAMETER1:  0000000000000007

STACK_TEXT:  
000000ce`170fca70 00007ff9`e7390ea3     : 000000ce`170fcad0 000000ce`170fcb20 000001ee`a11703b0 000000ce`170fd070 : Qt5Core!QCoreApplicationPrivate::qmljsDebugArgumentsString+0xb8
000000ce`170fcaa0 00007ff9`e755a1cc     : 000001ee`94f939b8 00007ff9`e7743a40 00000000`00050d02 00000000`00050d00 : Qt5Core!QMessageLogger::fatal+0x93
000000ce`170fcb00 00007ffa`19d2132a     : 00000000`00000000 00007ffa`19e00000 000001ee`a1740370 00007ffa`19e09fdb : Qt5Core!QObjectPrivate::QObjectPrivate+0x7c
000000ce`170fcb50 00007ffa`19d212e2     : 000000ce`170fd070 00007ff9`e7658f00 000001ee`00000007 00000000`00000000 : qsqlite+0x132a
000000ce`170fcba0 00007ffa`1f2e536c     : 00007ffa`1f3105b0 000000ce`170f0000 000001ee`99598149 00000000`00000000 : qsqlite+0x12e2
000000ce`170fcbd0 00007ffa`1f2e3c1e     : 000000ce`170fcce0 00000000`00000005 000000ce`170fd070 ffffffff`ffffffff : Qt5Sql!QSqlDatabase::hostName+0x1bc
000000ce`170fccb0 00007ffa`2357fd71     : 00000000`00000005 000000ce`170fd070 000001ee`afea5660 00007ff9`e7659188 : Qt5Sql!QSqlDatabase::addDatabase+0x3e
000000ce`170fcce0 00007ffa`2357f64f     : 000000ce`170f0020 00000000`00000000 000001ee`a13f7a30 000001ee`a13016a0 : CHITUBOX_Thumbnail!DllUnregisterServer+0x9a51
000000ce`170fce10 00007ffa`23576e71     : 000000ce`170fd070 000001ee`a1301ac0 000001ee`afea6120 000001ee`9a003b70 : CHITUBOX_Thumbnail!DllUnregisterServer+0x932f
000000ce`170fceb0 00007ffa`30e384e3     : 00000000`80004002 000001ee`94f939a8 00000000`00000000 00000000`00000000 : CHITUBOX_Thumbnail!DllUnregisterServer+0xb51
000000ce`170fd500 00007ffa`30ec969d     : 00000000`00000000 00000000`80004002 000000ce`170fd6f8 00000000`00000004 : windows_storage!InitializeFileHandlerWithFile+0x83
000000ce`170fd570 00007ffa`30ea34fc     : 00000000`00000000 00000000`00000000 00000000`00000002 00000000`00000000 : windows_storage!CFileSysItemString::HandlerCreateInstance+0x4d9
000000ce`170fd660 00007ffa`30ed7e1c     : 00000000`00000000 00000000`00000000 00007ffa`040bafa0 00007ffa`040bafa0 : windows_storage!CFileSysItemString::LoadHandler+0x1f4
000000ce`170fd7b0 00007ffa`30ea2cd6     : 00007ffa`3149e8b8 000001ee`a18067a0 000000ce`170fdf10 00000000`00000000 : windows_storage!CFSFolder::s_GetThumbnailExtractor+0x18c
000000ce`170fdb40 00007ffa`30f2ceb2     : 000000ce`170fe3c8 00000000`00000000 000000ce`170fdf88 000000ce`170fdf80 : windows_storage!CFSFolder::_BindHandler+0x5c6
000000ce`170fded0 00007ffa`30ed86c9     : 00000000`00000000 000001ee`99cd6cd0 000001ee`99cd6cf0 00000000`00000000 : windows_storage!CFSFolder::GetThumbnailHandler+0x62
000000ce`170fdf50 00007ffa`30eb6767     : 00000000`00000000 000000ce`170fe110 000001ee`99cd6cd0 00000000`80004001 : windows_storage!_CreateThumbnailHandler+0xa9
000000ce`170fe010 00007ffa`04084ca1     : 00000000`00000000 000000ce`170fe3f9 00000000`00000000 00000000`00000000 : windows_storage!CShellItem::BindToHandler+0x727
000000ce`170fe370 00007ffa`04084905     : 00000000`00002220 00000000`00002220 000001ee`acc77f20 00000000`00000000 : thumbcache!InitExtractor+0x255
000000ce`170fe440 00007ffa`04085b36     : 000000ce`00000000 000001ee`00006364 00007ffa`040cce90 00000000`00000000 : thumbcache!CThumbnailCache::_PerformFullExtractionCore+0x171
000000ce`170fe550 00007ffa`0407a607     : 000001ee`82a80000 00000000`00000000 000001ee`829c0000 00000000`00000000 : thumbcache!CThumbnailCache::_PerformFullExtraction+0xce
000000ce`170fe760 00007ffa`04079028     : 00000000`8004b200 00000000`00000000 00000000`00000000 00000000`00000000 : thumbcache!CThumbnailCache::_GetThumbnailImpl+0x4c7
000000ce`170fec50 00007ffa`04078187     : 00000000`00000000 00000000`00000000 0000338a`b27da887 00007ffa`3360a515 : thumbcache!CThumbnailCache::_GetThumbnailInternal+0x1b8
000000ce`170fed40 00007ffa`040843f1     : 00000000`00000000 000000ce`170fee80 00000000`00000001 000000ce`170fee10 : thumbcache!CThumbnailCache::GetThumbnailPrivate+0x67
000000ce`170fedc0 00007ffa`0408432f     : 000001ee`0000009c 00007ffa`314a0b58 000000ce`170fee88 00007ffa`30eeb703 : thumbcache!CThumbnailCacheAPI::GetThumbnailPrivate+0xb1
000000ce`170fee60 00007ffa`30eea2da     : 00000000`00929d9d 00000000`00000001 00000060`00000060 00000000`00000009 : thumbcache!CThumbnailCacheAPI::GetThumbnail+0x2f
000000ce`170feeb0 00007ffa`30eea209     : 000000ce`00000000 00000060`00000060 00000000`00000009 00000000`00000000 : windows_storage!CShellItem::_GetThumbnail+0xae
000000ce`170fef70 00007ffa`343e162b     : 000000ce`170ff0f0 00007ffa`3438e114 00000000`00000000 00000000`00000000 : windows_storage!CShellItem::GetSharedBitmap+0x159
[... cut ...]
000000ce`170ff800 00000000`00000000     : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : ntdll!RtlUserThreadStart+0x21

SYMBOL_NAME:  Qt5Core!QCoreApplicationPrivate::qmljsDebugArgumentsString+b8

MODULE_NAME: Qt5Core

IMAGE_NAME:  Qt5Core.dll

STACK_COMMAND:  ~9s ; .ecxr ; kb

FAILURE_BUCKET_ID:  FAIL_FAST_FATAL_APP_EXIT_c0000409_Qt5Core.dll!QCoreApplicationPrivate::qmljsDebugArgumentsString

OS_VERSION:  10.0.19041.1

BUILDLAB_STR:  vb_release

OSPLATFORM_TYPE:  x64

OSNAME:  Windows 10

IMAGE_VERSION:  5.13.0.0

FAILURE_ID_HASH:  {ce113b38-2589-35b3-af2b-4f6f27571558}

Followup:     MachineOwner
---------

smartavionics, 2020-07-08:
Please search your machine for QtCore.dll and see if you have other instances of that which may conflict with what is supplied in my release.

andymozza, 2020-07-08:
I have 'loads' of other instances of the file on the machine, but I don't think any should override yours as they are not in the path.
I do notice yours is a newer version that the one in 4.6.1

[image]

andymozza, 2020-07-08:
Ok!! 
Your hint for the qt dll and the mention of chitubox in the stack trace sent me off googling.

I found references to dell backup manager causing later qt to crash in other applications as it installed a file filter which did something inside the file manager.

Chitibox( sla/resin slicer) now includes a file preview thumbnail generator when you view files in windows file explorer......
Chitubox also had a qt5core dll in the list.
Chitubox was also in the stack trace for opening the file filter ( obviously it’s loaded when you open a file dialogue to browse and choose inside cura).

So - I unregistered the dll that chitubox supply for creating the preview of the thumbnail and thus resolved it.

Older cura QT version didn’t appear to have a problem with it.

As always thanks for pointing me in the right direction!

smartavionics, 2020-07-08:
> As always thanks for pointing me in the right direction!

It was a lucky guess, glad you found a solution, closing...

closed 2020-07-08 by smartavionics

dryphi, 2021-04-27:
> So - I unregistered the dll that chitubox supply for creating the preview of the thumbnail and thus resolved it.

How did you do this specifically?

andymozza, 2021-04-27:
There seems to be a post in the chitibox forum now.

https://forum.chitubox.com/?thread-419.htm

I did the regsvr32 /u dll-path option and restarted.

dryphi, 2021-04-27:
I just uninstalled ChituBox. Cura works fine now.