smartavionics/Cura #67: Retraction missing

eskeyaar, 2020-07-10:
Hi Mark,
As you see from the picture below, there is no retraction at the top of 4 pillars, is it due to towers are thin? 
Thanks for your time.

**Application Version**

 Cura-mb-master-win64-20200703  

**Platform**
Windows64Bit

**Printer**
Any printer with any profile

**Steps to Reproduce**
<for instance, choose UM2+, normal profile, slice it. and in my project file too.
[UM2_four_square.zip](https://github.com/smartavionics/Cura/files/4903414/UM2_four_square.zip)

>

**Actual Results**
< As soon as model converges into thin, there is no retraction at all, only at one of pillar its happening that too i think its because of retracting at layer change?. Because of no retraction, lot of stringing occurs on those top sections>
[image]

[image]

**Expected results**
There should be retraction all the way of model
Should be a clean model. 

**Additional Information**
Same model in Prusaslicer, does have retraction all the way to top. And seems like this is happening in older versions of cura too. Tested on Cura4.0 and 4.6, same result. or am i missing something in settings? 
Thanks
[image]

smartavionics, 2020-07-10:
Have you tried fiddling with the maximum retraction count or minimum extrusion distance window settings? They could have an influence on this behaviour.

eskeyaar, 2020-07-10:
Yes, tried, no effect on them.

eskeyaar, 2020-07-10:
> 
> 
> Have you tried fiddling with the maximum retraction count or minimum extrusion distance window settings? They could have an influence on this behaviour.

Update: i was wrong!, I entered value of max retraction count of 300 and now i see retraction. Earlier i checked with 120, didn't notice any difference on that value. But what is the safe value for these settings? more than 300?
Thanks for your input.

smartavionics, 2020-07-10:
I've never changed them so I can't say what is a good limit but if you have a decent extruder that doesn't mash the filament up too badly it should be OK to use quite a high value.

eskeyaar, 2020-07-10:
Thanks for the tip!
We can close this.

closed 2020-07-10 by eskeyaar