smartavionics/Cura #68: Too many segments?

eskeyaar, 2020-07-20:
Hi Mark!
An interesting thing in your build is that, when i checked my prints, it does produce too many segments but when compared to Cura4.6.0. it doesn't produce those many,
Used the same profile on both occasions.
Yours:
[image]
Cura 4.6.0
[image]

Left is cura 4.6 right is your version
[image]

**Application Version**
Cura-mb-master-win64-20200703

**Platform**
Win 10 64bit

**Printer**
UM2+

**Steps to Reproduce**
Attached link to download project file. ( size was too big to attach here)
https://wetransfer.com/downloads/5269d6cbfcd23f8b3f587398c24a70b220200720155726/e52a1c9bfc64ac4ba2103f93628d7f8220200720155747/22d63d

**Additional Information**
Attached Gcode of 4.6.0 too. 
[UM2_RockAB46.zip](https://github.com/smartavionics/Cura/files/4948762/UM2_RockAB46.zip)
Your Version Gcode:
[UM2_RockABmasterdef1.zip](https://github.com/smartavionics/Cura/files/4948776/UM2_RockABmasterdef1.zip)

smartavionics, 2020-07-20:
The maximum deviation setting has more effect than the max resolution setting in controlling the discarding of vertices. You are using a very small value for max deviation so it will not throw away many vertices even if you make the max resolution quite large. When you increase the value of max resolution, the line count drops dramatically.

Anyway, if I look at the number of gcode lines in your two files, I see:

  96671 UM2_RockAB46.gcode
  90972 UM2_RockABmasterdef1.gcode

Which suggests that there's more gcode in the 4.6.0 output than in the 20200703 output.

If I slice that project with my current release, I get:

  89709 UM2_RockAB.gcode

Hmm, I just noticed a problem with that slice which has been caused by a recent upstream change which will have affected my latest builds. Looks like I need to fix that and make new builds tomorrow.

smartavionics, 2020-07-21:
Increasing the max deviation to 0.1 reduces the gcode line count to 74871 and the preview doesn't look too bad.

eskeyaar, 2020-07-21:
> The maximum deviation setting has more effect than the max resolution setting in controlling the discarding of vertices. You are using a very small value for max deviation so it will not throw away [...]

The values I kept because it was default values for UM2+. and I was just testing my prints with different values, then i found your version was producing more compared to Cura.
What does it mean by the number of lines produced by 20200703 output is less? even though it produces more segments? 

Yes, increasing max deviation to 0.1, segments look good. I will try to print it out and see.
Thanks

eskeyaar, 2020-07-21:
Mark is the 20200403 based on cura 4.6? or which version?
Thanks

smartavionics, 2020-07-21:
It's based on whatever the UM master branch was at the time of the build. However, because the UM master branch often contains fresh bugs and regressions, I delay making releases so as to try and detect any problems. However, I would say there's a reasonably close correspondence between my latest releases and the 4.6 releases from UM.

smartavionics, 2020-07-21:
> What does it mean by the number of lines produced by 20200703 output is less? even though it produces more segments?

I'm not sure that master is producing more segments. When I count the number of G1 gcodes in the files you provided, I get:

grep G1 UM2_RockAB46.gcode | wc -l
75575

grep G1 UM2_RockABmasterdef1.gcode | wc -l
73889

So master is producing less G1 gcodes than 46 and G1 gcodes involve extruder movement (line extrusion or retract/un-retract).

eskeyaar, 2020-07-21:
> It's based on whatever the UM master branch was at the time of the build. However, because the UM master branch often contains fresh bugs and regressions, I delay making releases so as to try and [...]

Understood.

> So master is producing less G1 gcodes than 46
Yes, it does produce no of lines less, but the preview of that gcode does show more small segments compared to other. maybe that preview is showing wrong? I'm not sure.
Have you released your new build? 
Thanks!

smartavionics, 2020-07-21:
Yes, there is a release today.

eskeyaar, 2020-07-21:
Thanks. Will check that out.