smartavionics/Cura #69: ExtruderBox not working inside Cura.PrintMonitor element

adripo, 2020-08-13:
I am trying to use the MKS WiFi Plugin with this build
https://github.com/Jeredian/mks-wifi-plugin

At the beginning it didn't work because it loads some custom interaces for version 4.x so I created a pr to make the plugin work also with this plugin.
https://github.com/Jeredian/mks-wifi-plugin/pull/31

Now Everything works except the ExtruderBox (Ultimaker Cura master\resources\qml\PrinterOutput\ExtruderBox.qml) inside Cura.PrintMonitor (Ultimaker Cura master\resources\qml\PrintMonitor.qml) element
Here you can see the row where this element is created
https://github.com/Jeredian/mks-wifi-plugin/blob/master/MKSWifiPlugin/MonitorItem4x.qml#L121
But the only thing that I see in the interface is the Bed Plate. No Extruder/nozzle is visible.

[image]

I tried to debug some changes but with no luck.

**Application Version**

master

**Platform**

Windows

**Printer**

FB Ghost 4S

**Steps to Reproduce**

Download plugin from here https://github.com/adripo/mks-wifi-plugin/tree/patch-2 and put it inside plugin folder. Launch Cura master and go to Monitor tab after printer is connected.
I think the issue appears also by creating a Cura.PrintMonitor element and connect your printer with USB.

**Actual Results**

No Extrusion Element view

**Expected results**

Extrusion tab view

**Additional Information**

--

smartavionics, 2020-08-13:
Does this work as expected with the UM 4.7 beta release?

adripo, 2020-08-13:
I haven't tried it yet. Is there an exe ready to install or do I have to build cura manually?

smartavionics, 2020-08-14:
https://github.com/Ultimaker/Cura/releases/tag/4.7-beta