smartavionics/Cura #70: Print concentric and continuous

amagr0, 2020-08-17:
Hello @smartavionics,

This is not an issue related with you wonderful Cura version, is only me asking you how to resolve my problem.

I'm working on a project to develop an experimental filament with a high metal percentage, the parts printed then should be sinteret in order to obtain a full (98%) metal part.
This filament is at this moment very fragile and should be printed without retractions, and without to much change directions.  
The toolpath strategy is to print in a concentric way, I already tried diferent aprouches, but non of them is giving me what I want.

The desired toolpath should concentric (or continuous) printing, and should start from inside to outside, or in the opposite way, like in the image.
[image]

**Toolpath stratagys:**

**Only perimeters:**
- Wall Line Count: 100
- Top/Bottom Thickness: 0 mm
- Infill: 0 %

This option almost resolve my problem, but Cura starts from the middle, prints the inside perimeters, then goes to the middle to print the outside perimeters. Since I cannot make retractions, this option will result on some underextrusion in the beggining of the outside perimeters.

[image]

I also tried this options, but again, I not getting what I intend.
- Wall Line Count: 0
- Top/Bottom Thickness: 100 mm - concentric
- Infill: 0 %

---

- Wall Line Count: 0
- Top/Bottom Thickness: 0 mm
- Infill: 100 % - concentric

What you could suggest in order to solve my problem?
Thanks

Sophist-UK, 2020-08-17:
You can try turning retraction on but set the retraction speed and amount to very low values, and then use retraction prime to prevent the under extrusion at the beginning of an extrusion.

amagr0, 2020-08-17:
The problem with retraction is related with the friction on the same part of the filament, that could break the filament, definitely I could not use it.

Thanks for the tip.

Sophist-UK, 2020-08-17:
I am sorry if I did not explain myself clearly. Make the retraction so small that it is less than one step of the motor - so although retraction is on, it **won't actually do any retraction**. But after it has done the non-existent retraction and the non-existent prime, you can then do an extra prime amount which will extrude a little extra filament and prevent the under extrusion after the travel move.

smartavionics, 2020-08-18:
Hi, while it could be possible to make it print all those walls in the same direction from inside to outside (or the other way round) that would only be useful for a very limited class of models. How do you envisage printing something that had, say, two holes rather than just one hole as per your example?

Sophist-UK, 2020-08-18:
@smartavionics - For two holes you can achieve the same by printing a mobius strip. (Just kidding of course. ;-) )

 But actually a spiral layer only works for a completely circular layer with at most one completely circular hole in the precise centre of the layer. So the class of models for spiral layers is extremely small.

Thinking a bit more laterally, the class of shapes that can be printed without travel is quite a lot larger if you don't have outer walls, inner walls and infill but instead treat all layers as surface layer with line or zigzag pattern and zero walls. I am not certain whether you can tweak Cura to do this (I have never tried 0 walls and infinite surface layers, and even if you did print in this way, I am not sure what the structural strength or surface quality would be when printer or after sintering. And I am not sure what classes of layer shapes this allows without travel. But I know from experience that Cura sometimes cannot print surface layers without travel.

So in the end, I think that being able to print with a filament that cannot retract will come down to the settings I have suggested - a minuscule retraction length that is effectively 0 (or if Cura allows it an actual 0) and an extra prime to avoid under-extrusion issues, whilst also avoiding extra prime causing over extrusion blobs / zits. It may be that under extrusion can not ever be avoided completely without creating over-extrusion, in which case the next issue would be how to ensure that the under-extrusion happens in a non-visible area.

If these cannot be solved, then the class of models that can be printed by this method will be very small which will limit the commercial viability of this type of filament.