smartavionics/Cura #71: [fr] recovery end of bridges, lack of material

jbalibeux, 2020-10-20:
hello, I'm looking to perfect my bridges. especially the end of bridges.
I activate the bridge parameters. with an extrusion at 70 (approximately according to my tests) and with a little freewheel at the start to relieve the pressure, I have a very flat underside. OK
but at the end, when normal extrusion is resumed, the pressure is too low and generates a lack of material. (see photo below).

I think a parameter resembling "additional volume at the boot" but at the end of the bridge could solve this problem ... or a small pause before the resumption or other something else?

(sorry for this google translation, do not hesitate to ask me to rephrase if it is badly translated :))

[image]
[image]
I have a hard time displaying the image correctly here, I do not know github very well sorry :/

jbalibeux, 2020-10-20:
> hello, I'm looking to perfect my bridges. especially the end of bridges.
> I activate the bridge parameters. with an extrusion at 70 (approximately according to my tests) and with a little freewheel [...]
I'm trying to correct the display of images .. now

smartavionics, 2020-10-20:
Hi, no problem with the translation. There is a setting to help it is called Bridge Wall End Boost. It increases the flow for the wall lines immediately following the bride. Hope this helps!

jbalibeux, 2020-10-20:
this is exactly what I was looking for, thank you very much, and sorry for reporting a problem that came from me XD

I close this issue :)

closed 2020-10-20 by jbalibeux

smartavionics, 2020-10-20:
OK, happy printing!