smartavionics/Cura #72: I suspect that max resolution is only applied to outer walls?

kulfuerst, 2020-10-22:
Hey,

I have long been struggling with my printer stuttering in long curves when printing via USB. This is a well-known issue caused by low USB bandwidth. One way to fix this is to lower the max resolution in Cura. I have done this and I can now observe something very curious. My printer still stutters, but ONLY when printing inner walls. On outer walls, there is absolutely no stuttering. This is while printing both at the same speed.

It is just a suspicion, but could it be that Cura applies the max resolution setting only to the outer walls? It would be fantastic if you could look into this, since this could possibly solve a long going issue for me. Thank you!

smartavionics, 2020-10-22:
Hi, please supply a sample gcode file that has this problem. Thanks.

smartavionics, 2020-10-22:
And the project file would be useful too, thanks.