smartavionics/Cura #74: How hard would ARM64 support be?

Botspot, 2020-10-30:
I have a Raspberry Pi 4, and I use the 64-bit version of Raspberry Pi OS.  
This Cura build currently does not have an arm64 release as far as I can tell.

Would you consider adding a release of Cura that's compatible with my setup?

Thanks in advance.

smartavionics, 2020-10-30:
Yes, I will make an 64 bit release for the Pi 4 when the 64 bit OS becomes stable. It hasn't got there yet, has it?

smartavionics, 2020-10-30:
I decided to start down that road and have installed the 64 bit OS on a Pi 4. Will let you know as soon as I have anything to report.

Botspot, 2020-10-30:
> Yes, I will make an 64 bit release for the Pi 4 when the 64 bit OS becomes stable. It hasn't got there yet, has it?

I've been using it for months and it's been 100% stable as far as I can tell. The only reason it's not out of "beta" stage is because they haven't completely finished the graphics drivers. (I think that was their reason at least)

> I decided to start down that road and have installed the 64 bit OS on a Pi 4. Will let you know as soon as I have anything to report.

Thank you. BTW I've made a Raspberry Pi app store called [Pi-Apps](https://github.com/Botspot/pi-apps) and your Cura build is in it. Oh and Pi-Apps has over 14,000 users.

smartavionics, 2020-11-02:
Hi @Botspot , I have made an aarch64 release (https://github.com/smartavionics/Cura/releases/tag/20201023), it appears to work OK but has not been tested much. Please let me know if you come across any problems.

Botspot, 2020-11-04:
It works great for me.

As Cura is now in [Pi-Apps](https://github.com/Botspot/pi-apps), would you consider adding a little badge to your README?
```
[![badge](https://github.com/Botspot/pi-apps/blob/master/icons/badge.png?raw=true)](https://github.com/Botspot/pi-apps)
```
[image]

andrewufrank, 2020-11-16:
I hope that for the raspi 4 the solution is simpler because it has a streamlined booting procedure. The raspi4 starts with two partitions on a SD card (or an SSD) with a simpler boot sequence. 
I think it will only require a method to write the correct root device names into /boot/firmware commandline.txt or config.txt, probably adding also the kernel files required. I have the impression that changing in config.txt the 'root=LABEL=.." value is sufficient as long as the kernel remains the same.

Botspot, 2020-11-16:
@smartavionics The above comment is off-topic. I suggest you delete it.

closed 2021-02-14 by Botspot