smartavionics/Cura #76: z-hop on every travel move

rooiejoris, 2020-11-08:
**problem
At the moment I am printing quite a lot on a large FDM machine (1000x600x600) and then when traveling I would like to z hop. But z-hop in combination with connect infill doesn't work because then it doesn't retract (and so it doesn't z-hop)...
I think the same not hopping is in the bottom layers when the lines are not connected

**(possible)solution(s)
My request is a z-hop option for every travel move
or a retract every travel move (so the z-hop on retract works on every travel move)

I use the z-hop to prevent hitting the object. Expecially with large objects the speed goes up and the moves are longer. A workaround is to not connect infill lines, but hat has other mayor disadvantages (sometimes)

If any remarks/questions or if I forgot something to mention, please let me know.
(I have no idea how easy or difficult this request is.)

smartavionics, 2020-11-09:
Hi, doesn't turning off combing and enabling normal z-hop give you what you want? That should use retraction for all travel moves longer than the min distance and z-hop will be used on all retracted moves.

I don't understand how connecting the infill lines stops retraction because while it's connecting the infill lines it's extruding not simply travelling.

rooiejoris, 2020-11-09:
Ah sorry, my mistake...! you are right, it is the combing that affects the retraction and thus the z-moves, sorry...!!

But than the request still is: an option to do a z-hop on every travel move (because the no combing has other disadvantages)

smartavionics, 2020-11-09:
> But then the request still is: an option to do a z-hop on every travel move (because the no combing has other disadvantages)

What are the other disadvantages of no combing that would not be present if you could ask for z-hop on every travel move?

rooiejoris, 2020-11-09:
> What are the other disadvantages of no combing that would not be present if you could ask for z-hop on every travel move?

I think like the disadvatages of every 'no-combing' print, like stringing and small marks on the outer wall where the nozzle travels in and out the walls

Thinking of zo hop on travels, maybe retract on combing is a better solution. Then you can still use other options as 'retraction extra prime amount'....

(good questions btw... : )

smartavionics, 2020-11-09:
> I think like the disadvatages of every 'no-combing' print, like stringing and small marks on the outer wall where the nozzle travels in and out the walls

That is true when the nozzle is not z-hopped but if you are using retraction and z-hop it really shouldn't leave filament on walls when it crosses them (for PLA like stuff, at least). I guess if the filament is super runny then there could be more of an issue with ooze and then using a combed travel path could avoid blobs where the nozzle crosses a wall.

Perhaps a retract (and, therefore, z-hop) option on combing could be useful for some filament types...

rooiejoris, 2020-11-09:
> Perhaps a retract (and, therefore, z-hop) option on combing could be useful for some filament types...

That would be perfect...!!!

When can I test it... : )

cheers / joris

smartavionics, 2020-11-09:
Hold on, why can't you just set the max comb distance without retract to a small value and that would have the same effect as forcing combing to retract?

rooiejoris, 2020-11-09:
I was just kidding... : ) I like to think in solutions for the current situation.

> why can't you just set the max comb distance without retract to a small value

Didn't know of that setting...!! This could be the solution. Even better than 'my' suggested option I think, since you now can choose that for short distances to do no Z-hop which saves print time.

I think this issue can be closed then...!?

THANKS A LOT....!

Turning Combing off is quite a good solution by the way. I used that before (but forgot the trick?!).

smartavionics, 2020-11-09:
OK, I'll close this, cheers!

closed 2020-11-09 by smartavionics