smartavionics/Cura #77: Release 20201111 only has ARM assets e.g. no Windows installer

Sophist-UK, 2020-11-11:
Today's release only has the following executable assets:

- Cura-mb-master-aarch64-20201111.AppImage
- Cura-mb-master-armhf-20201111.AppImage
- Cura-mb-master-x86_64-20201111.AppImage
- mesa-aarch64-20201101.tar.gz
- mesa-armhf-20201107.tar.gz

The previous release had the following:

- Cura-mb-master-aarch64-20201023.AppImage
- Cura-mb-master-armhf-20201023.AppImage
- Cura-mb-master-macos-20201023.dmg
- Cura-mb-master-win32-20201023.exe
- Cura-mb-master-win64-20201023.exe
- Cura-mb-master-x86_64-20201023.AppImage
- mesa-aarch64-20201101.tar.gz
- mesa-armhf-20201107.tar.gz

smartavionics, 2020-11-11:
Fuck! I knew that someone would point this out. I do the easy builds first and then the others get done as I get around to it. Don't worry, they'll be done in the next 24 hours.

Sophist-UK, 2020-11-11:
Language, language!! ;-)

I am not worried about any delay in getting a Windows version - I just wanted to make sure that it hadn't been forgotten since there was nothing to indicate it was delayed.

closed 2020-11-11 by Sophist-UK

smartavionics, 2020-11-11:
OK, the win64 release has landed, enjoy!