smartavionics/Cura #78: [4.4.1] infill support unexpected behaviour

rooiejoris, 2020-11-15:
- (settings or bug?!)

**Application version**
4.4.1

**Platform**
Mac 10.14.6

**Printer**
Custom FFF and UM2+

**Reproduction steps**
Slice with infill support checked with Custom FFF and slice with custom support checked with UM2

**Screenshot(s)**
custom fff printer profile:
[image]

UM2+ printer profile:
[image]

**Actual results**
On the custom FFF printer profile, the infill and infill support structure are generated, but some layers the infill structure covers all of the object, even te 'hollow' parts
On the UM2+ printer profile only the infill support structure is generated, but not the infill structure (side note: i think this is intended, but would be a nice feature to combine infill and support infill, maybe this is possible with copying the same object and use the per object settings...)

On the custom fff printer profile the infill support structure is also not completely from the bottom to the top.
[image]

please note that I didn't find any stl errors with meshmixer, but I did upload it to netfabb cloud to be sure. No difference in the repaired file and e the original file.

**Expected results**
Not sure exactly, but what NOT should happen is that infill lines are present 'outside' the object in the hollow space.
I do think this should be a configuration error.

**Project file**
I can and will send if you like, but it is a pretty large file: 14,2MB (STL file is 32,2 MB)

**Log file**
[log.txt](https://github.com/smartavionics/Cura/files/5543148/log.txt)

**Additional information**
(Extra information relevant to the issue.)

rooiejoris, 2020-11-15:
Did some more testing, I suspect that the strange behaviour is when there is not enough space to make an infill structure. More dense infill and/or less infill overhang angle makes it dissapear.
This can maybe lead to a feature request: always print the 'outline' of the inner support structure then it doesn't matter if any infill can be printed or not.

smartavionics, 2020-11-16:
I think the Cura people are working in this area so it's probably worth waiting to see what they come up with.