smartavionics/Cura #79: Dual extrusion pauses during extruder change for heatup, oozing a blob of filament

Jdbye, 2020-11-16:
**Application version**
20201023

**Platform**
Windows 10

**Printer**
Geeetech A20M

**Reproduction steps**
Print with dual extrusion, single hotend, prime tower enabled, and "prime all extruders used on first layer" enabled.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
On the first extruder change, there is a "M104" (start heating extruder) and "M109" (heat extruder and wait) command inserted, which even though the nozzle is already hot, causes the printer to pause for a few seconds before it moves to the prime tower to prime the new color. Depending on where this happens it can leave a blob of filament attached to the first layer of the print, ruining the print once it tries to print over it and instead drags it along with it, at worst ripping the print loose, at best leaving a significant imperfection in the print.
This does not happen on subsequent layers, only on the first nozzle change (which in my case is on the first layer)

**Expected results**
Since the printer is using a shared hotend, there is no need to wait for the nozzle to heat on extruder change. The behavior should be the same as on subsequent layers, where there is no heating gcode inserted on extruder change, and the printer moves immediately to the prime tower to prime the new color.

**Project file**
https://www.dropbox.com/s/5di7g5izgxvv6hh/GA_3C_Lizard_3.3mf?dl=0

**Log file**
https://pastebin.com/rHWtBHBy

**Additional information**
My printer prints very well with dual extrusion/color mixing now apart from that one issue, and it's all thanks to this fork. Thanks for the hard work.

smartavionics, 2020-11-16:
Do you have extruders share heater option enabled in the printer setup? Enabling that should zap the extra M10[49]s.

smartavionics, 2020-11-16:
I tried loading your project but it didn't load OK so I can't slice your project, sorry.

andymozza, 2020-11-16:
I think the dual extrusion changes still let you set different temperatures for each nozzle though right? So it has to pause to get to these temperatures still if they are different, as people still wanted the option to print with two slightly differing filaments - do you have them set the same temp ? 
( check all temps including standby temp ).
The changes if I remember just remove the cool down of the old tool heater on nozzle change, not the temp change to the new tool if it's different?

smartavionics, 2020-11-16:
Hi Andy, thanks for that, I believe you're right, the temps need to be the same for all of the extruders.

Sophist-UK, 2020-11-16:
I think that the hot end not in use is allowed to cool to a stand-by temperature (typically about 150C instead of 190C-200C), so when you change extruder then the new hot end has to heat up again to temperature.

It might be possible (via a Post-Processing script) to work back from an extruder change and insert gCode for a partial pre-heat a few seconds earlier.

smartavionics, 2020-11-16:
Yes but here there is only one hotend so all the extruders need to have the same printing and standby temps to avoid waits.

andymozza, 2020-11-16:
@Sophist-UK if you enable the new shared heater option in the printer settings then the cool down when a tool is not used doesn't happen as this would affect the printing through the only shared heater, but the change in temp set for the individual nozzle printing temperatures would still happen.

Jdbye, 2020-11-16:
> 
> 
> Do you have extruders share heater option enabled in the printer setup? Enabling that should zap the extra M10[49]s.
Of course. That was the first thing I changed.

> 
> 
> I tried loading your project but it didn't load OK so I can't slice your project, sorry.
Strange. I just downloaded the file and opened it fine. Are project files printer specific or something?
I got the models from here, imported the .stls, set the extruder per model and then merged them, like you would typically do for multi color prints, so if it helps, you can try that: https://www.thingiverse.com/thing:2332398

smartavionics, 2020-11-16:
When I load that project file, it fails...

[image]

The log contains this...
```
2020-11-16 20:09:02,633 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [67]: Parse error in function (=3.5) for setting: invalid syntax (eval, line 1)
2020-11-16 20:09:02,633 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [67]: Parse error in function (=0.4) for setting: invalid syntax (eval, line 1)
2020-11-16 20:09:02,634 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [67]: Parse error in function (=0.0001) for setting: invalid syntax (eval, line 1)
2020-11-16 20:09:02,634 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [67]: Parse error in function (=machine_gcode_flavor != "UltiGCode") for setting: invalid syntax (eval, line 1)
2020-11-16 20:09:02,635 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [67]: Parse error in function (=machine_height) for setting: invalid syntax (eval, line 1)
2020-11-16 20:09:02,635 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [67]: Parse error in function (=0) for setting: invalid syntax (eval, line 1)
2020-11-16 20:09:02,635 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [67]: Parse error in function (=0) for setting: invalid syntax (eval, line 1)
2020-11-16 20:09:02,636 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [67]: Parse error in function (=0.001) for setting: invalid syntax (eval, line 1)
2020-11-16 20:09:02,636 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [67]: Parse error in function (=10) for setting: invalid syntax (eval, line 1)
2020-11-16 20:09:02,636 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [67]: Parse error in function (=4) for setting: invalid syntax (eval, line 1)
2020-11-16 20:09:02,637 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [67]: Parse error in function (=False) for setting: invalid syntax (eval, line 1)
2020-11-16 20:09:02,637 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [67]: Parse error in function (=machine_width) for setting: invalid syntax (eval, line 1)
2020-11-16 20:09:02,638 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [67]: Parse error in function (=machine_nozzle_offset_x) for setting: invalid syntax (eval, line 1)
2020-11-16 20:09:02,638 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [67]: Parse error in function (=machine_depth) for setting: invalid syntax (eval, line 1)
2020-11-16 20:09:02,638 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [67]: Parse error in function (=machine_nozzle_offset_y) for setting: invalid syntax (eval, line 1)
2020-11-16 20:09:02,638 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [67]: Parse error in function (=False) for setting: invalid syntax (eval, line 1)
2020-11-16 20:09:02,661 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [67]: Parse error in function (=3.5) for setting: invalid syntax (eval, line 1)
2020-11-16 20:09:02,661 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [67]: Parse error in function (=0.4) for setting: invalid syntax (eval, line 1)
2020-11-16 20:09:02,662 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [67]: Parse error in function (=0.0001) for setting: invalid syntax (eval, line 1)
2020-11-16 20:09:02,663 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [67]: Parse error in function (=machine_gcode_flavor != "UltiGCode") for setting: invalid syntax (eval, line 1)
2020-11-16 20:09:02,663 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [67]: Parse error in function (=machine_height) for setting: invalid syntax (eval, line 1)
2020-11-16 20:09:02,664 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [67]: Parse error in function (=0) for setting: invalid syntax (eval, line 1)
2020-11-16 20:09:02,665 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [67]: Parse error in function (=0) for setting: invalid syntax (eval, line 1)
2020-11-16 20:09:02,665 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [67]: Parse error in function (=0.001) for setting: invalid syntax (eval, line 1)
2020-11-16 20:09:02,666 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [67]: Parse error in function (=10) for setting: invalid syntax (eval, line 1)
2020-11-16 20:09:02,666 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [67]: Parse error in function (=4) for setting: invalid syntax (eval, line 1)
2020-11-16 20:09:02,667 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [67]: Parse error in function (=False) for setting: invalid syntax (eval, line 1)
2020-11-16 20:09:02,667 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [67]: Parse error in function (=machine_width) for setting: invalid syntax (eval, line 1)
2020-11-16 20:09:02,667 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [67]: Parse error in function (=machine_nozzle_offset_x) for setting: invalid syntax (eval, line 1)
2020-11-16 20:09:02,668 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [67]: Parse error in function (=machine_depth) for setting: invalid syntax (eval, line 1)
2020-11-16 20:09:02,668 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [67]: Parse error in function (=machine_nozzle_offset_y) for setting: invalid syntax (eval, line 1)
[... cut ...]
2020-11-16 20:09:02,685 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [67]: Parse error in function (=machine_depth) for setting: invalid syntax (eval, line 1)
2020-11-16 20:09:02,686 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [67]: Parse error in function (=machine_nozzle_offset_y) for setting: invalid syntax (eval, line 1)
2020-11-16 20:09:02,686 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [67]: Parse error in function (=False) for setting: invalid syntax (eval, line 
```

Jdbye, 2020-11-17:
> 
> 
> When I load that project file, it fails...
> 
> [image]
> 
> The log contains this...
> 

Could it be because I've modified the Geeetech A20M definitions/extruders (in main Cura folder) for virtual extruder support?
[geeetech_A20M.def.zip](https://github.com/smartavionics/Cura/files/5553380/geeetech_A20M.def.zip)

smartavionics, 2020-11-17:
> Could it be because I've modified the Geeetech A20M definitions/extruders (in main Cura folder) to increase the extruders to 5 (for virtual extruders)?

Perhaps, I don't know.

Now, have you seen what has been written above about ensuring that all the extruders have been configured to have the same printing and standby temperatures? When that is done, do you still see any M109s that shouldn't be there?

Jdbye, 2020-11-17:
> Perhaps, I don't know.
>
> Now, have you seen what has been written above about ensuring that all the extruders have been configured to have the same printing and standby temperatures? When that is [...]

I've just double checked, everything is identical. Both the filament settings, and the extruder specific settings.
Can you see if you are able to load the project with the modified definitions/extruders?

smartavionics, 2020-11-17:
Can you please provide a sample gcode file, I'll just take a look at that, thanks.

Jdbye, 2020-11-17:
> 
> 
> Can you please provide a sample gcode file, I'll just take a look at that, thanks.

Sure.
[GA_3C_Lizard_3.zip](https://github.com/smartavionics/Cura/files/5554490/GA_3C_Lizard_3.zip)

smartavionics, 2020-11-17:
My hunch is that because your start gcode contains M104/M109, cura is failing to realise that the temperature is already 210 when it gets to the first tool change. Could you please comment out/remove the M104/M109 in the start gcode and then see if the M104/M109 still get generated before the first tool change. Thanks.

Jdbye, 2020-11-17:
> My hunch is that because your start gcode contains M104/M109, cura is failing to realise that the temperature is already 210 when it gets to the first tool change. Could you please comment [...]

That has indeed solved it, but it's not without complications. Since now Cura inserts its own heating G-code before my start G-code, the nozzle will be hot when the bed leveling code is running, so it's going to ooze and maybe leave behind filament on the bed, which can ruin the print if I'm not paying attention for a few minutes at the start of every print.

It's certainly better than oozing during the print though, so it will work for now.
Is this behavior easy to change, so that I won't need this workaround in the future?

smartavionics, 2020-11-17:
So at the moment, when the start gcode contains M10[49], cura doesn't output its own M10[49] and assumes the temperature is still 0 and therefore does the wait when the extruder change occurs. I think there's an argument for making cura assume the temperature is the printing temperature (the temp set in the profile) even if it doesn't output the initial temperature setting gcode. That would then stop the subsequent wait on extruder change.  It would be then up to the user to ensure that the temperature set in the profile matches the temperature used in the start gcode. I will ponder this...

Jdbye, 2020-11-17:
The strange thing is that there is code in Cura that's supposed to handle this. It just doesn't seem to be working correctly in this particular scenario.
It's not just any M104/M109 command that works to stop Cura inserting its own heating G-code. It has to specifically contain {material_print_temperature_layer_0}
So they clearly thought of this already, and it seems like an oversight that it works as expected in other cases but not on the first nozzle change.

smartavionics, 2020-11-18:
I have created a new CuraEngine.exe that contains a fix for this issue, could you please drop that into your Cura installation and try it out. You can find it at https://www.dropbox.com/s/nd0f0cyp4gceyix/CuraEngine.exe?dl=0

Jdbye, 2020-12-07:
Sorry for the late reply. I tried your fix and it seems to slice correctly. I haven't tested printing with it yet, as I haven't been printing anything with color mixing, it's not something I use often. I will update when I do, but looking at the G-code I see no reason why it wouldn't work correctly.

smartavionics, 2020-12-07:
Hi, you may wish to install the latest release (https://github.com/smartavionics/Cura/releases/tag/20201121) which contains the above mentioned fix and a few other changes.

Jdbye, 2020-12-07:
Will do. And thanks for coming up with a fix so quickly!