smartavionics/Cura #80: [4.4.1] connect infill lines stutter while walls move smoothly

rooiejoris, 2020-11-19:
**Application version**
[4.4.1]

**Platform**
MAC

**Printer**
custom FDM

**Reproduction steps**
slice with connect infill lines

**Screenshot(s)**
[image]
gcode with gcode.ws website

**Actual results**
some 'moves' are introduces which causes stuttering

**Expected results**
no extra 'non-moves' included

**Project file**
can be provided, but attached gcode file

where does these green spots come frome...!?
I suspect them from making themachine stuttering

smartavionics, 2020-11-20:
Does this still occur using my latest release? Perhaps it has already been fixed?

rooiejoris, 2020-11-20:
just sliced in 4.8.0 and it is A LOT better...!!
But there is still a 'darker' spot on some places (almost nowhere though)
[image]

The thing is, I don't have/know the tools to see which line is this gcode? i use gcode.ws website and repetier, but both are not ideal to figure out what exactly happens.

(btw sorry for bothering more than i normally do, let me know if it get annoying...)

attached the gcode
[CFFFP_printtest01b.gcode.zip](https://github.com/smartavionics/Cura/files/5572061/CFFFP_printtest01b.gcode.zip)

smartavionics, 2020-11-20:
My builds are quite different from the Ultimaker releases with regard to this kind of thing so I can only really comment on the performance of my builds.

The code viewer I use is an old copy of the craftware slicer, it has a really excellent code viewer which gives you all the info you need when looking at line segments.

rooiejoris, 2020-11-20:
with 4.8.0  i mean your latest release...! not using ultimaker cura anymore...! I like all the features and overall better performance of 'your' cura.

will check out craftware...!

smartavionics, 2020-11-20:
Ah, OK, well it would perhaps be better if you report my build dates rather than the UM release versions.

Anyway, here's a sample of the the craftware slicer being used as a gcode viewer, you can see that it provides good info. It's a little buggy but still miles (kms?) better than most other viewers.

[image]

rooiejoris, 2020-11-20:
now investigating with craftware, but I think I have a setting enabled that wants to adjust E values

[image]

will investigate more, but I have some ideas... with extra prime settings...

rooiejoris, 2020-11-20:
> Ah, OK, well it would perhaps be better if you report my build dates rather than the UM release versions.

where do I find those... : ) i was already struggling to find the 4.8.0...

smartavionics, 2020-11-20:
The gcode files have a comment like this...

> ;Generated with Cura_SteamEngine mb-master-20201111

And the build files all are dated so it should be pretty obvious.

rooiejoris, 2020-11-20:
OK, will use the date of the application files then...!
Before this morning: 20200208
Now: 20201111

rooiejoris, 2020-11-20:
> EDIT:

(and have to learn etiquettes here as well, what is best practice, edit a post with more info or a reply?)

smartavionics, 2020-11-20:
I think new replies are better because edits to earlier posts often get missed.

rooiejoris, 2020-11-20:
Maybe close the topic. I think the conclusion is:
- a user/settings problem with 'extra prime min travel' set to 0 causes some unnecessary gcode
- why is this very small travel move made (but that's a very low priority, since there is a good workaround)

(THANKS again...!)

smartavionics, 2020-11-20:
Where exactly is this extra travel occurring? Can you locate it in the g-code? Perhaps it's worth investigating?

rooiejoris, 2020-11-20:
> now investigating with craftware, but I think I have a setting enabled that wants to adjust E values
> 
> [image]

yes here...!
but don't know which line number
I noticed because i saw the comment 'adjusted e by'  and then made the link to the extra prime

rooiejoris, 2020-11-20:
btw, there are some 'funky' move in this gcode as well regarding to connect infill lines
some unnecessary up forward an backward moves on the same path...

smartavionics, 2020-11-20:
I found that small travel move and can't imagine where it's coming from. Do you have a project file for this I can look at?

rooiejoris, 2020-11-20:
here we go...!
[CFFFP_resp03c_bottom.3mf.zip](https://github.com/smartavionics/Cura/files/5572766/CFFFP_resp03c_bottom.3mf.zip)

now printing with the right settings and almost no stuttering with speed cranked to 200% original speeds are 35 outer wall and 50 for the rest. a little stuttering n the infill or inner walls is not a problem now, I suspect octoprint for that and have other options to test.

are you familiar with this plugin for octoprint to convert gcode to arcs...!?
https://github.com/FormerLurker/ArcWelderPlugin
has some other disadvantages as well, but still interesting...

smartavionics, 2020-11-20:
Thanks for the project file, it's rather big, can you provide a project file for just the part that corresponds to the gcode file you provided earlier?

rooiejoris, 2020-11-20:
oops, my bet, wrong file... I indeed made a section of that file for testing:

[CFFFP_printtest01.3mf.zip](https://github.com/smartavionics/Cura/files/5573339/CFFFP_printtest01.3mf.zip)

I think here the setting 'extra prime min travel' is still on 0 here, please adjust to something >1 or what I did to 20mm or 50mm

smartavionics, 2020-11-20:
Thanks, I have made a change to the slicer to ignore travel moves < 5 um long when generating line sequences (used when outputting infill and skins). It removed several thousand short travels in your test so I am attaching it here for you to take a look at.

[CFFFP_printtest01-mb.gcode.zip](https://github.com/smartavionics/Cura/files/5573597/CFFFP_printtest01-mb.gcode.zip)

rooiejoris, 2020-11-20:
Several thousands...!!! that sound like a bug...  : )
its now printing for the next 24 hours, but will check!

Did it also remove that much short travels when setting  'extra prime min travel' on 20mm...?!
In other words, is the bug related to that setting?!

smartavionics, 2020-11-20:
> Did it also remove that much short travels when setting 'extra prime min travel' on 20mm...?!
In other words, is the bug related to that setting?!

Err, there's no such setting, do you mean "retraction minimum travel" ?

Sophist-UK, 2020-11-20:
I also noticed that the SmartAvionics-only infill types have stutters when they are curved infill - despite having maximum resolution / deviation set. Does your special Engine code for these infill types use the max resolution / deviation settings?

I am also beginning to think that Cura needs a Printer definition setting for maximum number of commands per second, and to slow down the print speed if this is exceeded. I think a post-processing script can also be used to achieve this - and may indeed be better because it has access to the entire layer and can calculate the print time for each segment and create a rolling calculation for the rate of commands, and then adjust the feed rate as necessary to slow them down and keep the commands to below the defined number per second. I will try to code such a post-processing script when I have the time.

smartavionics, 2020-11-20:
> I also noticed that the SmartAvionics-only infill types have stutters when they are curved infill - despite having maximum resolution / deviation set. Does your special Engine code for these infill types use the max resolution / deviation settings?

The Gyroid and Schwarz infill patterns don't abide by max res/dev but they do have 3 levels of resolution so you can make them low-res if required...

[image]
[image]
[image]

rooiejoris, 2020-11-20:
> Err, there's no such setting, do you mean "retraction minimum travel" ?

Nope I mean the setting, but now I remember this is a plugin, so postprocessed, and therefore i think more fragile for unnecessary travel moves.
These options are more or less only for larger scale printers, not for regular use i think.

[image]

smartavionics, 2020-11-20:
Ah, I don't have that plugin and I never use extra prime anyway.

rooiejoris, 2020-11-21:
> I am also beginning to think that Cura needs a Printer definition setting for maximum number of commands per second, and to slow down the print speed if this is exceeded.

@Sophist-UK would be very interesting for debugging/research purposes to have a gcode viewer who can show this kind of information. I have no idea how much commands my printer can handle and also no idea how much quality I loose when lowering the res and deviation settings.
And is there a relation when making sharp turnes? can it handle less commands then or does the accelartion nd jerk prevent the speed so it is not a problem...?!
(But this is a bit off topic... : )

Sophist-UK, 2020-11-21:
Well - I am on the way to having a PostProcessing script written for this.

AFAIK the limitation is on the USB baud rate rather than on the processors ability to actually process the gCode commands, but this may not be true and the processor may have limitations particularly on 8-bit boards. But either way it doesn't really matter because you still need to slow down the printing so that transmitting or processing the gCode can keep pace and not stutter.

Of course, there is nothing to stop you simply lowering your speed settings in order to prevent stutter, but this makes long straight extrusions slower as well.

And I think that this PostProcessing script will be complementary to the use of Max Resolution / Max Deviation because if you don't use those settings, then you **WILL** get many tiny extrusion segments and this script will slow the printing of those to an absolute crawl.

As for turns, yes it is off topic and I don't know the answer anyway.

Sophist-UK, 2020-11-22:
I now have a post processing script working. But to use it you need some way of working out approximately how many gCode instructions your printer can process per second. Once I have made a way to measure this using a specially crafted gCode file, I will upload these to a GitHub repo for people to try. If it works, then we can think about adding this to Cura itself.

Sophist-UK, 2020-11-22:
Ok - I printed samples of a ring saved as a high-resolution STL (i.e. lots of tiny faces) in PLA without this slowdown script and without Max Resolution / Max Deviation and I had a LOT of stuttering and poor quality walls. When I turned on the script, the printer slowed to the minimum speed I set and quality improved quite a lot, but the Min Speed I set was still too high and there was some small amount of stuttering.

Turning on Max Resolution  Max Deviation without my script, I had a very fast print and pretty good quality. When I turned on the script as well, I didn't have any stuttering, but the speed varied wildly up and down and as a consequence surface quality was slightly poorer.

So it seems like I need to implement some sort of averaging algorithm which takes into account the number of commands that the printer can queue and changes the feedrate more gradually in order to smooth out the printing. I will need to research how many commands the firmware queues ahead and enhance the script with the smoothing functionality, and see if that improves things.

In the mean time, I will add the current script to https://github.com/Sophist-UK/Cura-PostProcessingScripts at some point this evening in case anyone else wishes to try it.

Sophist-UK, 2020-11-22:
Script added for you to experiment with.

rooiejoris, 2020-11-22:
Maybe also drop this idea here...?!
https://community.octoprint.org/t/new-plugin-anti-stutter-need-testers/18077/270
It's pretty related, eg solving stuttering

Sophist-UK, 2020-11-22:
@rooiejoris The post you linked to is essentially about ArcWelder - which is an alternative way of reducing the number of line segments to Max Resolution / Max Deviation - but is only for those printers that supports arcs and splines i.e. G2/G3. ArcWelder is complementary to what I am looking at.

Actually @chendo is also looking at solving buffer underruns in Marlin - but from a completely different and complementary direction that may be more synergistic. But I can't find his email address or alternative to reach out to him.

bartwaw, 2020-11-25:
Hi, I've probably have similar problem.
Please look at picture. In this places printer stuttering.
[image]

smartavionics, 2020-11-25:
What you are seeing there are artifacts that are caused by the somewhat broken wall overlap compensation. UM are working on a new implementation of walls that should address this kind of problem.