smartavionics/Cura #81: [macos-20201121] Crash to desktop on macOS Big Sur

drelich, 2020-11-21:
**Application version**
Unable to open but downloaded file is macos-20201121

**Platform**
macOS Big Sur on late 2015 Macbook Pro (13 inch)

**Reproduction steps**
1. Download DMG file, mount it, copy app file to Application folder
2. Open app, confirm in Security it's okay to open it

**Actual results**
App icon bounces once or twice in the dock and then disappears. App does not launch.

**Expected results**
App window to appear.

smartavionics, 2020-11-21:
Hi, yes, my MacOS releases don't work on Big Sur because I haven't included the hacked version of geos.py that is required to work around a change that Apple have made in that release. I notice that Cura uses a rather old version of Shapely (which is the project that contains geos.py) so I shall investigate whether Cura works with a more recent version of Shapely which may work OK on Big Sur. If I can build a working Cura using a recent Shapely, I will make a MacOS build that you can test. Stay tuned...

drelich, 2020-11-21:
Thanks for letting me know. I recalled there was a problem launching Cura on Big Sur beta releases and copying an updated version of the geos.py helped. I used one found on https://github.com/Ultimaker/Cura/files/4935355/geos.py.gz

It sort of helped. The app looked like it launched (the icon in the dock stopped bouncing and basic Ultimaker Cura menu in the top bar popped up). But that's about it. Here's the Console readout again, if it helps.

```
Nov 21 17:25:15 Davids-MacBook-Pro com.apple.xpc.launchd[1]: Coalition Cache Hit: app<application.nl.ultimaker.cura.12388318.12388325(501)> [5699]
Nov 21 17:25:17 Davids-MacBook-Pro cura[32267]: getattrlist failed for /System/Library/Extensions/AppleIntelBDWGraphicsGLDriver.bundle/Contents/MacOS/AppleIntelBDWGraphicsGLDriver: #2: No such file or directory
Nov 21 17:25:17 Davids-MacBook-Pro cura[32267]: getattrlist failed for /System/Library/Frameworks/OpenGL.framework/Resources//GLRendererFloat.bundle/GLRendererFloat: #2: No such file or directory
```

smartavionics, 2020-11-21:
Hmm, those errors look like a completely different problem. Does the UM 4.8 release run OK ?

drelich, 2020-11-21:
Yep, 4.8 downloaded off Ultimaker's website runs just fine.

smartavionics, 2020-11-21:
Well, I don't know why that is occurring. Anyway, I have built a release using a more recent version of Shapely, you can find it at https://github.com/smartavionics/Cura/releases/download/20201121/Cura-mb-master-macos-20201121-1.dmg

Please let me know if that is any better/worse/same. Thanks.

drelich, 2020-11-21:
That was quick! :) Will test it but seems like my Mac is acting up. Now my Mail app does launch but freezes, no emails shown, etc. It seem to be referring to the same kinda problem as I reported earlier on. Will let you know when all is fixed. Cheers.

drelich, 2020-11-21:
Nope. Mail fixed but still getting error for the new version of your app:

```
Nov 21 20:07:53 Davids-MacBook-Pro com.apple.xpc.launchd[1]: Coalition Cache Hit: app<application.nl.ultimaker.cura.12411311.12411320(501)> [708]
Nov 21 20:07:55 Davids-MacBook-Pro cura[904]: getattrlist failed for /System/Library/Extensions/AppleIntelBDWGraphicsGLDriver.bundle/Contents/MacOS/AppleIntelBDWGraphicsGLDriver: #2: No such file or directory
Nov 21 20:07:55 Davids-MacBook-Pro cura[904]: getattrlist failed for /System/Library/Frameworks/OpenGL.framework/Resources//GLRendererFloat.bundle/GLRendererFloat: #2: No such file or directory
```

drelich, 2020-11-21:
Just checked something, your app seems to be having a different folder structure compared to the official 4.8 Cura app. Would this be the actual problem?

[image]

smartavionics, 2020-11-21:
At the moment, I think the error you are seeing is unrelated to Shapely. I will continue to investigate...

smartavionics, 2020-11-21:
I did some internet searching and it seems that quite a few applications are failing with similar messages so it's not just a Cura problem.

drelich, 2020-11-21:
Yeah, found a few discussions like that, too. But how come official Cura 4.8 works then? Your app is a fork, is it not?

smartavionics, 2020-11-21:
Yes, it's a fork but there's quite a few differences between my builds and the UM builds. I'm using different versions of various libraries and also it's being built on a different version of MacOS and using a different set of compilers/libraries, etc. The Mac I am using is very old and I am not using a recent release of the build tools. The Mac was given to me and apart from building Cura releases, I don't use it for anything else so it is unlikely that I will get a more recent machine.

drelich, 2020-11-21:
Ah, okay. Guess will need to settle for official Cura's inability to place the z-seam where I actually want it to and not where they want it :D Thanks again for all your help! It is very much appreciated :)

closed 2020-11-21 by drelich

smartavionics, 2020-11-21:
You're welcome. I'll let you know if I or anyone else can make a build of this cura that works on Big Sur.

smartavionics, 2020-11-24:
Hi, I've got another release to try, you can find it at https://github.com/smartavionics/Cura/releases/download/20201121/Cura-mb-master-macos-20201124.dmg

drelich, 2020-11-24:
Hi! I'm afraid that did not work. It starts and then it just freezes. I'd guess the difference between your macOS and the Big Sur is so significant that compiling a version for Big Sur may be impossible. 

Don't think it's worth wasting your time, to be honest. How many Big Sur users would actually use it? Thanks for trying though!

smartavionics, 2020-11-24:
I'm going to keep trying, bye for now...

dnasir, 2020-12-12:
> Hi! I'm afraid that did not work. It starts and then it just freezes. I'd guess the difference between your macOS and the Big Sur is so significant that compiling a version for Big Sur may be [...]

Don't say that. I'm one of those Big Sur users who would like to try out this version of Cura over the official one.

smartavionics, 2020-12-12:
Hi, I'm afraid no progress has been made with this issue. At the moment, I don't know what to try next to get it to work and because there are issues with the UM Cura release on Big Sur (and lots of other programs appear to have issues with Big Sur) I have decided to stop wasting my time and wait for a couple more releases from Apple before looking at this again. I will return to this at some point in the future.