smartavionics/Cura #82: Custom FFF printer slices to empty gcode

chambm, 2020-12-03:
**Application version**
master 20201121, upgraded in place from master 20200721

**Platform**
Windows 10.0.19042.546, Nvidia 770GTX

**Printer**
Custom FFF printer

**Reproduction steps**
1. I installed 20201121 over 20200721. My Custom FFF Printer setting was working fine in the latter.
2. Slicing in 20201121 with this printer always produces empty gcode.
3. I created a new Custom FFF printer, same result: empty gcode.
4. I created a new Ultimaker S5 printer, result is correct non-empty gcode.
5. I created a new Smoothie custom printer, result is correct non-empty gcode.
6. I tried many slicing profiles with the Custom FFF Printers, all produces empty gcode.
7. I tried disabling all my plugins except Octoprint, no change from the above steps.

**Screenshot(s)**
[image]

**Actual results**
Custom FFF printer creates empty gcode.

**Expected results**
Custom FFF printer creates non-empty gcode.

**Project file**
Shouldn't matter, it happens on any model for me.

**Log file**
```
2020-12-03 11:20:25,656 - DEBUG - [MainThread] UM.Controller.setActiveStage [180]: Setting active stage to PrepareStage
2020-12-03 11:20:25,659 - DEBUG - [MainThread] UM.Controller.setActiveView [108]: Setting active view to SolidView
2020-12-03 11:20:25,688 - DEBUG - [MainThread] cura.Machines.Models.NozzleModel._update [28]: Updating NozzleModel.
2020-12-03 11:20:25,869 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura master\resources\shaders\xray.shader]...
2020-12-03 11:20:33,159 - DEBUG - [MainThread] cura.Machines.Models.QualityManagementModel._update [306]: Updating QualityManagementModel.
2020-12-03 11:20:33,164 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [62]: Updating QualityProfilesDropDownMenuModel.
2020-12-03 11:20:33,169 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [33]: Updating CustomQualityProfilesDropDownMenuModel.
2020-12-03 11:20:36,574 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [155]: New error check scheduled.
2020-12-03 11:20:36,623 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [214]: Error check finished, result = False, time = 0.1s
2020-12-03 11:20:39,105 - INFO - [MainThread] cura.Settings.MachineManager.clearUserSettingAllCurrentStacks [453]: Clearing the setting [wall_thickness] from all stacks
2020-12-03 11:20:39,312 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [155]: New error check scheduled.
2020-12-03 11:20:39,364 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [214]: Error check finished, result = False, time = 0.1s
2020-12-03 11:20:39,986 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2020-12-03 11:20:40,135 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [247]: Starting to slice...
2020-12-03 11:20:40,135 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [266]: Going to slice build plate [0]!
2020-12-03 11:20:40,139 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [231]: Previous socket existed. Closing that first.
2020-12-03 11:20:40,142 - DEBUG - [Thread-6] CuraEngineBackend.StartSliceJob.run [164]: Before error checker took 0.0 seconds
2020-12-03 11:20:40,142 - DEBUG - [Thread-6] CuraEngineBackend.StartSliceJob.run [170]: Error checker took 0.0 seconds
2020-12-03 11:20:40,143 - DEBUG - [Thread-6] CuraEngineBackend.StartSliceJob.run [201]: Per object check took 0.0 seconds
2020-12-03 11:20:40,146 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [188]: Socket state changed to Listening
2020-12-03 11:20:40,146 - DEBUG - [Thread-6] CuraEngineBackend.StartSliceJob.run [210]: Remove layer old data took 0.0 seconds
2020-12-03 11:20:40,147 - DEBUG - [Thread-6] CuraEngineBackend.StartSliceJob.run [264]: Grouping objects took 0.0 seconds
2020-12-03 11:20:40,148 - DEBUG - [Thread-6] CuraEngineBackend.StartSliceJob.run [300]: Filtering object groups took 0.0 seconds
2020-12-03 11:20:40,170 - INFO - [MainThread] UM.Backend.Backend.startEngine [90]: Started engine process: C:\Program Files\Ultimaker Cura master\CuraEngine.exe
2020-12-03 11:20:40,253 - DEBUG - [MainThread] UM.Backend.Backend._backendLog [110]: [Backend] Calling engine with: ['C:\\Program Files\\Ultimaker Cura master\\CuraEngine.exe', 'connect', '127.0.0.1:49675', '']
2020-12-03 11:20:40,259 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2020-12-03 11:20:40,264 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] Cura_SteamEngine version mb-master-20201121
2020-12-03 11:20:40,266 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [192]: Socket state changed to Connected
2020-12-03 11:20:40,267 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] Copyright (C) 2020 Ultimaker
2020-12-03 11:20:40,271 - DEBUG - [MainThread] UM.Backend.Backend._onSocketStateChanged [181]: Backend connected on port 49675
2020-12-03 11:20:40,274 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2020-12-03 11:20:40,277 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] This program is free software: you can redistribute it and/or modify
2020-12-03 11:20:40,280 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] it under the terms of the GNU Affero General Public License as published by
2020-12-03 11:20:40,282 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] the Free Software Foundation, either version 3 of the License, or
2020-12-03 11:20:40,285 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] (at your option) any later version.
2020-12-03 11:20:40,288 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2020-12-03 11:20:40,291 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] This program is distributed in the hope that it will be useful,
2020-12-03 11:20:40,293 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] but WITHOUT ANY WARRANTY; without even the implied warranty of
[... 3 lines cut ...]
2020-12-03 11:20:40,305 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] You should have received a copy of the GNU Affero General Public License
2020-12-03 11:20:40,308 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] along with this program.  If not, see <http://www.gnu.org/licenses/>.
2020-12-03 11:20:40,352 - DEBUG - [Thread-6] CuraEngineBackend.StartSliceJob.run [346]: Building object group messages took 0.2036898136138916 seconds
2020-12-03 11:20:40,354 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onStartSliceCompleted [460]: Sending slice message took 0.2176969051361084 seconds
2020-12-03 11:20:40,477 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] [WARNING] Mesh has overlapping faces!
2020-12-03 11:20:40,593 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] [WARNING]  -s wireframe_roof_outer_delay="0.2" -s bridge_wall_material_flow="50" -s wipe_pause="0" -s mesh_temperature_dawdle_speed="10" -s infill_pattern="grid" -s infill_multiplier="1" -s machine_settings="0" -s jerk_travel_layer_0="30.0" -s cool_fan_lag="1" -s acceleration_wall_x="3000" -s extruder_prime_pos_abs="False" -s material_diameter="2.85" -s wireframe_enabled="False" -s cool_fan_speed_min="100.0" -s prime_tower_flow="100" -s wireframe_straight_before_down="20" -s material_flow_dependent_temperature="False" -s skirt_brim_speed="30.0" -s xy_offset="0" -s platform_adhesion="0" -s support_interface_offset="0.0" -s skirt_brim_material_flow="100" -s wall_extruder_nr="-1" -s machine_head_with_fans_polygon="[[-20, 10], [10, 10], [10, -10], [-20, -10]]" -s layer_start_x="0.0" -s machine_height="220" -s bottom_thickness="0.8" -s support_extruder_nr_layer_0="0" -s machine_nozzle_size="0.4" -s bridge_over_support="False" -s skirt_brim_minimal_length="250" -s wall_line_width_x="0.4" -s small_feature_max_length="0.0" -s draft_shield_height_limitation="full" -s material_bed_temp_prepend="False" -s mold_enabled="False" -s minimum_polygon_circumference="1.0" -s skin_edge_support_thickness="0.6" -s brim_line_count="20" -s travel_retract_before_outer_wall="False" -s wall_0_wipe [... cut ...]
2020-12-03 11:20:40,981 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [196]: Socket state changed to Closing
2020-12-03 11:20:40,983 - DEBUG - [MainThread] UM.Backend.Backend._onSocketError [220]: Socket debug: Arcus Error (13): Closing socket because other side requested close.
2020-12-03 11:20:40,985 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [198]: Socket state changed to Closed
2020-12-03 11:20:40,986 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [902]: Backend quit with return code 0. Resetting process and socket.
```

smartavionics, 2020-12-03:
Hi, please provide a project file that I can try to slice, thanks.

chambm, 2020-12-03:
Thanks for the fast reply.
[CFFFP_UPHILL_CONE.zip](https://github.com/smartavionics/Cura/files/5637588/CFFFP_UPHILL_CONE.zip)

smartavionics, 2020-12-03:
So that model has very thin walls, you need to enable "print thin walls" to get any walls...

[image]

chambm, 2020-12-03:
Sorry I just picked the first small simple model I could find. The model follows the same behavior as initially described. For both my old and new Custom FFF printer settings, I get empty gcode output from slicing it (not even the brim). For my Smoothie custom printer setting, I get the behavior you describe: brim only without thin walls, same as your picture with thin walls enabled.

smartavionics, 2020-12-03:
I tested your project on Linux, I will try on Windows now...

smartavionics, 2020-12-03:
I get the same results on Windows as I do on Linux. Please provide the complete cura.log and stderr.log files, thanks.

chambm, 2020-12-03:
That was the entire log. I emptied it out before reslicing to isolate the log around the slice. Here's a log with a slice on the Smoothie custom printer (takes 17 seconds to slice the more complex tank model) and on the Custom FFF printer (takes <1) second. 12:28 is the Smoothieware working slice, 12:42 is the Custom FFF slice.
```
2020-12-03 12:28:00,082 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [155]: New error check scheduled.
2020-12-03 12:28:00,086 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [214]: Error check finished, result = False, time = 0.0s
2020-12-03 12:28:01,381 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [247]: Starting to slice...
2020-12-03 12:28:01,382 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [266]: Going to slice build plate [0]!
2020-12-03 12:28:01,385 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [231]: Previous socket existed. Closing that first.
2020-12-03 12:28:01,388 - DEBUG - [Thread-3] CuraEngineBackend.StartSliceJob.run [164]: Before error checker took 0.0 seconds
2020-12-03 12:28:01,388 - DEBUG - [Thread-3] CuraEngineBackend.StartSliceJob.run [170]: Error checker took 0.0 seconds
2020-12-03 12:28:01,391 - DEBUG - [Thread-3] CuraEngineBackend.StartSliceJob.run [201]: Per object check took 0.0 seconds
2020-12-03 12:28:01,393 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [188]: Socket state changed to Listening
2020-12-03 12:28:01,394 - DEBUG - [Thread-3] CuraEngineBackend.StartSliceJob.run [210]: Remove layer old data took 0.0 seconds
2020-12-03 12:28:01,402 - INFO - [MainThread] UM.Backend.Backend.startEngine [90]: Started engine process: C:\Program Files\Ultimaker Cura master\CuraEngine.exe
2020-12-03 12:28:01,406 - DEBUG - [MainThread] UM.Backend.Backend._backendLog [110]: [Backend] Calling engine with: ['C:\\Program Files\\Ultimaker Cura master\\CuraEngine.exe', 'connect', '127.0.0.1:49674', '']
2020-12-03 12:28:01,408 - DEBUG - [Thread-3] CuraEngineBackend.StartSliceJob.run [264]: Grouping objects took 0.013037443161010742 seconds
2020-12-03 12:28:01,408 - DEBUG - [Thread-3] CuraEngineBackend.StartSliceJob.run [300]: Filtering object groups took 0.0 seconds
2020-12-03 12:28:01,463 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2020-12-03 12:28:01,508 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [192]: Socket state changed to Connected
2020-12-03 12:28:01,523 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] Cura_SteamEngine version mb-master-20201121
2020-12-03 12:28:01,524 - DEBUG - [MainThread] UM.Backend.Backend._onSocketStateChanged [181]: Backend connected on port 49674
2020-12-03 12:28:01,527 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] Copyright (C) 2020 Ultimaker
2020-12-03 12:28:01,531 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2020-12-03 12:28:01,534 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] This program is free software: you can redistribute it and/or modify
2020-12-03 12:28:01,537 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] it under the terms of the GNU Affero General Public License as published by
2020-12-03 12:28:01,541 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] the Free Software Foundation, either version 3 of the License, or
2020-12-03 12:28:01,544 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] (at your option) any later version.
2020-12-03 12:28:01,546 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2020-12-03 12:28:01,549 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] This program is distributed in the hope that it will be useful,
2020-12-03 12:28:01,552 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] but WITHOUT ANY WARRANTY; without even the implied warranty of
2020-12-03 12:28:01,555 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
2020-12-03 12:28:01,558 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] GNU Affero General Public License for more details.
2020-12-03 12:28:01,561 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2020-12-03 12:28:01,565 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] You should have received a copy of the GNU Affero General Public License
2020-12-03 12:28:01,569 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] along with this program.  If not, see <http://www.gnu.org/licenses/>.
2020-12-03 12:28:01,727 - DEBUG - [Thread-3] CuraEngineBackend.StartSliceJob.run [346]: Building object group messages took 0.31857848167419434 seconds
2020-12-03 12:28:01,729 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onStartSliceCompleted [460]: Sending slice message took 0.3476064205169678 seconds
2020-12-03 12:28:01,977 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] [WARNING] Mesh has overlapping faces!
2020-12-03 12:28:01,993 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] [WARNING]  -s retraction_hop_after_extruder_switch_height="1" -s default_material_print_temperature="2 [... cut ...]
2020-12-03 12:42:56,141 - DEBUG - [Thread-1] CuraEngineBackend.ProcessSlicedLayersJob.run [254]: Processing layers took 0.06299877166748047 seconds
2020-12-03 12:42:56,142 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [198]: Socket state changed to Closed
2020-12-03 12:42:56,143 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [902]: Backend quit with return code 0. Resetting process and socket.
2020-12-03 12:42:56,152 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onProcessLayersFinished [933]: See if there is more to slice(2)...
```

chambm, 2020-12-03:
Actually there's 2 slice attempts in the Custom FFF section. I toggled thin walls on/off.

smartavionics, 2020-12-03:
Please zip up and attach cura.log and stderr.log, thanks.

chambm, 2020-12-03:
I cleared out `%USERPROFILE%\AppData\Roaming\cura\stderr.log` earlier and after slicing it's empty. 
[cura.zip](https://github.com/smartavionics/Cura/files/5637995/cura.zip)

smartavionics, 2020-12-03:
Thanks for the files. At the moment, I can't see anything obviously wrong. BTW, does it behave the same if you don't use the Octoprint plugin?

chambm, 2020-12-03:
I haven't tried disabling that yet. It isn't causing problems with the Smoothie custom printer or in Cura 4.8, so I figure that isn't it. Quite a mysterious issue. However the new master did fix some issues I was having with tree supports, so I'm glad I upgraded! (adding a new custom printer and copying the settings isn't too a big a deal).

smartavionics, 2020-12-04:
Just been looking at your cura.log file and it doesn't appear to be (much) related to the printer in the .3mf file. So could you please shutdown cura, delete cura.log, start cura, load the .3mf file you gave me, do a slice, shutdown cura and attach the cura.log file to this issue, thanks.

chambm, 2020-12-07:
[cura.log](https://github.com/smartavionics/Cura/files/5655519/cura.log)

smartavionics, 2020-12-08:
Thanks for the log file. I've just twigged that you have wall_extruder_nr, wall_0_extruder_nr, infiil_extruder_nr all set to -1 (aka disabled). That's why there's no g-code being produced.

chambm, 2020-12-11:
I don't see where to adjust those in the GUI. Why did it change when upgrading from Cura-master 20200721 to 20201121? But not when upgrading from Cura-official 4.6 to 4.8? And why does creating a new Custom FFF printer keep those disabled extruder settings but new Custom Smoothie FFF printer has valid extruder settings? :/

smartavionics, 2020-12-11:
Sorry, I can't answer any of those questions as I have not worked on that part of Cura, ever.

chambm, 2020-12-11:
OK, well hopefully this helps anyone else that has the same issue. Thanks for investigating.

closed 2020-12-11 by chambm