smartavionics/Cura #83: [20201121] interesting behaviour connect infill lines

rooiejoris, 2020-12-18:
**Application version**
20201121, but also in 20201111

**Platform**
Mac OS

**Printer**
custom fff

**Reproduction steps**
slice with connect infill lines checked

**Screenshot(s)**
[image]

**Actual results**
The lines don't only connect, many times they 'end' somewhere.
Interesting thing is that this is sometimes a feature for my projects when i think 2 walls takes too long and then with connect infill a  large part of the wall is 2 walls.

**Expected results**
Connect the infill lines and don't print to a 'dead end'

**Project file**
I only have a rather large file atm and cannot really share it in publi

**Additional information**
not really high prio, but wanted to let you know... : )

cheers / joris

smartavionics, 2020-12-18:
Hi, please save a copy of the project file with the model removed and attach that to this issue. That way I will get to see the settings you are using even if I don't have the model. Thanks.

rooiejoris, 2020-12-19:
I will post the 3mf here including the model and will remove it afterwards
[CFFFP_block05_2020_infill.3mf.zip](https://github.com/smartavionics/Cura/files/5719143/CFFFP_block05_2020_infill.3mf.zip)

I 'traced' several layer, but 1223 is a good example I think

smartavionics, 2020-12-19:
OK, got it!

smartavionics, 2020-12-19:
Ah, for all infill patterns other than the ones that I have provided (gyroid, schwarz and honeycomb), the connect infill lines feature is implemented by UM code that I have never looked at (until today). It is complicated and I am too lazy to put the effort in to understanding where it could be failing. Sorry about that, life is too short! if this is a real problem for you, I recommend that you open an issue on the UM Cura repo.

closed 2020-12-19 by smartavionics

rooiejoris, 2020-12-19:
hey hey, no apologies needed...!!!!!
I am already so happy with all the other features and solved bugs and the support I get.

Maybe I can abuse the honeycomb infill for this, the other are too much 3d for now I think. But who knows, maybe I give that a try as well.

(good that you closed the issue, life is too short...!!)