smartavionics/Cura #88: Inner wall overlap compensation causing extremely thin inner wall.

kvnper, 2020-12-28:
**Application version**
v20201121

**Platform**
Win10 x64

**Printer**
QIDI X-Max

**Reproduction steps**
See 3mf in zip attachment and simply slice it.

**Screenshot(s)**
Inner wall overlap compensation on:
[image]
[image]

Inner wall overlap compensation off:
[image]
[image]

**Actual results**
Inner wall not printed correctly. Seems to be too thin when **inner wall overlap compensation** is enabled.

**Expected results**
Inner walls printed at correct width.

**Project file**

See 3mf file in zip attached
[XMAX_red cover v4.zip](https://github.com/smartavionics/Cura/files/5745868/XMAX_red.cover.v4.zip)

**Log file**

[cura.log](https://github.com/smartavionics/Cura/files/5745873/cura.log)

**Additional information**
n/a

kvnper, 2020-12-28:
Something else of note is that the inner wall line alternates between a green line and yellow line. The yellow side also has some odd behaviour with inner wall overlap compensation:

Green and yellow wall separation
[image]

The other end of the yellow wall with odd behaviour:
[image]

**NOTE:** This does not go away when turning off inner wall overlap compensation (Even though the green line does get fixed):
[image]
[image]

kvnper, 2020-12-28:
Update on yellow line: Turning on **Filter Out Tiny Gaps**  fixed this yellow line thinning. No difference seen with Compensate Inner Wall Overlap enabled or not with Filter Out Tiny Gaps enabled.

[image]

kvnper, 2020-12-28:
Update 2 on yellow thinning lines: I found more locations where this occurs but is not fixed by enabling **Filter Out Tiny Gaps**:

Screenshot from layer 14 amongst the grills:

[image]

smartavionics, 2020-12-28:
Yeah, those thinning yellow lines are not good but I'm not going to do any more work on that part of the slicer as all of my thin wall and gap filling code is going to be removed in the future when I merge in the upstream arachne code. However, I don't plan to merge that code until after UM have released a couple of versions of Cura that uses it.

smartavionics, 2020-12-28:
A couple of suggestions, using a wall line width of 0.5 I think helps and also setting the min wall flow to something like 50% removes a lot of the thin green lines.

kvnper, 2020-12-28:
Arachne looks quite nice. How far off, in your opinion, is it from release?

Yeah when I set the minimum wall flow it removes that wall completely and does not print anything there. I'll try the 0.5mm wall suggestion.

Thanks for your time.

smartavionics, 2020-12-28:
I am not involved with the development of Arachne so I don't have any idea of their timescales. I would think that unless they find some show-stopping problem that it will appear within the next couple of months. The first release and maybe the second that uses it will likely have some problems so I won't be looking to merge for a while. It's going to be quite a big job because my Cura has diverged from UM Cura over the years.

closed 2020-12-28 by smartavionics