smartavionics/Cura #89: Supports extend substantially beyond overhang

Sophist-UK, 2021-01-09:
**Application version**
Cura-mb-master-win64-20201121.exe (I think - but not certain)

**Platform**
Win 10 64

**Printer**
Dagoma DiscoEasy200

**Reproduction steps**
Sliced project in Ultimaker Cura 4.8.0.
Sliced same project in SmartAvionics Cura,

**Screenshot(s)**
Ultimaker:
[image]

SmartAvionics:
[image]

**Actual results**
1. Right hand supports very odd shape
2. Odd support 2/3 the way up the model

**Expected results**
Same supports in SmartAvionics as Ultimaker

**Project file**
[TEST.zip](https://github.com/smartavionics/Cura/files/5791830/TEST.zip)

smartavionics, 2021-01-10:
I sliced your project on Linux and it looks like your 4.8 image. I will try it on Windows and let you know if it is any different.

smartavionics, 2021-01-10:
I can reproduce the issue, a workaround is to reduce the maximum deviation setting to 0.05.

Sophist-UK, 2021-01-10:
'tis ok - I was only trying your version for this print to see if it corrected some printing flaws introduced by Ultimaker. But it doesn't.

Going to try Prusa Slicer next.

smartavionics, 2021-01-10:
What flaws, perhaps a setting change can help?

smartavionics, 2021-01-10:
looking at your profile, I would expect the print quality to be poor with noticeable underextrusion and ringing.

Sophist-UK, 2021-01-10:
I get underextrusion on a few layers at the start of the layer which reduces to zero gradually over 3 or 4 layers.

But generally print quality is excellent and I don't get underextrusion or ringing. If anything, I get slight over extrusion on inner walls and slight blobs at retraction which I have been unable to stop and which result in slight bumps on the top surface (which i can pare off with a stanley knife.

Sophist-UK, 2021-01-10:
On further consideration, the blobbing at retraction may be related - i.e. it is the material that has oozed at the end of the layer that is missing from the beginning of the next layer. Perhaps tuning of retraction parameters would help.

smartavionics, 2021-01-10:
As an experiment, try halving all your print speeds (keep the travel speed the same) and see what it does for your print quality.