smartavionics/Cura #90: Lines not rendering in preview

DeathDealer33, 2021-01-12:
The line preview is not rendering anything at all for me. Im on Ubuntu MATE 20.10 (Groovy Gorilla) 64-bit. Ive tried the 3 latest releases and they all do the same thing for me. I was on Ubuntu 20.10 and the layer lines were working. Not sure what preventing it from rendering any help would be greatly appreciated.
[image]

smartavionics, 2021-01-12:
Maybe the log files tell us what's going wrong. Please zip up and attach ~/.local/share/cura/stderr.log and ~/.local/share/cura/master/cura.log.

Sophist-UK, 2021-01-12:
I also had this issue one time - and had to close and restart Cura to get it to work.

DeathDealer33, 2021-01-12:
> I also had this issue one time - and had to close and restart Cura to get it to work.

I've tried 3 different versions, so that means I tried restarting it many times.

DeathDealer33, 2021-01-12:
> I also had this issue one time - and had to close and restart Cura to get it to work.

I'll get those uploaded in just a few minutes, I did end up trying the cura that is supplied by "sudo apt-get install cura". It's 4.6 cura but the lines are rendering in the preview.

DeathDealer33, 2021-01-12:
[cura.log](https://github.com/smartavionics/Cura/files/5802634/cura.log)
[stderr.log](https://github.com/smartavionics/Cura/files/5802636/stderr.log)

Here are the 2 logs, i looked at them but could identify what the issue is.

smartavionics, 2021-01-12:
Ah, this is on a Pi4 ? Looking at the logs, it appears you have a fairly recent mesa installation. In that case, try forcing the GLES version by prefixing the command like this...

MESA_GLES_VERSION_OVERRIDE="3.2" Cura-mb-master-aarch64-20201121.AppImage

DeathDealer33, 2021-01-12:
> Ah, this is on a Pi4 ? Looking at the logs, it appears you have a fairly recent mesa installation. In that case, try forcing the GLES version by prefixing the command like this...
> 
> MESA_GLES_VERSION_OVERRIDE="3.2" Cura-mb-master-aarch64-20201121.AppImage

Im pretty new to linux, i dont actually know what your talking about... I tried running that command in the appimage's directory in the terminal but it said the command isnt found.. Sorry to ask but what do you mean?

smartavionics, 2021-01-12:
What that is doing is running the appimage with an environment variable set just for that command. Another way of achieving the same is to first set and export the environment variable by typing this into the shell...

export MESA_GLES_VERSION_OVERRIDE="3.2"

And then running the appimage from the shell, perhaps you need to prefix the name with ./ ? i.e. ./Cura-mb-master-aarch64-20201121.AppImage

DeathDealer33, 2021-01-12:
Ok that did it^_^, do need to run cura this way every time it boots?

[image]

smartavionics, 2021-01-12:
You need that environment variable set every time you start Cura so you could put that export line into a script or you could put it in your ~/.bashrc file. Doing that would mean it was set for all programs that you run. That shouldn't cause a problem AFAIK.

DeathDealer33, 2021-01-12:
> You need that environment variable set every time you start Cura so you could put that export line into a script or you could put it in your ~/.bashrc file. Doing that would mean it was set for all programs that you run. That shouldn't cause a problem AFAIK.

Awesome I'll give that a shot, thanks so much for the help and the work you do here^_^ I finally have 4.8 Cura on my ras pi 4!! 🕺

smartavionics, 2021-01-12:
You're welcome.

I'll leave this issue open in case anyone else has the same issue.

DeathDealer33, 2021-01-12:
> You're welcome.
> 
> I'll leave this issue open in case anyone else has the same issue.

[image]

Ive added the line to my .bashrc file with nano in terminal and restarted a few times as well but no success. Am i doing something wrong?

smartavionics, 2021-01-12:
So what gets printed in a terminal if you enter "echo $MESA_GLES_VERSION_OVERRIDE" ?

DeathDealer33, 2021-01-12:
> So what gets printed in a terminal if you enter "echo $MESA_GLES_VERSION_OVERRIDE" ?

It print "3.2"

DeathDealer33, 2021-01-12:
> So what gets printed in a terminal if you enter "echo $MESA_GLES_VERSION_OVERRIDE" ?

but lines are not rendered unless i boot from a terminal with the "export MESA_GLES_VERSION_OVERRIDE="3.2" then booted. Thats weird : /

smartavionics, 2021-01-12:
Hmm, maybe your normal shell isn't bash and so .bashrc isn't being read?

DeathDealer33, 2021-01-12:
> Hmm, maybe your normal shell isn't bash and so .bashrc isn't being read?
Like i said i sort of new to ras pi/ linux, not sure what else i would be using.. Fresh Ubuntu MATE 20.10 (Groovy Gorilla) 64-bit on this ras pi 4 8gb. Should be using the default, i havnt changed anything.

smartavionics, 2021-01-12:
So if you enter "echo $SHELL" in your shell, what does it say?

DeathDealer33, 2021-01-12:
> So if you enter "echo $SHELL" in your shell, what does it say?

It prints "/bin/bash"

smartavionics, 2021-01-12:
I just noticed that in your image above, the editor says that .bashrc is only used with non-login shells. My .profile, that is executed for login shells does include .bashrc, does yours?
```
# ~/.profile: executed by the command interpreter for login shells.
# This file is not read by bash(1), if ~/.bash_profile or ~/.bash_login
# exists.
# see /usr/share/doc/bash/examples/startup-files for examples.
# the files are located in the bash-doc package.

# the default umask is set in /etc/profile; for setting the umask
# for ssh logins, install and configure the libpam-umask package.
#umask 022

# if running bash
if [ -n "$BASH_VERSION" ]; then
    # include .bashrc if it exists
    if [ -f "$HOME/.bashrc" ]; then
	. "$HOME/.bashrc"
    fi
fi

# set PATH so it includes user's private bin if it exists
if [ -d "$HOME/bin" ] ; then
    PATH="$HOME/bin:$PATH"
fi
```

DeathDealer33, 2021-02-01:
i ended up making a script to run the same command i do from the terminal to boot the program.