smartavionics/Cura #91: Allow changing  temperature per block of layers

paoletto, 2021-01-19:
I suppose it's impractical/impossible to change the extrusion temperature in specific areas like for the fan, but perhaps per block of layers?
If i were to try this to myself, what command should i insert manually in the g-code, to increase the temperature say from 205 to 230 after layer N?

smartavionics, 2021-01-20:
Use M104 SNUM to set the temperature without waiting and M109 SNUM to set the temperature and wait until that temp is reached. So, for example, to set the temperature to 220 without waiting put "M104 S220" into the gcode file at the point you want the temperature to start to increase. Obviously, because the temp changes are relatively slow, you would need to put it somewhat before the start of the region that has to be at that temp. Hope this helps.

paoletto, 2021-01-20:
thanks! that's pretty much what i had in mind.. If i were to add this to your fork, where would i be start looking at?
(in the medium run it will probably worth the time, instead of manually adding M109s and also losing the ability to print over ethernet)

smartavionics, 2021-01-20:
My fork already has the ability to change the temperature within a mesh so you can use a modifier mesh to change the temperature within a particular region of your print. Check out the mesh temperature delta and mesh dawdle speed and mesh dawdle time settings. This feature has not received much testing.

paoletto, 2021-01-21:
@smartavionics oh i see, then i suppose the issue i opened is invalid.
Out of curiosity, do you use M109s or M104s? and if the former, how does it work inside an arbitrary mesh possibly cutting through a layer?

smartavionics, 2021-01-22:
It uses M104s so it doesn't wait for the temperature to reach the set value.

closed 2021-02-01 by paoletto

paoletto, 2021-02-02:
@smartavionics so i tried to find this feature, but i failed..
This is what i get: https://imgur.com/bM7HSHh
(i have **show all** checked)

smartavionics, 2021-02-02:
Look in the experimental settings section...