smartavionics/Cura #92: Where did the ARM builds go?

Botspot, 2021-02-14:
I'm in charge of the world's most popular Raspberry Pi app store called [Pi-Apps](https://github.com/Botspot/pi-apps), and your ARM Cura builds are there for anyone to conveniently install and use. **Over 1500 people** have installed Cura through the store.

And that brings me to the present concern: In your [latest release](https://github.com/smartavionics/Cura/releases/tag/20210214), both aarch64 and armhf builds are **missing**.  
Was this a mistake? Or is this the new normal?

smartavionics, 2021-02-14:
> And that brings me to the present concern: In your latest release, both aarch64 and armhf builds are missing.
Was this a mistake? Or is this the new normal?

They will be uploaded shortly, it takes a while to create the 6 different builds for each release and I tend to do them in random order. The Linux builds are often first because they're easiest but today I did the MacOS build first because someone was waiting for it.

smartavionics, 2021-02-14:
They're there now...

closed 2021-02-14 by smartavionics

Botspot, 2021-02-14:
> They will be uploaded shortly, it takes a while to create the 6 different builds for each release and I tend to do them in random order. The Linux builds are often first because they're easiest but today I did the MacOS build first because someone was waiting for it.

Thank you.  
PS. have you considered adding the Pi-Apps badge to your readme?
`[[image]](https://github.com/Botspot/pi-apps)  `
[image]