smartavionics/Cura #96: Bridge mode visualisation

5axes, 2021-02-21:
It would be nice to be able to visualize also in the line types the visualization of the lines in bridge mode. A new type bridge could be added A new type could be added to the other existing types.

[image]

smartavionics, 2021-02-21:
Yes, that would be nice but it's actually not that easy to implement. However, UM have just added a new colour scheme which is line width and that will display bridge lines whose flow has been modified in different colours from the normal wall and skin lines. I've actually modified the UM code because when calculating the max and min line widths they include the travel lines and as they have bogus widths the display changes colour when you enable/disable the travel moves so that needed fixing. Anyway, my next release will have this feature. Here's an example that shows what I mean...

[image]

Sophist-UK, 2021-02-21:
I assume that the visualisation routine have to determined line type from comments in the gCode placed there by the Cura Engine. So adding better colouring for bridging would need changes to both Cura and Cura Engine. You might get a better response from the folks at Ultimaker if you raise this same issue on their repo.

smartavionics, 2021-02-21:
> I assume that the visualisation routine has to determined line type from comments in the gCode placed there by the Cura Engine. So adding better colouring for bridging would need changes to both Cura and Cura Engine. You might get a better response from the folks at Ultimaker if you raise this same issue on their repo.

Actually, it only looks at those comments when loading a gcode file. When slicing afresh, the line types are communicated from the engine to the UI.

I just think that the amount of work required to be able to show bridge lines in a different colour to the shell lines is too much for what is, arguably, a slightly redundant feature given that the print speed colour scheme and the new line width colour scheme will both make bridge lines more visible as long as they don't have the same print speed and flow as the normal wall and skin lines.

5axes, 2021-02-21:
As BRIDGE identification are already implemented in the G-Code, I was expecting an easiest integration.
[image]

Sophist-UK, 2021-02-21:
I assume that bridging finishing is indicated simply by the next `;TYPE` comment?

5axes, 2021-02-21:
> You might get a better response from the folks at Ultimaker if you raise this same issue on their repo.

My real enhancement request would be in this case a granularity of line type similar to PrusaSlicer :)

[image]

smartavionics, 2021-02-21:
TBH, I don't think the Cura visualisation is ever going to be much good. There's some fundamental problems it has which I doubt that UM are interested in fixing. Personally, If I want to get a good look at the gcode and all the details, I load the gcode into the craftware slicer which works as a very nice gcode viewer and gives you pretty much all the info you want.