smartavionics/Cura #97: Wrong restart value

5axes, 2021-02-21:
Release 20210217  Windows.

Wrong Retract value : Retraction value set to 4mm.   The value of the E axis increases gradually to finish at 140 mm at the end of the print.

[image]

[test-dog_50mm.zip](https://github.com/smartavionics/Cura/files/6018057/test-dog_50mm.zip)

5axes, 2021-02-21:
Just beacuse I'm Stupid ... Postprocessor script still active !!

closed 2021-02-21 by 5axes

smartavionics, 2021-02-21:
Ah, thanks for the update, I just sliced your project on Linux and it was using 4mm retraction as expected.

5axes, 2021-02-21:
Yes I'm Sorry  But it's not possible to delete an issue even if it's just because you are too stupid and  you open a repprt too quikly wihout deep analyse

Sophist-UK, 2021-02-21:
@5axes We have all done this at some point. Don't beat yourself up about it - just close the issue.