smartavionics/Cura #98: Error?

qwerty8224, 2021-02-24:
please see the last layer of filling (the nozzle passes 2 times for filling) is this normal?
[DXU_1x1_fixed.zip](https://github.com/smartavionics/Cura/files/6034074/DXU_1x1_fixed.zip)

smartavionics, 2021-02-24:
Hi, I don't understand the problem, can you provide some screen shots or a better description?

qwerty8224, 2021-02-24:
[image]
the nozzle is already moving over the printed filling

qwerty8224, 2021-02-24:
[image]

qwerty8224, 2021-02-24:
[image]

smartavionics, 2021-02-24:
I can see the extra infill problem now, I will investigate...

qwerty8224, 2021-02-24:
Thank. I tried my best))) sorry for my english.

smartavionics, 2021-02-24:
OK, I have found the bug, I am surprised it hasn't been discovered earlier. I will make a new release today that is fixed.

Thanks a lot for the report.

smartavionics, 2021-02-24:
Hi, I have made a new release (20210224), please try it and see if the problem has gone away.

I expect I will merge the arachne work into my cura but not until at least 2 releases by UM have been made that includes it. My cura and UM cura are now quite different so the merge will be painful.

qwerty8224, 2021-02-24:
OK. if there are any mistakes I will write.

smartavionics, 2021-02-24:
OK, thanks for the report.

closed 2021-02-25 by qwerty8224