smartavionics/Cura #101: 20210306 Crash on W10

5axes, 2021-03-06:
Last release Crash on Windows 10 just when I launch the slice operation. 

Occurs on every geometry not dedicate to a special part.   Printer Ender 3

[image]

In the Log file I have an error in relation to the "PrintInformation.py" 

And something regarding an user "**burto** ????" . It's not a account existing on my Pc.

IE : 
`File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\cura\UI\PrintInformation.py",
[cura.log](https://github.com/smartavionics/Cura/files/6095842/cura.log)
`

smartavionics, 2021-03-06:
Hi, I think it is the custom job prefix plugin that has a problem, can you try disabling that and see if the slice succeeds. If it does, then you need to raise an issue either on the upstream UM Cura repo, or on the provider of that plugin.

5axes, 2021-03-06:
I'm sorry I have switched back to the previous release .. Bu I don't think that issue can come from plugin because before to send my report , I have a fresh install in a new folder and the user  Roaming have been completely delete so normaly no 3d party plugin was used with this release.

smartavionics, 2021-03-06:
Well, in the log file it refers to:
```
File "C:\Users\5axes\AppData\Roaming\cura\master\plugins\CustomJobPrefix\CustomJobPrefix\PrintInformationPatches.py"
```

That's where the error is occurring.

5axes, 2021-03-06:
Yes I have seen this line ???  I should not have any plugin ... Ok I will try to re_install the last release.

smartavionics, 2021-03-06:
As far as I can tell, in the upstream UM code they recently removed the function that your log file says is undefined. I note that the custom job prefix plugin has recently been updated and it appears to include that function so maybe you just need to update that plugin?

5axes, 2021-03-06:
Ok no more crash :( but I have now something very strange on my first test part

Visualisation Layer 301 

[image]

Visualisation layer 302

[image]

[Slant_Roof_Paddle_Boat_Master.zip](https://github.com/smartavionics/Cura/files/6095918/Slant_Roof_Paddle_Boat_Master.zip)

5axes, 2021-03-06:
Ok I think it's time to stop ... for today.

Close and Save the project. Load again and everything is OK 

[image]

I'm Sorry to bother you with this false report !!

5axes, 2021-03-06:
I'm Sorry to bother you with this false report !!

closed 2021-03-06 by 5axes

smartavionics, 2021-03-06:
No problem. When you are living on the leading edge, stuff (code, plugins, cache files, etc.) can get out of sync. It happens to me also from time to time.