smartavionics/Cura #102: Error?

qwerty8224, 2021-03-13:
Layer 37 stands out from the rest ...
[image]

smartavionics, 2021-03-13:
Can you provide the project file, please?

qwerty8224, 2021-03-13:
[1DXU_Крепление дуги  к плоту (1).zip](https://github.com/smartavionics/Cura/files/6134477/1DXU_.1.zip)

smartavionics, 2021-03-13:
That change in the diameter is due to the hole horizontal expansion, setting it to 0, makes the step go away. There's nothing I can do to avoid this.

[image]

qwerty8224, 2021-03-13:
sorry for bothering you (((I just wanted to check this but didn't have time. Thanks)

closed 2021-03-13 by qwerty8224

smartavionics, 2021-03-13:
No problem, cheers!