smartavionics/Cura #109: [20210516] Strange skin line printing order

5axes, 2021-05-16:
If we don't use the Monotonic option the skin line printing order could be very strange. not realy a bug but could be more constant.
[image]

Same part in the regular 4.9.1 release.

[image]
[plate_with_indent.zip](https://github.com/smartavionics/Cura/files/6489932/plate_with_indent.zip)

smartavionics, 2021-05-16:
You don't say what layer that is but I can't reproduce the top image on any layer. For me, I get similar to the bottom image (but not identical).

5axes, 2021-05-16:
It's the last one ( N°25 with a Layer height of 0.2 mm) and the strangest thing is that indeed the lower layers are similar to what we could have with a standard version. Ie Layer N°23.
[image]

smartavionics, 2021-05-16:
My layer 25 looks like this...

[image]

I can't explain why yours doesn't look similar. What system are on, perhaps it is system dependent?

I need to finish for tonight now and I won't be able to look at this again until Tuesday. See you then...

5axes, 2021-05-16:
ok .. Yes it a very weird issue. I'm running Cura on Windows 10. But I cannot reproduce the issue every time.
Ie : Layer 25 .
[image]

new project , this time it's seems to be reproducible.
[plate_with_indent_a.zip](https://github.com/smartavionics/Cura/files/6490062/plate_with_indent_a.zip)

5axes, 2021-05-16:
But it's very difficult to reproduce, just moving the part position of 0.6 mm in this case can totaly change the result.
[image]

smartavionics, 2021-05-17:
Ah, yes, moving the model on the bed triggers this behaviour. I think what we're seeing here is due to an upstream change made a little while ago, I will look into this tomorrow.

smartavionics, 2021-05-18:
Hi, I've looked into this and now mostly understand the behaviour you are seeing.

1 - the skin lines are different on the top layer from, say layer 23, because the start point on those layers are different. This is because the "sharpest corner" algorithm is very sensitive to small changes in the shape of the polygons. Because the start points are different, the first skin line gets drawn in different directions on those two layers. Workaround is to not use the sharpest corner setting (I very rarely use it myself).

2 - the skin line drawing order changes when the part is moved on the bed because the skin lines are not anchored to the part  but remain static WRT the print bed. So moving the part changes not only the lengths of the individual skin lines but also their positions relative to the part outline. Probably no easy workaround for this behaviour as it is fundamental to how skin lines are created.

3 - as far as I can tell, the lines are being drawn in the expected order given the shape of the outline and holes. The odd looking order occurs because having finished a skin line it simply looks for the nearest line end to start from and when skin lines meet a wall at a shallow angle, the nearest line end point can belong to a line that isn't adjacent to the line that has just been drawn and so a gap appears which gets filled in later.

I don't know right now why the 4.9.1 release behaves differently but I will continue to think about it.