smartavionics/Cura #110: Random tiny gaps between inner and outer walls on 2 walls, no infill print

d-schmidt, 2021-05-16:
**Application version**
20210516

**Platform**
win 10 64

**Printer**
ender 3v2

**Reproduction steps**
0 infill
0 bottom
2 walls
moai fixed bottom
https://github.com/Ultimaker/Cura/issues/9811

**Screenshot(s)**
[image]

**Actual results**
there are tiny gaps between inner and outer wall which cura tries to fill

**Expected results**
constant inner wall offset without gaps

**Project file**

[moai-flat-fixed.zip](https://github.com/smartavionics/Cura/files/6490078/moai-flat-fixed.zip)

**Log file**

**Additional information**

5axes, 2021-05-16:
That's why we have the option "Filter Out Tiny Gap" ? or Even don't use the option Fill Gaps Between Walls ?
[image]

smartavionics, 2021-05-17:
Those are good recommendations to work around this issue.

d-schmidt, 2021-05-17:
@5axes  of course I can filter, but these gaps shouldn't be there in the first place. Are they the necessary side effect of another nice optimization? Imho the offset math for the inner wall should not produce gaps between the wall. This makes me doubt wall stability. At this place it is large enough for cura to fill, what about the rest of the wall?

5axes, 2021-05-17:
I think that  @smartavionics could give a better answer than me . But on my point of view yes it's a side effect of a better result concerning thin wall and gap filling in the Master release.   Arachne will perhaps change this statement with the UM Cura release but right now that's one of the main reason why I'm personnaly using the Master release instead of the UM Cura version.  Now if you want to also get a better accuracy on the wall geometry; using a smaller tolerance is also on other solution to supress this tiny gap .

[image]

d-schmidt, 2021-05-17:
This also happens in solid prints:
[image]

and changeing resolution doesn't help:
[image]

5axes, 2021-05-17:
In this case the most revelant parameter is the "Maximum deviation" ( deviation between the original geometry and the calculated   path) try 0.015

d-schmidt, 2021-05-17:
This is a solid model, so I don't see why thin walls code should apply here.  
Maximum deviation 0.015:  
[image]

smartavionics, 2021-05-18:
> @5axes of course I can filter, but these gaps shouldn't be there in the first place. Are they the necessary side effect of another nice optimization? Imho the offset math for the inner wall should not produce gaps between the walls. This makes me doubt wall stability. At this place it is large enough for cura to fill, what about the rest of the wall?

You're absolutely right that the gaps shouldn't be there in the first place. But they are and the easiest workaround is to filter the tiny gaps as has been suggested. My thin wall and gap filling implementation is not perfect but it generally works well enough for me and I am unwilling to put more time into it now as Arachne should make it completely redundant. Sorry about that.

d-schmidt, 2021-05-18:
Filter it is.
I'm still wondering if there really are tiny gaps in the walls or if the gap filling is just overzealous.

@smartavionics  You want to close this as "wont fix"?

5axes, 2021-05-18:
Hello Mark, 
I have a small question about the algorithm for creating the walls. I don't think that the spaces between the internal and external walls are abnormal. On the contrary, this seems rather logical and "desirable". The creation of a profile from a segmented model necessarily requires a filtering action of the points. One can think that the filtering applied on the inner walls can be more important because the quality of the contours does not need to be as high as on the outside. And a more important filtering decreases the size of the files and avoids jerkiness problems on the printers.  This difference in filtering can be the source of the gaps between the walls. But is this the case today in the Cura engine? Is there a different filtering applied on the internal and external walls?

smartavionics, 2021-05-18:
@d-schmidt , let's leave it open in case anyone else is interested in the discussion.

@5axes , the filtering is applied to the outline of each sliced layer and then the walls are derived from the filtered outline. The walls are generated by offsetting the outline inwards, i.e. the outer wall path is generated by shrinking the outline by 1/2 the width of the outer wall. Each inner wall path is generated by shrinking the previous wall path. The shrinking algorithm (which is provided by the clipper library that Cura uses) sometimes produces some small variations in the distance between the original path and an shrunk (or expanded) path. Thus, tiny gaps can appear. If the wall overlap compensation is enabled, other gaps can be created as that part of Cura is far from perfect.