smartavionics/Cura #111: [20210516] Speed preview is broken after slice

d-schmidt, 2021-05-16:
**Application version**
20210516

**Platform**
win 10 64

**Printer**
ender 3v2

**Reproduction steps**
slice with speed active
dis/enable any line type in speed drop down to get back correct view

**Screenshot(s)**
[image]

**Actual results**
10-0mm/s all blue

**Expected results**
colors

**Project file**
-

**Log file**
-

**Additional information**
-

5axes, 2021-05-16:
Same issue for me : 
**Application version** 20210516

**Platform** win 10 64

The maximum speed value is not correctly initialized after a calculation, an option must be changed to force the change.

[image]

smartavionics, 2021-05-17:
Yes, UM have made some changes in this area, I think this is an upstream bug.