smartavionics/Cura #113: hilbert curve top/bottom pattern

gz2k2, 2021-05-21:
hi,

I know that hilbert curves as top/bottom pattern result in an insane amount of printer movements, but the result just looks great. please implement it like slic3r does :)

smartavionics, 2021-05-21:
I can't imagine I would use it myself but as it's been a rainy day I've got something working...

[image]

It's still got some problems so I won't be releasing it just yet but it will appear in the future.

I don't use slic3r myself so perhaps you can answer a question for me. If you slice an object whose cross section varies with height (like a sphere or pyramid), does the hilbert infill remain the same pattern through all of the layers or does it alter in any way (i.e scale, shift or rotate)?

5axes, 2021-05-21:
On my point of view Hilbert infill in Slic3r or Prusa Slicer is just crap. It's not a real Hilbert trajectory (Ie : fill one layer starting from one point and finishing to an other point without retraction) but just a pattern with a lot a direction change and retraction so utility of this solution ????

[image]

5axes, 2021-05-21:
Same for the Top/Bottom pattern , it's base on a pattern, start from one corner, Switch to the opposite with a lot of direction change not very suitable for this technology.  Don't realy sea the benefits in this case . 
[image]

5axes, 2021-05-21:
From my point of view if you really want to occupy your rainy days the integration of decorative geometric patterns as on the example below would really be a solution that would interest many peoples (not me) but all those who make decorative boxes with their printers. Here the demand is strong and recurrent (just my point of view) 

[image]

smartavionics, 2021-05-22:
Try https://github.com/smartavionics/Cura/releases/tag/20210522

gz2k2, 2021-05-22:
Thank! I will try it! :)

smartavionics, 2021-05-23:
I can see how this could produce a decorative effect when using translucent filament...

[image]