smartavionics/Cura #114: 'Support Roof Top Layer Density' setting issue

kvnper, 2021-05-24:
When changing 'Support Roof Top Layer Density', to a value different from the default, the density becomes 100% for the support roof top layer.

Note: I'm also using a lower line width for support/support interface lines than the nozzle size (1mm)

3mf file attached
[FVTREX3_LEFT CALF Cover v3 logo.zip](https://github.com/smartavionics/Cura/files/6530486/FVTREX3_LEFT.CALF.Cover.v3.logo.zip)

[image]

[image]

smartavionics, 2021-05-24:
Hi, yes, I've verified the bug, sorry about that. Will be fixed in my next release. In the meantime, the workaround is not to use a different density for the top layer of the support roof.

smartavionics, 2021-05-24:
What system are you using? I can make a build for that right now.

kvnper, 2021-05-24:
Windows 10, no rush, I don't need it right now

kvnper, 2021-05-24:
Also no need to apologise, thank you for your work

smartavionics, 2021-05-24:
I'll do it anyway as I have a couple of minor fixes for the new hilbert pattern. It will appear as https://github.com/smartavionics/Cura/releases/download/20210522/Cura-mb-master-win32-20210522-1.exe when it has been built this morning.