smartavionics/Cura #115: Missing UI Icons

randellhodges, 2021-06-15:
**Application version**
20210608

**Platform**
Windows 10 20H2

**Reproduction steps**
1. Run program

**Screenshot(s)**
[image]

[image]

**Expected results**
Icons should appear

**Log file**
[cura.log](https://github.com/smartavionics/Cura/files/6652086/cura.log)
[stderr.log](https://github.com/smartavionics/Cura/files/6652092/stderr.log)

**Additional information**
Cura 4.9 and 4.9.1 works on the same machine.

smartavionics, 2021-06-15:
Hi, this is caused by an incompatibility between this release and the last release. The UM people used a new set of icons in the last release and then in this release they have reverted to the original icons and somehow Cura is getting confused. A workaround is to remove the Cura installation and then reinstall the latest release.

5axes, 2021-06-15:
Yes same issue for me: 
Solution remove the previous Cura Master version ( Just delete the installation directory ) and install the last release and it's ok

randellhodges, 2021-06-15:
> Hi, this is caused by an incompatibility between this release and the last release. The UM people used a new set of icons in the last release and then in this release they have reverted to the [...]

@smartavionics  This is a first time installed your version of Cura on this machine. I do have also have a 4.9.1 install on the computer, in a different directory of course.

@5axes You hit it on the head with the Ultimaker Dark theme. I changed to the Ultimaker theme, restarted and all the icons appeared. I changed back to Ultimaker Dark them, restarted and verified they no longer show again.

smartavionics, 2021-06-15:
Do the icons appear when you use the dark theme with 4.9.1?

randellhodges, 2021-06-15:
> Do the icons appear when you use the dark theme with 4.9.1?

They do

smartavionics, 2021-06-17:
Maybe you need to clear the qml cache? On Linux it is stored in ~/.cache/Ultimaker\ B.V./cura/qmlcache, Windows will have a similar folder.

5axes, 2021-06-17:
Doesn't change anything on Windows : C:\Users\XXX\AppData\Local\Ultimaker B.V\Cura\cache\qmlcache

5axes, 2021-06-17:
The work around to "solved" this issue is to duplicate the icons folder of the **cura-light** Theme into the **cura-dark** theme

[image]

smartavionics, 2021-06-17:
If you have 4.9.1 installed, what files are in the resources/themes/cura-dark/icons folder that you will find in the installed tree?

I just see sign_in_to_cloud.svg and nothing else whereas the light theme has 100 or so icons in it.

5axes, 2021-06-17:
Yes same for me .. Normaly You don't need to re-copy the icons in the folder icons of a new theme Cura should use the "default" icons but in the case of your master release it's not the case. I don't know why.

smartavionics, 2021-06-17:
Thanks, I don't know why either and I haven't made any changes to the themes part of Cura. There have been upstream changes over the last month or two which they reverted but, obviously, things have not gone back to how they were previously. So for the moment, you will have to use the workaround suggested above. It will be interesting to see if the same problem arises when they release 4.10.

smartavionics, 2021-06-17:
@5axes, can you see any differences between the themes trees in 4.9.1 compared to my master?

5axes, 2021-06-17:
I searched a bit but found nothing .... In the themes we still have the "inherits": "cura-light" so we should not need to redefine the icons. Sorry no more info I think we will have to wait for 4.10

```
    "metadata": {
        "name": "Ultimaker Dark",
        "inherits": "cura-light"
    },
```

smartavionics, 2021-06-17:
There are some differences in the Python code between 4.9.1 and the current upstream master that are concerned with the location of icon files so there's every possibility that the UM folks have broken something. I suggest that when the 4.10 beta is released that you test that and if this issue is present then someone opens a new issue on UM's repo so that they can investigate.

5axes, 2021-06-18:
So Cura 4.10-Beta every thing is normal : 
[image]

smartavionics, 2021-06-19:
Yes, they must have noticed it was broken and made a change just before 4.10-beta was released. I'm building a release at the moment....

5axes, 2021-06-20:
Solved for me with your last release

randellhodges, 2021-07-30:
All good

closed 2021-07-30 by randellhodges