smartavionics/Cura #116: cx_Freeze: Python error on win64 version

jmcazaban, 2021-07-14:
Hi Mark, when I install your latest win64 release I get this as soon as I try to run the program:

[image]

The win32 release works fine.
I'm on W10 (20H2) 64 bit version.
Any hint?

Thanks!

smartavionics, 2021-07-15:
I will investigate, can you translate the ImportError message for me please?

smartavionics, 2021-07-15:
Can you please provide the cura.log file? Thanks.

jmcazaban, 2021-07-16:
Hi, 
I tried installing in another machine and I get the same error.
The program installs perfectly without errors, but when I try to run it just crashes instantly and shows the error.
The cura.log isn`t even generated, 
This time it is in english:

[image]

jmcazaban, 2021-07-16:
In the same computers, the Ultimaker 4.10.0 release is working fine.
Also your win32 version works fine.

Looks like it´s looking for something in C:\Users\burto ... which I believe is your user
I don´t even have such directory created with my user.

Thanks!

smartavionics, 2021-07-17:
Could you please look in where Cura is installed (c:\program files\ultimaker cura master) and there should be in the lib directory a file Savitar.pyd and the length should be 176128.

jmcazaban, 2021-07-17:
Yes it is and the size is correct.
I tried also replacing it with Savitar.pyd from my working UM 4.10.0 installation, same results.

When I rename or delete the file the error changes to module not found:

[image]

smartavionics, 2021-07-17:
I still don't know why this problem occurs but I have updated my build process a little and made a new win64 build, it's https://github.com/smartavionics/Cura/releases/download/20210629/Cura-mb-master-win64-20210629-1.exe

jmcazaban, 2021-07-17:
Mark thanks for the new build, but I get the same result as the original.
Anyway, as the 32bit version works fine for me I don't want to keep wasting your time on this.

Also I wanted to thank you for your work, because your version allowed me to start using Cura.
The way it handles internal thin walls is great and very important for the things I print, similar to the single extrusions of S3D, instead of the crazy micro zigzag of UM release.
Also the features and options you add allows me to get noticeably better prints.

I started recommending Cura when I found your version, and I only recommend SmartAvionics release.

Thanks!!

smartavionics, 2021-07-18:
I'm sorry it's still not right but it sounds like you can survive. I don't actually use Windows myself so my knowledge of the quirks of the system is limited. Have you used any earlier builds from me? If so, which was the last 64 bit build that worked OK.

jmcazaban, 2021-07-19:
Mark, this 64 bit release works fine:

https://github.com/smartavionics/Cura/releases/tag/20201121 

I was on google searching about spiralized mode in Cura because I was a geting a noticeable seam (in the preview).
I never had a problem with this mode on S3D, so I was surprised it wasn't working.

Someone suggested your release 20201121, and as you were helping me with the 64 bit issue, I tried installing it and it worked fine.
The vase mode still looked bad, but then in another thread you said that the renderer in Cura couldn't show spiralized mode very well.
So I checked the file sliced with Cura on SD3 and it looked fine, you were right!!

Hope it helps!

jmcazaban, 2021-11-30:
Hi,

I solved the problem by installing the latest Microsoft Visual C++ 2015-2019 Redistributable

Maybe it helps with other Pyton cx_Freeze errors.

closed 2021-11-30 by jmcazaban