smartavionics/Cura #117: Perrformance issue since Cura 4.9 in remote access (xrsh)

Bi-Doux, 2021-07-22:
**Application version**
Cura 4.9 and followings

**Platform**
Arm64, Raspberry Pi running RaspiOS 64bits

**Printer**
Anycubic Mega Zero but issue is independent from chosen printer

Hello,

I don' know what has been changed since Cura 4.9 but I experience very slow MMI when running Cura from a remote access.
It was not the case with Cura 4.8 therefore I suspect that some change introduced in graphical aspects of MMI with Cura 4.9 is rather incompatible with remote access or has poor performance on headless platforms (I am curious to learn whether other users running also in remote access but other flavors e.g. Windows x86, Ubuntu... experience same issue).
When I write very slow, it is almost unusable: some 30 seconds just to start the program and see the parameters submenus. Several tenths seconds also to deploy each parameter submenus (I configure "see all" in parameters) and any interaction (rotations, moves, changes in camera position...etc) with the loaded object on plate takes also ages and is not practicable.
To clarify why it is an issue, I can say that 4.8 had not such problem and I was really able to use Cura in my headless configuration with previous versions.
Additionally, it works also as in the past when I exceptionally connect directly a monitor to my Raspberry.
So, not sure whether the solution is in Cura or in some library used by Cura from version 4.9 but I would appreciate any hint to restore previous performances because Cura is the only application running on my RPI which experiences this drop of performandes when I am running it in a remote access session (note: I am not using Vnc but xrsh sessions sometimes from Windows remote access control app).

As I am not familiar with Cura architecture, please don't hesitate to point this issue to the right repository: it is not a Cura Engine issue but most probably something with its graphical interfaces.

Thanks in advance for any hint (tweaking of MMI libraries to better support xrsh).

Cheers

Laurent

smartavionics, 2021-07-22:
Hi, I develop using 64 and 32 bit headless Pi-4 systems and get good UI performance using VNC. I don't use xrsh. Startup time is still quite slow but once the UI has loaded, all other interactions are acceptably fast. Can you use VNC?

Cura is based on Python, Qt and PyQt plus many other libraries and there was a big change after 4.8 to make it use more recent versions so I guess that is what has caused the change you are seeing.

Bi-Doux, 2021-07-22:
Hi,
I will have a try with Vnc (I need to activate it in my Pi and install it on the PC) just to see whether the performance drop is still there. When you mean start-up is slow, you just refer to the initialisation phase (before any menu or workspace is displayed) ? This initialisation is slow for me on any cases (even with the Windows version) but once loaded, UI is very fluent on direct displays unlike remote accesses. I will check on respective forums whether this is a known regressions of the latest versions of those libraries.

Thanks for all

BR
Bi-Doux

Bi-Doux, 2022-02-28:
Suspicion of an issue from QT5 is not confirmed. I have installed today another application using Qt5 and it works very fluently in remote access. Giving up xrdp, I have even switched recently to another mean to access my headless RPI4: I uses a X-client and X11 SSH forwarding but it is still very slow unlike all other applications that I run that way remotely.