smartavionics/Cura #123: [20210629] Weird bumps in concentric circles

m-sundman, 2021-08-21:
**Application version**
Cura-mb-master-x86_64-20210629.AppImage

**Platform**
Linux/kubuntu

**Printer**
Artillery Genius

**Reproduction steps**
1. Slice something with edges that are concentric circles.
2. Use concentric pattern.

**Screenshot(s)**
Layer 1 is horrible:
[image]

Layer 2 is fine:
[image]

Layer 5 has some bumps too:
[image]

Even the top surface layer has small bumps:
[image]

**Actual results**
There are random bumps in the pattern. The bumps are usually bigger on the first layer, and the location of the bumps seem to vary if you adjust the line width.

**Expected results**
Concentric circles should be concentric circles without any random bumps disrupting the pattern.

**Project file**
[Wheel.3mf](https://github.com/smartavionics/Cura/files/7025933/Wheel.zip)

**Log file**
[cura.log](https://github.com/smartavionics/Cura/files/7025942/cura.log)

**Additional information**
Seems to be present in upstream, too, although perhaps to a lesser extent.

smartavionics, 2021-08-22:
It's not perfect but increasing max deviation to 0.25 and setting fill gaps between walls to everywhere helps a lot...

[image]

smartavionics, 2021-08-22:
In fact increasing max deviation to 0.05 is sufficient.

m-sundman, 2021-08-23:
Still bumpy here, although only on the first layer:
[image]

Project:
[Bumpy.3mf](https://github.com/smartavionics/Cura/files/7028734/Bumpy.zip)

smartavionics, 2021-08-23:
I just fiddled with the resolution and deviation values. Again, not perfect but better. This model is not ideal for concentric skin or lots of walls because the hole isn't central or round.

[image]

m-sundman, 2021-08-31:
Why is it always so much worse in the first layer? Like, if I could get the same concentric algorithm for the first layer as it uses for the second layer then in most cases that would be enough.

smartavionics, 2021-08-31:
Good question, I don't know why the first layer is worse than the 2nd as the line widths are the same.