smartavionics/Cura #124: Adaptive layers don't respect the Top/Bottom Thickness

m-sundman, 2021-08-28:
**Application version**
Cura-mb-master-x86_64-20210629.AppImage

**Platform**
Linux/kubuntu

**Printer**
Artillery Genius

**Reproduction steps**
1. Enable Use Adaptive Layers
2. Use a model with an edge angle high enough to change the layer thickness near the top/bottom
3. Set Top/Bottom Thickness (i.e., Top Layers and Bottom Layers are calculated values).

**Screenshot(s)**
N/A

**Actual results**
The top/bottom thicknesses will be the number of layers in the Top Layers and Bottom Layers fields.

**Expected results**
The top/bottom thicknesses should be the number of layers that will make the specified top/bottom layer thicknesses.

**Project file**
E.g., a sphere.

**Log file**
N/A

**Additional information**
When specifying a thickness (rather than a number of layers), the actual number of top/bottom layers to use should be calculated **after** cura calculates the thicknesses of each layer.

smartavionics, 2021-08-28:
Hi, I thought my build already did what you are requesting but looking at the code shows that the work was not completed or it has been reverted somewhere along the line. I will get it working again for the next release.

timperlabs, 2022-04-09:
I stumbled here from  [this thread](https://community.ultimaker.com/topic/22998-top-thickness-not-respected-when-using-adaptive-layers/page/2/#comments)  after failing to print a sphere with adaptive layers during new printer calibration.  I am using Cura 4.13.    Was this fix (or the new variant) ever PR-ed back to Cura itself?

smartavionics, 2022-04-09:
It is unlikely that it was incorporated into UM Cura.