smartavionics/Cura #125: angled overhangs not detected?

halfshavedyaks, 2021-09-05:
**Application version**
Cura-mb-master-x86_64-20210629.AppImage

**Platform**
linux Mint 20

**Printer**
modified CR10S

**Reproduction steps**
set overhang speed (70%) and overhang angle (20deg) and slice on a model with overhangs at various XY angles

**Screenshot(s)**
[overhang-proj&screen.zip](https://github.com/smartavionics/Cura/files/7111430/overhang-proj.screen.zip)

**Actual results**
angled overhang isn't detected

**Expected results**
print speed should slow down on angled overhang same as on orthogonal overhang

**Project file**
included in above zip file

**Log file**
is the log needed? I'll find it if requested

halfshavedyaks, 2021-09-05:
[image]
added image separately for convenince

smartavionics, 2021-09-05:
Hmm, that's interesting, I'll investigate...

smartavionics, 2021-09-05:
I've come up with a fix and will test it some more before my next release. Thanks for posting the issue!

[image]

halfshavedyaks, 2021-09-05:
that was fast - that's great!

is there a way I can test it without having to build cura myself?

smartavionics, 2021-09-05:
I would need to build a new AppImage for you to test it and I will be doing that when the upstream 4.11 release arrives which should be coming along in a few days I guess.

halfshavedyaks, 2021-09-05:
sounds great! Will this fix apply to the overhang fan settings too?

smartavionics, 2021-09-05:
Yes, the problem was that it wasn't recognising those lines as being overhung at all but now it will be so both the speed and the fan will be altered for them.

halfshavedyaks, 2021-09-06:
I've been testing and I see that the existing release does work on most overhangs, but has trouble with some. I found another case with weird behaviour in the current release that might be worth checking if your fix covers it.

[image]

[overhangborkage.zip](https://github.com/smartavionics/Cura/files/7114661/overhangborkage.zip)

incidentally and OT that project also seems to show an apparent bug where 3 skirt lines are printed even though only one is set, but I'm very new to cura so maybe I missed something.

halfshavedyaks, 2021-09-06:
I've been looking at the g-code files in prusa g-code viewer, and the fan speed increases are not happening for the same lines that the print speed decreases are, they are being triggered separately, and they are wrong in different ways from each other.

fan speed is being increased for adjacent lines and previous/subsequent lines where it should not be.

I'll post screenshots and files when I get time. but this is just a alert if debugging this feature to check both fan and print speed since they are not reacting the same way to overhangs.

smartavionics, 2021-09-06:
I'm not at home now to check but remember that the fan can't speed up instantly so the speed increase will be commanded some distance before the overhang is printed.Mark

halfshavedyaks, 2021-09-06:
yes you are right, as per my edit, sorry if you didn't see that before replying.

The fan will turn on early, but it can't turn on mid line (I presume) so it will turn on at the start of the line that is closest to the overhang plus the spinup time... so the amount of the print that looks like it has faster fan varies depending on the line length. It was bit confusing to see until I realised what was happening.

halfshavedyaks, 2021-09-07:
On closer examination I'm still seeing some strange fan behaviour that isn't just the spinup time. Here's a project that shows odd fan behaviour on the bridge, the fan stops early on the bridge (is there spindown time too?) and it also doesn't spinup in advance for some of the perimeter bridge sections.

There's also a small area where overhangs are not detected properly, and some fan issues there too, although maybe that overhang area will be fixed with your fix.

[image]

[oh&bridgetest.zip](https://github.com/smartavionics/Cura/files/7122785/oh.bridgetest.zip)

smartavionics, 2021-09-07:
> On closer examination I'm still seeing some strange fan behaviour that isn't just the spinup time. Here's a project that shows odd fan behaviour on the bridge, the fan stops early on the bridge 

> (is there spindown time too?) 

yes

> and it also doesn't spinup in advance for some of the perimeter bridge sections.

That's true.

> There's also a small area where overhangs are not detected properly, and some fan issues there too, although maybe that overhang area will be fixed with your fix.

yes

halfshavedyaks, 2021-09-08:
I'm not seeing spindown applied to perimeters anywhere. I'm not sure it bothers me but it seems inconsistent.

also I think spinup length (which is always whole lines) may be being calculated using the overhang printing speed, but since it happens before the overhang begins, it should actually be calculated on the speed of the preceding perimeter(s)

my cooling fan won't start from stopped if the pwm amount is below about 40%. I suggest adding a feature like in s3d, that blips the fan to 100% briefly when starting from stopped. S3D pauses for 500ms to do this, which i think is excessive, I suspect a 100ms shorter pause would do, or better still to do it while still printing.

halfshavedyaks, 2021-09-10:
I've been using the new release and it is much better thanks!

There are still some small areas where overhang is not detected correctly, but they are ignorable, I haven't found anything likely to cause a significant problem with the print yet.