smartavionics/Cura #127: Add XY Homing per layer, to prevent layer shifts

daT4v1s, 2021-09-05:
**Is your feature request related to a problem? Please describe.**
in certain prints, I occasionally get layer shifts, thus wasted material, (and looks bad.)
If it would home the x/y esp if there is spare time, extra travel moves are better than loosing a print and the time lost and to re-setup/debug and redo.

- it seems to be related to a print that has lots intermediate travel moves (many sub-parts >15),
 that includes a lot of movement (per-layer)

**Describe the solution you'd like**

add homing every (either or weighted average) of various factors:
- layer
- sub-part of layer (e.g.  stl-part, line-type..)
- after certain mm of filament extruded
- after certain number of travel moves
- after cumulative acceleration or jerk that the print experienced, esp those adjacent to travel(faster) moves

homing moves are slow, thus it should move to a small priming-tower to retract&wipe, thus less stringing, near the homing spot (different for different printer models/types)
if UI it could allow printing the homing speed, and location into Cura, to keep the time-estimation of the print accurate, these challenges may help stress/test the code for improvement

(it would maybe help) if it's homing frequency/sensitivity could be locally modified 
e.g. [modify mesh by parts], or a [similar object that only edits the z-height] (since the former is limited, in the settings it can change, and inserts a gash/wall at the boundary)

would it be possible to (based on/via timing, and if using Usb interface) to detect if a layer-shift occurred?

a single mis-fired layer (or even better sub-layer) is easier to sand-down or repair, ...

- maybe tuning some settings, but i don't know exactly which ones would not result in side-effects

**Describe alternatives you've considered**
- monitoring it real-time and aborting and modding the g-code to restart at layer, not-human friendly

- lowering `jerk` values, but lower would make the print blobby
- updating the printer firmware to support linear-advance and s-spline (WIP) , to allow lower jerk values

- using insert g-code (plugin) at z-layer-change, for homing, but cura doesn't like that when re-opening the program (complains about corrupt profile settings, and removing the insert@ z-layer fixes this profile-warning consistently) , something built-in may be better

- dismissals pointing to printer, may not be completely optimal, there may be smarter/modern software workarounds, the main printer isn't always available

**Affected users and/or printers**
people who get layer-shifts, printers whose motors aren't closed-loop

**Additional context**
different from what i print usually, it uses `Random` `Z-seam alignment`
previously, `random infill start` for a different caused no issues

different from typical prints, when i abort it takes a while for it to stop, (waiting buffer more full, thus less g-code (density) amount per time)
but this time, it almost immediately stops, indicating there is little spare buffer waiting, it may be CPU-bottle-necked, almost all the walls are curved, but the tolerances are relatively tight & close-fitting

smartavionics, 2021-09-06:
Hi, sorry to hear you are having mechanical problems with your printer. However, I don't believe that the proposed solution (add XY homing between chosen layers) is the best way to solve the issue. You need to either fix the mechanical problem or use slower speeds, lower acceleration and jerk, etc. You don't say what printer you have but I have two cheap Sovol printers and they will print for many hours and never suffer from XY shifts. Perhaps that's because I don't try to drive them too hard. What speeds and accelerations are you using?

daT4v1s, 2021-09-06:
I appreciate the reply, solving the root cause if the root cause if the shifts is more important than those workarounds.

typical printing speeds: 8.4mm/s to 27mm/s , tested up to 70mm/s
travel speed 70mm/s
acceleration: 500mm/s² (print), 800mm/s²(travel), tested up to 1200mm/s²
jerk threshold: 6.5mm/s (print), 6.7mm/s (travel), 4.8mm/s (intial layer), tested up to 15mm/s, quality goes down (blobbier) when below jerk of approx 6.5 mm/s
z-hop: 0.3mm (since undecided)

retraction distance: 7.5mm
retraction: 10.5mm/s

gyroid gradient infill + vertical multiplier
[image]

I'm wondering if gyroid infill is subjecting the printer to too much wear, 
(or  recent g-code buffer shortage pauses, or occasional power outages)

The printer is 
1.75mm filament, 0.4mm nozzle, direct-drive, with saguino-melzi-derivative board, modified with TL-smoothers, printing with PLA
with gcode to increase jerk values to 6.5mm/s from 2.0mm/s, and max speed from 40mm/s

the motors are 24mm, 2.5W, 12V, and typically run at 60-70°C (infrared estimate)

I'm unsure if lowering some of the speeds would introduce stringing and by how much.

Current Cura Printer settings (set via gcode)
[image]

smartavionics, 2021-09-08:
> the motors are 24mm, 2.5W, 12V, and typically run at 60-70°C (infrared estimate)

I'm not a stepper motor expert but I am wondering whether your steppers are losing torque at that temperature such that they occasionally they miss steps. 70 deg isn't so hot but perhaps they are running at the torque limit?

bartwaw, 2021-09-24:
For me that shifting in layers looks like motors are loosing steps. Also higher microstepping will introduce smaller torque.
If i get such problems i will start inspecting printer. 
Loose all belts and check that all axes are moving freely. 24mm motors are quite small and they have small torque. Such motors (nema17 20-22mm) are generally dedicated for extruders not for moving gantry.