smartavionics/Cura #129: What exactly is the obstacle to BigSur compatibility?

halfshavedyaks, 2021-09-06:
**Is your feature request related to a problem? Please describe.**
Although I am mostly using linux for maker stuff atm, I would also like to be able to run Cura on my M1 mac

**Describe the solution you'd like**
a universal binary for apple silicon and BigSur

**Describe alternatives you've considered**
using linux instead

**Affected users and/or printers**
owners of recent macs

**Additional context**
Really this is just a question, what is the actual problem preventing a BigSur/Universal build, and can I help? 

I have a M1 mac with Xcode installed, and although my knowledge of compiling things is very limited I have successfully compiled my own C++ app as a universal binary.

halfshavedyaks, 2021-09-07:
I have a M1 mac mini, and the latest Xcode. but I don't know anything about building python apps. I'd be willing to try building it if you (or anyone) can provide some guidance.

halfshavedyaks, 2021-09-07:
hmm, I looked up how to build cura, and they really make you jump through hoops. seems like you have to install cmake AND Xcode AND git AND svn and maybe brew as well before you can even start compiling dependencies...

so I'm thinking I don't want a mac version all that much...

werdnum, 2021-11-21:
Can you explain for the record what the obstacle is to building for MacOS, and what upstream change broke it? It would be helpful to know if it's surmountable with some effort.

smartavionics, 2021-11-22:
Hi @werdnum , I think the problem boiled down to the simple fact that the MacBook I was building my releases on is ancient and so the versions of compilers/libraries that I was using wasn't compatible with what was needed for more recent MacOS releases and didn't support the changes that UM made with regard to PyQt version, etc.

werdnum, 2021-11-23:
@smartavionics gotcha, do you have a script or something that you use to build it? I wonder how hard it would be for me to try building it on my own laptop.

smartavionics, 2021-11-23:
Hmm, well you can have a go but it's a bit of a slog.

You need to clone a couple of my repos (https://github.com/smartavionics/cura-build-environment) and (https://github.com/smartavionics/cura-build).

In both of those repos you need to check out the mb-master branch.

First you must build the environment by making a directory called build and then within that directory do ../run-cmake.sh which will setup some enviroment variables and call cmake to produce the make files. It assumes that you already have installed the standard development tools (compilers, SDKs, etc). Once you have successfully generated the makefiles simply type make to compile and install all the stuff that cura is based on. This will take quite a long time!

Once the environment has been built, you can do the same in the cura-build repo, mkdir build, cd build, ../run-cmake.sh, make. When that has finished you are left with a folder containing the application and its resources. I have a script that makes a dmg file from that folder, I'll find that script and post it here.

In the run-cmake.sh files there are hard-coded numbers for the target osx  version and sdk versions, you may need to change those.

Good luck!

werdnum, 2021-11-23:
Thanks! I'm going to take a look into something like nixos.org for setting up a consistent build environment, because builds that can't be done in one step make me sad.

xerootg, 2022-02-17:
My personal slicing machine is a 2011 iMac, so I'm out to dry with recent changes to Cura. I hate to 'get a new machine' just cause cura isn't compiling for older versions of macos. Is there any way I can help resurrect the macos build?