smartavionics/Cura #132: Gradual infill step height Broken

MakManIfest, 2021-09-12:
**Application version**
20210908

**Platform**
Windows 10 Pro

**Printer**
Custom - Voxelab Aquila

**Reproduction steps**
Using Gradual Infill Steps and Gradual Infill Step Height

**Screenshot(s)**
[image]
[image]
[image]<br /><a target='_blank' href='https://whatsmyscreenresolution.com/'>popular desktop screen resolutions</a><br />

**Actual results**
Cura increases the infill according to the parameters for a single layer then reverts to lower infill for a number of layers that seems to be tied to the Gradual Infill Step Height some how. I have tried different Gradual Infill Step Heights but this increases or decreases the number of lower infill layers between the single denser infill layer.

**Expected results**
In Cura 4.11 vanilla, the infill is varied according to the parameters Gradual Infill Steps at the Gradual Infill Step Height relative to the 1st top layer. This Is what I would expect to also happen in 20210908.

**Project file**
I cant share the fang I'm trying to slice due to it's license but there is an earlier version on thingyverse that I have tinkered to reproduce the same issue.
3fm - https://mega.nz/file/TDIAWJoR#WZZILfvy0-JPypTy2WoKtSaRf3kb0c3ePShTkoCBiIo
STL - https://mega.nz/file/uGYykZ7S#HzXkZsNPNUMN7-AZsOrb91H7tVkiX9oWTBOkMrdKFYs

**Log file**
Cura.log - https://mega.nz/file/DSRynZgT#kZ0lNyK3EyyubbKx84ZtlkOFZPXO3427BR--7Sxif9Y

**Additional information**
I need the Fang to be lite so I'm using low infill with 2 walls and I'm printing in PETG so I need the infill density to increase near the top layer so the top layers form properly without holes.
Previously I have been slicing in S3d and use multiple processes to vary the infill density and pattern around 3 layers below the 1st top layer until either the top has formed or the geometry of the object allows me to reduce it again without the risk of holes forming.
Looking back into Cura as it has some cool features and S3d seems to stalled.

MakManIfest, 2021-09-12:
I would like to add that slicing a simple cube with these settings seems to produce the expected result.

smartavionics, 2021-09-12:
Thanks for the report, I will investigate.

smartavionics, 2021-09-14:
Hi, I have fixed a problem with the skin support feature not working in conjunction with the gradual infill feature but  I have noticed that the honeycomb pattern doesn't work well with gradual infill as it generates multiple lines lying on top of each other.