smartavionics/Cura #135: Unable to set up custom FDM Printer

bigbeau, 2021-09-29:
**Application version**
20210908

**Platform**
Windows 10

**Printer**
That's the problem, I can't

**Reproduction steps**
1. Freshly install the version above.
2. Try to add a custom printer via the Add Printer utility.
3. Notice that there is no option to add a custom, non-networked printer.
4. Investigate the Custom definition json in the definitions folder in the install directory.
5. Notice it's set to visible in the json, but still doesn't appear after multiple restarts.

**Screenshot(s)**
[image]
[image]

**Actual results**
The option to set up a custom printer is not there

**Expected results**
I expect to be able to set up my printer lmao

**Log file**
Copied parts that mentioned the fdmprinter definition
```line 187: 2021-09-29 10:41:35,835 - DEBUG - [JobQueueWorker [0]] UM.Mesh.MeshData.calculateNormalsFromVertices [558]: Calculating normals took 0.0 seconds
line 188: 2021-09-29 10:41:35,835 - WARNING - [MainThread] LocalContainerProvider.LocalContainerProvider._saveCachedDefinition [295]: The definition cache for definition fdmprinter failed to pickle.
line 189: 2021-09-29 10:41:35,851 - DEBUG - [JobQueueWorker [0]] UM.Mesh.MeshData.approximateConvexHull [505]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.004000425338745117 seconds. 8 input vertices. 8 output vertices.

line 473: 2021-09-29 11:39:06,281 - DEBUG - [JobQueueWorker [0]] UM.Mesh.MeshData.calculateNormalsFromVertices [558]: Calculating normals took 0.0010008811950683594 seconds
line 474: 2021-09-29 11:39:06,426 - WARNING - [MainThread] LocalContainerProvider.LocalContainerProvider._saveCachedDefinition [295]: The definition cache for definition fdmprinter failed to pickle.
line 475: 2021-09-29 11:39:06,440 - DEBUG - [JobQueueWorker [0]] UM.Mesh.MeshData.approximateConvexHull [505]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.05299782752990723 seconds. 8 input vertices. 8 output vertices.

line 753: 2021-09-29 12:03:13,181 - INFO - [MainThread] cura.CuraApplication.run [811]: Initializing container manager
line 754: 2021-09-29 12:03:13,435 - WARNING - [MainThread] LocalContainerProvider.LocalContainerProvider._saveCachedDefinition [295]: The definition cache for definition fdmprinter failed to pickle.
line 755: 2021-09-29 12:03:13,441 - DEBUG - [JobQueueWorker [0]] UM.Mesh.MeshData.calculateNormalsFromVertices [558]: Calculating normals took 0.0 seconds

line 1195: 2021-09-29 12:13:33,322 - DEBUG - [JobQueueWorker [0]] UM.Mesh.MeshData.calculateNormalsFromVertices [558]: Calculating normals took 0.019001007080078125 seconds
line 1196: 2021-09-29 12:13:33,326 - WARNING - [MainThread] LocalContainerProvider.LocalContainerProvider._saveCachedDefinition [295]: The definition cache for definition fdmprinter failed to pickle.
line 1197: 2021-09-29 12:13:33,333 - DEBUG - [JobQueueWorker [0]] UM.Mesh.MeshData.approximateConvexHull [505]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.006000041961669922 seconds. 8 input vertices. 8 output vertices.

line 1487: 2021-09-29 12:23:41,798 - INFO - [MainThread] cura.CuraApplication.run [811]: Initializing container manager
line 1488: 2021-09-29 12:23:42,054 - WARNING - [MainThread] LocalContainerProvider.LocalContainerProvider._saveCachedDefinition [295]: The definition cache for definition fdmprinter failed to pickle.
line 1489: 2021-09-29 12:23:42,073 - DEBUG - [JobQueueWorker [0]] UM.Mesh.MeshData.calculateNormalsFromVertices [558]: Calculating normals took 0.0 seconds

lines 1599-1628
2021-09-29 12:23:59,326 - DEBUG - [MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.009519815444946289 seconds
2021-09-29 12:24:14,117 - INFO - [MainThread] cura.Settings.MachineManager.addMachine [397]: Trying to add a machine with the definition id [fdmprinter]
2021-09-29 12:24:14,203 - DEBUG - [MainThread] cura.Machines.ContainerTree.__getitem__ [120]: Adding container tree for fdmprinter took 0.08200645446777344 seconds.
2021-09-29 12:24:14,221 - WARNING - [MainThread] cura.Settings.CuraStackBuilder.createExtruderStackWithDefaultSetup [115]: Could not find preferred nozzle . Falling back to empty.
2021-09-29 12:24:14,241 - INFO - [MainThread] cura.Settings.GlobalStack.addExtruder [202]: Extruder[fdmextruder #2] added to [FDM Printer Base Description] at position [0]
2021-09-29 12:24:14,244 - INFO - [MainThread] cura.UI.MachineActionManager.addDefaultMachineActions [74]: Default machine actions added for machine definition [fdmprinter]
2021-09-29 12:24:14,250 - WARNING - [MainThread] cura.Settings.GlobalStack.addExtruder [197]: Extruder [fdmextruder #2] has already been added to this stack [FDM Printer Base Description]
2021-09-29 12:24:14,255 - INFO - [MainThread] UM.Logger.info [129]: Initializing Active Machine...
2021-09-29 12:24:14,261 - WARNING - [MainThread] UM.OutputDevice.OutputDeviceManager.removeOutputDevice [235]: Could not find output device with id moonraker-upload to remove
2021-09-29 12:24:14,262 - DEBUG - [MainThread] MoonrakerPlugin.MoonrakerPlugin._checkMoonrakerOutputDevices [34]: Moonraker is not available for printer: id:FDM Printer Base Description, name:FDM Printer Base Description
2021-09-29 12:24:14,305 - DEBUG - [MainThread] cura.Machines.Models.NozzleModel._update [28]: Updating NozzleModel.
2021-09-29 12:24:14,438 - DEBUG - [MainThread] cura.Machines.Models.NozzleModel._update [28]: Updating NozzleModel.
2021-09-29 12:24:14,448 - DEBUG - [MainThread] cura.Machines.Models.NozzleModel._update [28]: Updating NozzleModel.
2021-09-29 12:24:14,453 - DEBUG - [MainThread] cura.Machines.Models.NozzleModel._update [28]: Updating NozzleModel.
2021-09-29 12:24:14,482 - DEBUG - [MainThread] cura.UI.WelcomePagesModel.goToNextPage [134]: Page [machine_actions] should not be displayed, look for the next page.
2021-09-29 12:24:14,598 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [155]: New error check scheduled.
2021-09-29 12:24:14,891 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [62]: Updating QualityProfilesDropDownMenuModel.
2021-09-29 12:24:14,898 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [33]: Updating CustomQualityProfilesDropDownMenuModel.
2021-09-29 12:24:15,075 - INFO - [MainThread] FirmwareUpdateChecker.FirmwareUpdateChecker.checkFirmwareVersion [79]: No machine with name FDM Printer Base Description in list of firmware to check.
2021-09-29 12:24:15,127 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [62]: Updating QualityProfilesDropDownMenuModel.
2021-09-29 12:24:15,133 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [33]: Updating CustomQualityProfilesDropDownMenuModel.
2021-09-29 12:24:15,902 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [214]: Error check finished, result = False, time = 1.3s
2021-09-29 12:24:16,410 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend._createSnapshot [269]: Creating thumbnail image (just before slice)...
2021-09-29 12:24:16,416 - DEBUG - [MainThread] UM.Scene.Camera._preferencesChanged [255]: Changing perspective mode to perspective.
2021-09-29 12:24:16,417 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [284]: Starting to slice...
2021-09-29 12:24:16,417 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [303]: Going to slice build plate [0]!
2021-09-29 12:24:16,417 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [311]: Build plate 0 has no objects to be sliced, skipping
2021-09-29 12:24:19,156 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [421]: file:///C:/Program Files/Ultimaker Cura master/plugins/MachineSettingsAction/MachineSettingsAction.qml:81:27: QML MachineSettingsExtruderTab: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
2021-09-29 12:24:25,154 - DEBUG - [MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.03889155387878418 seconds
2021-09-29 12:24:28,321 - INFO - [MainThread] cura.Settings.MachineManager.removeMachine [730]: Attempting to remove a machine with the id [FDM Printer Base Description]

smartavionics, 2021-10-02:
Hi, sorry for the slow response. I just tried this with the Linux build and it shows this...

[image]

Don't know why you aren't seeing that.

bigbeau, 2021-10-02:
I tried the latest build in Ultimaker's repository and it's the exact same issue (on windows) there. I've not tried their Linux build yet. I'll probably post this issue over there since the problem is with their version and only affects this one as a side effect of being a fork.

As a workaround I suppose I could set up the profile in an older version and export it.

I'll try disabling my plugins first in case something is messing things up.

smartavionics, 2021-10-02:
Hi, yes, I haven't made any changes to this part of Cura so if there is a fresh bug, then it's probably UM's doing.

bigbeau, 2021-10-03:
Turns out it was due to a plugin. Closing.

closed 2021-10-03 by bigbeau