smartavionics/Cura #138: Cura Master not running on Ubuntu 21.10 ???

Jens5395, 2021-11-11:
Has anyone had problems running Cura (Master branch) on the newest Ubuntu (21.10) ?
Probably something simple/stupid but it just will not start up ... no errors either..
When run from terminal, I get the following message:

[image]

Jens5395, 2021-11-11:
Further info - Although this problem showed up when trying to run the most recent release (Nov), the same thing happens when trying to run the previous release that ran fine on the previous version of Ubuntu (21.03 I think) so it seems to be related to upgrading Ubuntu.

smartavionics, 2021-11-11:
Hi, that's because your system is using wayland and Cura expects X11 to be present. I'll take a look at whether it is possible to build Cura with wayland support.

smartavionics, 2021-11-11:
A quick investigation shows that building Cura to run natively on wayland would take some effort which, in the long term, may need to be done. In the short term, could you investigate the possibility of running Cura within Xwayland?

Jens5395, 2021-11-11:
After switching to X11 windowing mode, the following slightly different error displays in the terminal:
[image]

smartavionics, 2021-11-11:
Hmm, it's probably missing some libraries. Let's see if we can find out. Run the AppImage again but this time add the --appimage-mount option, i.e. `./Desktop/Cura-mb-master-x86_64-20211110.AppImage --appimage-mount`

It should print the name of the directory that the image has been mounted on, this will be different for you but for me it was
`/tmp/.mount_Cura-mt2qqzl`.

With the appimage mounted, we can inspect its contents.

Now run the following command to list the dynamic libraries that the qpa xcb  plugin requires (don't forget to use the directory that you get rather than my example):

`ldd /tmp/.mount_Cura-mt2qqzl/usr/bin/qt/plugins/platforms/libqxcb.so`

You should get output rather like this:
```
/tmp/.mount_Cura-mt2qqzl/usr/bin/qt/plugins/platforms/libqxcb.so: /lib/x86_64-linux-gnu/libQt5Core.so.5: version `Qt_5.15' not found (required by /tmp/.mount_Cura-mt2qqzl/usr/bin/qt/plugins/platforms/libqxcb.so)
/tmp/.mount_Cura-mt2qqzl/usr/bin/qt/plugins/platforms/libqxcb.so: /lib/x86_64-linux-gnu/libQt5Core.so.5: version `Qt_5.15' not found (required by /tmp/.mount_Cura-mt2qqzl/usr/bin/qt/plugins/platforms/libQt5XcbQpa.so.5)
	linux-vdso.so.1 (0x00007fff74d21000)
	libQt5XcbQpa.so.5 => /tmp/.mount_Cura-mt2qqzl/usr/bin/qt/plugins/platforms/libQt5XcbQpa.so.5 (0x00007f9a90a1b000)
	libQt5Gui.so.5 => /lib/x86_64-linux-gnu/libQt5Gui.so.5 (0x00007f9a9040b000)
	libQt5Core.so.5 => /lib/x86_64-linux-gnu/libQt5Core.so.5 (0x00007f9a8fec2000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f9a8fce0000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f9a8faee000)
	libfontconfig.so.1 => /tmp/.mount_Cura-mt2qqzl/usr/bin/qt/plugins/platforms/libfontconfig.so.1 (0x00007f9a8faa3000)
	libfreetype.so.6 => /lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007f9a8f9e4000)
	libQt5DBus.so.5 => /lib/x86_64-linux-gnu/libQt5DBus.so.5 (0x00007f9a8f948000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f9a8f925000)
	libX11-xcb.so.1 => /lib/x86_64-linux-gnu/libX11-xcb.so.1 (0x00007f9a8f920000)
	libxcb-icccm.so.4 => /lib/x86_64-linux-gnu/libxcb-icccm.so.4 (0x00007f9a8f919000)
	libxcb-image.so.0 => /lib/x86_64-linux-gnu/libxcb-image.so.0 (0x00007f9a8f712000)
	libxcb-shm.so.0 => /lib/x86_64-linux-gnu/libxcb-shm.so.0 (0x00007f9a8f70d000)
	libxcb-keysyms.so.1 => /lib/x86_64-linux-gnu/libxcb-keysyms.so.1 (0x00007f9a8f708000)
	libxcb-randr.so.0 => /lib/x86_64-linux-gnu/libxcb-randr.so.0 (0x00007f9a8f6f5000)
	libxcb-render-util.so.0 => /lib/x86_64-linux-gnu/libxcb-render-util.so.0 (0x00007f9a8f6ee000)
	libxcb-render.so.0 => /lib/x86_64-linux-gnu/libxcb-render.so.0 (0x00007f9a8f6df000)
	libxcb-shape.so.0 => /lib/x86_64-linux-gnu/libxcb-shape.so.0 (0x00007f9a8f6d8000)
	libxcb-sync.so.1 => /lib/x86_64-linux-gnu/libxcb-sync.so.1 (0x00007f9a8f6ce000)
	libxcb-xfixes.so.0 => /lib/x86_64-linux-gnu/libxcb-xfixes.so.0 (0x00007f9a8f6c4000)
	libxcb-xinerama.so.0 => /lib/x86_64-linux-gnu/libxcb-xinerama.so.0 (0x00007f9a8f6bf000)
	libxcb-xkb.so.1 => /lib/x86_64-linux-gnu/libxcb-xkb.so.1 (0x00007f9a8f6a1000)
	libxcb-xinput.so.0 => /tmp/.mount_Cura-mt2qqzl/usr/bin/qt/plugins/platforms/libxcb-xinput.so.0 (0x00007f9a8f676000)
	libxcb.so.1 => /lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f9a8f64a000)
	libX11.so.6 => /lib/x86_64-linux-gnu/libX11.so.6 (0x00007f9a8f50d000)
	libSM.so.6 => /tmp/.mount_Cura-mt2qqzl/usr/bin/qt/plugins/platforms/libSM.so.6 (0x00007f9a8f501000)
	libICE.so.6 => /tmp/.mount_Cura-mt2qqzl/usr/bin/qt/plugins/platforms/libICE.so.6 (0x00007f9a8f4e1000)
	libxkbcommon-x11.so.0 => /lib/x86_64-linux-gnu/libxkbcommon-x11.so.0 (0x00007f9a8f4d6000)
	libxkbcommon.so.0 => /lib/x86_64-linux-gnu/libxkbcommon.so.0 (0x00007f9a8f494000)
	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f9a8f369000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f9a8f21a000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f9a8f1ff000)
	libGL.so.1 => /lib/x86_64-linux-gnu/libGL.so.1 (0x00007f9a8f177000)
	libpng16.so.16 => /lib/x86_64-linux-gnu/libpng16.so.16 (0x00007f9a8f13f000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f9a8f121000)
	libharfbuzz.so.0 => /tmp/.mount_Cura-mt2qqzl/usr/bin/qt/plugins/platforms/libharfbuzz.so.0 (0x00007f9a8f018000)
	libicui18n.so.66 => /lib/x86_64-linux-gnu/libicui18n.so.66 (0x00007f9a8ed19000)
	libicuuc.so.66 => /lib/x86_64-linux-gnu/libicuuc.so.66 (0x00007f9a8eb33000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f9a8eb2d000)
	libpcre2-16.so.0 => /tmp/.mount_Cura-mt2qqzl/usr/bin/qt/plugins/platforms/libpcre2-16.so.0 (0x00007f9a8eaa9000)
	libdouble-conversion.so.3 => /tmp/.mount_Cura-mt2qqzl/usr/bin/qt/plugins/platforms/libdouble-conversion.so.3 (0x00007f9a8ea8f000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f9a90ba5000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007f9a8ea61000)
	libuuid.so.1 => /tmp/.mount_Cura-mt2qqzl/usr/bin/qt/plugins/platforms/libuuid.so.1 (0x00007f9a8ea57000)
	libdbus-1.so.3 => /tmp/.mount_Cura-mt2qqzl/usr/bin/qt/plugins/platforms/libdbus-1.so.3 (0x00007f9a8ea02000)
	libxcb-util.so.1 => /tmp/.mount_Cura-mt2qqzl/usr/bin/qt/plugins/platforms/libxcb-util.so.1 (0x00007f9a8e7fb000)
	libXau.so.6 => /lib/x86_64-linux-gnu/libXau.so.6 (0x00007f9a8e7f3000)
	libXdmcp.so.6 => /lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f9a8e7eb000)
	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007f9a8e7d1000)
	libpcre.so.3 => /tmp/.mount_Cura-mt2qqzl/usr/bin/qt/plugins/platforms/libpcre.so.3 (0x00007f9a8e75d000)
[... cut ...]
	liblzma.so.5 => /tmp/.mount_Cura-mt2qqzl/usr/bin/qt/plugins/platforms/liblzma.so.5 (0x00007f9a8ca95000)
	liblz4.so.1 => /tmp/.mount_Cura-mt2qqzl/usr/bin/qt/plugins/platforms/liblz4.so.1 (0x00007f9a8ca73000)
	libgcrypt.so.20 => /tmp/.mount_Cura-mt2qqzl/usr/bin/qt/plugins/platforms/libgcrypt.so.20 (0x00007f9a8c953000)
	libgpg-error.so.0 => /tmp/.mount_Cura-mt2qqzl/usr/bin/qt/plugins/platforms/libgpg-error.so.0 (0x00007f9a8c92f000)
```

Please report the output you get, thanks.

Type ^C to unmount the appimage.

Jens5395, 2021-11-11:
Once I figured out that I needed to open a second terminal window to run the ldd command, things went according to plan :)

[image]

Jens5395, 2021-11-11:
I see a number of 'not found' lines .... doesn't an appimage carry all of it's dependencies and files within the appimage file ?

smartavionics, 2021-11-11:
> I see a number of 'not found' lines .... doesn't an appimage carry all of it's dependencies and files within the appimage file ?

In this case, some of the dependencies are in the appimage and some (those dependencies that are standard system files) are not. The libQt5 files are in the appimage, the one that could be causing the problem is libxcb-xinerama.so.0. So install that by doing `sudo apt install libxcb-xinerama0` and see if that helps.

smartavionics, 2021-11-11:
Just noticed that I do include libxcb-xinput.so.0 and libxcb-util.so.1 in the appimage but I needn't as they should be installed on the system along with the other libxcb libraries. I'll probably take those out for future releases.

Jens5395, 2021-11-12:
For anyone that reads this thread, doing sudo apt install libxcb-xinerama0 fixed the issue.
Thank you Mark !

smartavionics, 2021-11-12:
Thanks, I'm going to mention the need to install libxcb-xinerama0 in the notes in case others have a similar problem.

smartavionics, 2021-11-12:
Hi @Jens5395, maybe getting Cura to run under wayland isn't too hard, I've built a release that includes the Qt wayland module but can't test it here. See https://github.com/Ultimaker/Cura/issues/10815

phrxmd, 2021-11-12:
Hi @smartavionics - I tried your Wayland build, it gives an error message "Could not probe OpenGL. This program requires OpenGL 2.0 or higher. Please check your video card drivers."

Here is the console output: 
```
> QT_QPA_PLATFORM=wayland ./Cura-mb-master-x86_64-20211110-wayland.AppImage
<frozen zipimport>:259: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
Attribute Qt::AA_UseDesktopOpenGL must be set before QCoreApplication is created.
qt.qpa.wayland: Failed to load client buffer integration: "wayland-egl"
qt.qpa.wayland: Available client buffer integrations: ("xcomposite-glx")
```

smartavionics, 2021-11-12:
Hi @phrxmd , thanks for the speedy feedback. What happens if you try `QT_QPA_PLATFORM=wayland-xcomposite-glx` ?

phrxmd, 2021-11-12:
Nothing. I get this in the console:
```
> QT_QPA_PLATFORM=wayland-xcomposite-glx ./Cura-mb-master-x86_64-20211110-wayland.AppImage
<frozen zipimport>:259: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
Using XComposite-GLX
```
...and then it hangs, and can't be terminated with Ctrl+C either.

smartavionics, 2021-11-12:
OK, well, I'll work on this when I can (quite busy) and report progress here.

smartavionics, 2021-11-12:
Hi @phrxmd, I have made some progress. I installed weston which provides wayland in a window on X11 and have managed to build an appimage that appears to work with QT_QPA_PLATFORM=wayland. So I have updated the release, the link is the same as before, make sure you get a new download.

smartavionics, 2021-11-12:
[image]

smartavionics, 2021-11-13:
Hmm, there's a problem with the menus (File, Edit, ...) but maybe that's just with weston?

phrxmd, 2021-11-13:
Hi @smartavionics - some progress, I see the splash screen now, but before the main window opens I get a segfault:
```
QT_QPA_PLATFORM=wayland ./Cura-mb-master-x86_64-20211110-wayland-2.AppImage
<frozen zipimport>:259: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
Attribute Qt::AA_UseDesktopOpenGL must be set before QCoreApplication is created.
Cyclic dependency detected between "file:///tmp/.mount_Cura-mrS9YbY/usr/bin/resources/qml/Actions.qml" and "file:///tmp/.mount_Cura-mrS9YbY/usr/bin/resources/qml/Actions.qml"
WARNING: Cannot find style "default" - fallback: "/tmp/.mount_Cura-mrS9YbY/usr/bin/qt/qml/QtQuick/Controls/Styles/Desktop"
/tmp/.mount_Cura-mrS9YbY/usr/bin//cura.sh: line 40: 25006 Segmentation fault      (core dumped) cura "$@"
```

smartavionics, 2021-11-13:
Could you please post the ~/.local/share/cura/master/cura.log and ~/.local/share/cura/stderr.log files? Thanks.

smartavionics, 2021-11-14:
A little background reading...

https://gist.github.com/probonopd/9feb7c20257af5dd915e3a9f2d1f2277

smartavionics, 2021-11-14:
As for the menu problem, I have found that if you open the desired menu using the keyboard (i.e. alt-f or alt-v, etc.) then the correct menu will stay open.

However, I have also discovered that opening some file dialogs causes the app to hang.

To be honest, I don't think Cura is usable under wayland yet.

ajshell1, 2022-01-11:
> To be honest, I don't think Cura is usable under wayland yet.

I would disagree with that statement. The version of Cura provided by the Arch Linux repositories does indeed work for me under Wayland (specifically Sway), out of the box. Unfortunately, this version does have this bug: https://github.com/Ultimaker/Cura/issues/10785

I'd be willing to help you do some more testing, if you're willing.

smartavionics, 2022-01-11:
> I'd be willing to help you do some more testing, if you're willing.

What happens when you try the following with the latest release?

 `QT_QPA_PLATFORM=wayland ./Cura-mb-master-x86_64-20220111.AppImage`

ajshell1, 2022-01-12:
> > I'd be willing to help you do some more testing, if you're willing.
> 
> What happens when you try the following with the latest release?
> 
> `QT_QPA_PLATFORM=wayland ./Cura-mb-master-x86_64-20220111.AppImage`

Huh. It runs. Without issue. I imported an STL file, sliced it, saved the gcode, and without any obvious issues. 

I'm not sure what you did since I necro'd this issue, but thanks!

Meanwhile, Ultimaker_Cura-4.13.0.AppImage only runs properly if I run "QT_QPA_PLATFORM=xcb ./Ultimaker_Cura-4.13.0.AppImage" . So congratulations!

phrxmd, 2022-01-13:
> > I'd be willing to help you do some more testing, if you're willing.
> 
> What happens when you try the following with the latest release?
> 
> `QT_QPA_PLATFORM=wayland ./Cura-mb-master-x86_64-20220111.AppImage`

Hm. I still get segfaults with that one:

````
> QT_QPA_PLATFORM=wayland LANGUAGE=C ./Cura-mb-master-x86_64-20220111.AppImage
<frozen zipimport>:259: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
Attribute Qt::AA_UseDesktopOpenGL must be set before QCoreApplication is created.
Cyclic dependency detected between "file:///tmp/.mount_Cura-mmSkIjQ/usr/bin/resources/qml/Actions.qml" and "file:///tmp/.mount_Cura-mmSkIjQ/usr/bin/resources/qml/Actions.qml"
WARNING: Cannot find style "default" - fallback: "/tmp/.mount_Cura-mmSkIjQ/usr/bin/qt/qml/QtQuick/Controls/Styles/Desktop"
/tmp/.mount_Cura-mmSkIjQ/usr/bin//cura.sh: line 40: 11962 Segmentation fault      (core dumped) cura "$@"
````

smartavionics, 2022-01-13:
Hi @phrxmd. Unfortunately, your cura.log doesn't show anything helpful so I'm no wiser as to what's going wrong. Sorry.

phrxmd, 2022-01-13:
> Hi @phrxmd. Unfortunately, your cura.log doesn't show anything helpful so I'm no wiser as to what's going wrong. Sorry.

No problem. Is there any way I can provide further information - increase the verbosity of the output, extra logging or something else?

smartavionics, 2022-01-13:
Hi, I've just uploaded https://github.com/smartavionics/Cura/releases/download/20220111/Cura-mb-master-x86_64-20220111-gdb.AppImage which lets you run gdb on the cura process. To do that, prefix the command line with CURA_GDB=gdb, so the whole command line would be...
`CURA_GDB=gdb QT_QPA_PLATFORM=wayland ./Cura-mb-master-x86_64-20220111-gdb.AppImage`
When it runs you should get a gdb prompt so type "run" to that to start cura running and lets see what gets reported when it crashes...

phrxmd, 2022-01-13:
Thank you. However, Cura doesn't start from that image:

````
> CURA_GDB=gdb QT_QPA_PLATFORM=wayland ./Cura-mb-master-x86_64-20220111-gdb.AppImage
GNU gdb (GDB; openSUSE Tumbleweed) 11.1
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-suse-linux".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://bugs.opensuse.org/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from cura...

warning: Loadable section ".dynstr" outside of ELF segments
  in /tmp/.mount_Cura-mzkPApa/usr/bin/cura
(No debugging symbols found in cura)
(gdb) run
Starting program: /tmp/.mount_Cura-mzkPApa/usr/bin/cura 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
Fatal error: Unable to determine absolute path for executable!
[Inferior 1 (process 29380) exited with code 01]
(gdb) 
````

phrxmd, 2022-01-14:
For what it's worth, when building from source I also get a segfault under Wayland that goes away when starting with `QT_QPA_PLATFORM=xcb`. I'm building OpenSUSE packages, [here](https://build.opensuse.org/package/show/home:rxmd/cura) is the .spec file I'm using.

But here at least Python tells me a bit more:
```
cura
/usr/lib/python3.8/site-packages/UM/PluginRegistry.py:4: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import imp
/usr/lib/python3.8/site-packages/UM/Settings/InstanceContainer.py:53: DeprecationWarning: invalid escape sequence \d
  version_regex = re.compile("\nversion ?= ?(\d+)")
/usr/lib/python3.8/site-packages/UM/Settings/InstanceContainer.py:55: DeprecationWarning: invalid escape sequence \w
  type_regex = re.compile("\ntype ?= ?(\w+)")
/usr/lib/python3.8/site-packages/UM/VersionUpgradeManager.py:98: DeprecationWarning: invalid escape sequence \.
  ".*\.lock",       # Don't upgrade the configuration file lock. It's not persistent.
/usr/lib/python3.8/site-packages/cura/UI/ObjectsModel.py:72: DeprecationWarning: invalid escape sequence \(
  self._naming_regex = re.compile("^(.+)\(([0-9]+)\)$")
Attribute Qt::AA_UseDesktopOpenGL must be set before QCoreApplication is created.
2022-01-14 14:37:48,850 - DEBUG - [MainThread] UM.Application.initialize [161]: Initializing Ultimaker Cura
2022-01-14 14:37:48,851 - DEBUG - [MainThread] UM.Application.initialize [162]: App Version 4.13.0
2022-01-14 14:37:48,851 - DEBUG - [MainThread] UM.Application.initialize [163]: Api Version 7.9.0
2022-01-14 14:37:48,851 - DEBUG - [MainThread] UM.Application.initialize [164]: Build type RPM
2022-01-14 14:37:48,851 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [200]: Plugin location /usr/bin/share/uranium/plugins must be a folder.
2022-01-14 14:37:48,851 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [200]: Plugin location /usr/bin/share/cura/plugins must be a folder.
2022-01-14 14:37:48,851 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [200]: Plugin location /usr/lib64/uranium must be a folder.
2022-01-14 14:37:48,851 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [200]: Plugin location /usr/lib32/uranium must be a folder.
2022-01-14 14:37:48,851 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [200]: Plugin location /usr/bin/plugins must be a folder.
2022-01-14 14:37:48,851 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [200]: Plugin location /usr/Resources/uranium/plugins must be a folder.
2022-01-14 14:37:48,852 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [200]: Plugin location /usr/Resources/cura/plugins must be a folder.
2022-01-14 14:37:48,852 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [447]: Initializing storage paths
2022-01-14 14:37:48,852 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [457]: Config storage path is /home/reichmuth/.config/cura/4.13
2022-01-14 14:37:48,852 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [465]: Data storage path is /home/reichmuth/.local/share/cura/4.13
2022-01-14 14:37:48,852 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [477]: Cache storage path is /run/user/1000/.cache/cura/4.13
2022-01-14 14:37:48,852 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [200]: Plugin location /usr/lib/python3.8/site-packages/UM/../plugins must be a folder.
2022-01-14 14:37:48,852 - INFO - [MainThread] UM.PackageManager.__init__ [55]: Found bundled packages JSON file: /usr/share/cura/resources/bundled_packages/cura.json
2022-01-14 14:37:48,852 - INFO - [MainThread] UM.PackageManager.__init__ [55]: Found bundled packages JSON file: /usr/share/uranium/resources/bundled_packages/uranium.json
2022-01-14 14:37:48,852 - DEBUG - [MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [123]: Trying OpenGL context 4.1...
2022-01-14 14:37:48,852 - DEBUG - [MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [131]: Yay, we got at least OpenGL 4.1 core: 4.6 Core profile
2022-01-14 14:37:48,852 - DEBUG - [MainThread] UM.Qt.QtApplication.initialize [177]: Detected most suitable OpenGL context version: 4.1 Core profile
2022-01-14 14:37:48,852 - INFO - [MainThread] UM.Qt.QtApplication.initialize [184]: Initializing job queue ...
2022-01-14 14:37:48,852 - INFO - [MainThread] UM.Qt.QtApplication.initialize [188]: Initializing version upgrade manager ...
2022-01-14 14:37:48,853 - INFO - [MainThread] UM.PackageManager._loadManagementData [170]: Loaded bundled packages data from /usr/share/cura/resources/bundled_packages/cura.json
2022-01-14 14:37:48,853 - INFO - [MainThread] UM.PackageManager._loadManagementData [170]: Loaded bundled packages data from /usr/share/uranium/resources/bundled_packages/uranium.json
2022-01-14 14:37:48,853 - INFO - [MainThread] UM.PackageManager._loadManagementData [197]: Loaded user packages management file from /home/reichmuth/.local/share/cura/4.13/packages.json
2022-01-14 14:37:48,853 - INFO - [MainThread] UM.PackageManager._saveManagementData [270]: Package management file /home/reichmuth/.local/share/cura/4.13/packages.json was saved
2022-01-14 14:37:48,853 - INFO - [MainThread] UM.PluginRegistry.initializeBeforePluginsAreLoaded [119]: Loading plugin configuration file '/home/reichmuth/.config/cura/4.13/plugins.json'
[... cut ...]
  File "/usr/lib64/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib64/python3.8/threading.py", line 890 in _bootstrap

Current thread 0x00007fcc15d1cb80 (most recent call first):
  File "/usr/lib/python3.8/site-packages/UM/Qt/QtApplication.py", line 319 in initializeEngine
  File "/usr/lib/python3.8/site-packages/cura/CuraApplication.py", line 942 in runWithGUI
  File "/usr/lib/python3.8/site-packages/cura/CuraApplication.py", line 859 in run
  File "/usr/bin/cura", line 236 in <module>
[1]    31984 segmentation fault (core dumped)  cura
```

smartavionics, 2022-01-15:
> Fatal error: Unable to determine absolute path for executable!

I've made a small change and uploaded https://github.com/smartavionics/Cura/releases/download/20220111/Cura-mb-master-x86_64-20220111-gdb-2.AppImage, could you please try again. Thanks.

phrxmd, 2022-02-06:
Sorry for the delay in answering, life got in the way. I still get the same error:

```
> CURA_GDB=gdb QT_QPA_PLATFORM=wayland ./Cura-mb-master-x86_64-20220111-gdb-2.AppImage
GNU gdb (GDB; openSUSE Tumbleweed) 11.1
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-suse-linux".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://bugs.opensuse.org/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /tmp/.mount_Cura-mjOY6Li/usr/bin/cura...

warning: Loadable section ".dynstr" outside of ELF segments
  in /tmp/.mount_Cura-mjOY6Li/usr/bin/cura
(No debugging symbols found in /tmp/.mount_Cura-mjOY6Li/usr/bin/cura)
(gdb) run
Starting program: /tmp/.mount_Cura-mjOY6Li/usr/bin/cura 
Downloading 0.58 MB separate debug info for /lib64/ld-linux-x86-64.so.2
Downloading 0.01 MB separate debug info for /lib64/libpthread.so.0 
Downloading 0.01 MB separate debug info for /lib64/libdl.so.2      
Downloading 0.01 MB separate debug info for /lib64/libutil.so.1
Downloading 1.72 MB separate debug info for /lib64/libm.so.6       
Downloading 5.88 MB separate debug info for /lib64/libc.so.6       
[Thread debugging using libthread_db enabled]                      
Using host libthread_db library "/lib64/libthread_db.so.1".
Fatal error: Unable to determine absolute path for executable!
[Inferior 1 (process 28694) exited with code 01]
(gdb) 
```

smartavionics, 2022-02-06:
I haven't a clue how to progress this. Sorry.

phrxmd, 2022-02-06:
Thanks anyway :)