smartavionics/Cura #146: Fan = 0, toggle for initial fan speed @ Layer....currently defaults to regular fan speed. (current, 4.12)

PhilBaz, 2022-01-03:
**Application version**
(The version of the application this issue occurs with.)
 4.12
**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
windows
**Printer**
(Which printer was selected in Cura?)
Voron 2.4 350
**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)
Just using initial fan speed & initial fan speed at layer
**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

Using initial fan & initial fan speed at layer. When initial fan speed at layer is used (lets say layer 10) the layers prior (1-9) use the 'regular fan speed'.

**Expected results**
(What should happen after the above steps have been followed.)

When initial fan speed at layer is used (lets say layer 10) would be nice if the layers prior (1-9) Fan was/could be set to zero. Often it would be nice to have 0 fan<layer 3, then initial fan speed at layer 3, ramping up to regular speed at 'regular fan speed at layer'. 

As proposed above then you could have zero fan, to slow fan, ramping to high fan. 3 phases. Right now it goes from high fan, to low fan, back to high. 

Thanks for your awesome fork!!!!

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

smartavionics, 2022-01-03:
Hi, I thought it did behave as you are expecting. Could you please provide a project file that shows the issue? Thanks.

PhilBaz, 2022-01-03:
Here is the gcode and the 3mf.....just sliced this one to make sure, in case I got it wrong some how... But you can see the initial m106 goes to reg. speed rather than zero

PS...would you happen to know the angle for overhang display (in red)?

Thanks again!
[fan speed issue.zip](https://github.com/smartavionics/Cura/files/7802605/fan.speed.issue.zip)
.

smartavionics, 2022-01-03:
OK, it's probably a bug (in the original UM code) but when the Regular/max Fan Speed Threshold value is <= Minimum Layer Time, then it sets the fan speed to the min fand speed. You have both set to 5.0. Setting Reg/Max Fan Speed Threshold to 6.0 makes the fan be zero on the first layer and it starts ramping up from layer 10.

The overhang display is controlled by the Support Overhang Angle setting.

PhilBaz, 2022-01-03:
> OK, it's probably a bug (in the original UM code) but when the Regular/max Fan Speed Threshold value is <= Minimum Layer Time, then it sets the fan speed to the min fand speed. You have both set to [...]

Thank you & thank you !!!