smartavionics/Cura #147: Running the AppImage aarch64 on a Chromebook

castaway, 2022-01-07:
**Application version**
./Downloads/Cura-mb-master-aarch64-20211119.AppImage

**Platform**
Acer Spin 513 Chromebook (8Gb RAM) - Linux penguin 5.4.151-16908-gff376e5d5ee1 #1 SMP PREEMPT Sat Dec 11 19:42:45 PST 2021 aarch64 GNU/Linux

**Printer**
n/a

**Reproduction steps**
1. Download aarch64 AppImage build
2. chmod +x it
3. Install zlibg1q (libz)
4. Attempt to run AppImage

**Screenshot(s)**
[image]

**Actual results**
Get a tiny window with an empty text area with an  "ok" button, that just closes it

**Expected results**
Cura to start up

**Log file**
None existing

**Additional information**
This is a chromebook, not a Pi, so this may be all wishful thinking.. There's no libgles-dev available, so it could be that? Any ideas?

smartavionics, 2022-01-07:
Hi.

Although the x86_64 appimage will run on a chromebook that uses an Intel processor, I don't think the aarch64 appimage will work on a chromebook because it is expecting the OpenGL to be GLES based on Broadcom GPU.

Have you tried using the mesa build I provide? That may provide some missing libraries you need?

castaway, 2022-01-07:
ooh, I'm a nitwit, I do have a "libgles2" however, now it starts!

smartavionics, 2022-01-07:
So is the UI usable? Can you slice and preview the results?

castaway, 2022-01-07:
Slice + save yes, preview says "you havent sliced yet" (!?) ..

smartavionics, 2022-01-08:
That's weird, what does it show in the popup you get when you hover the cursor over the (i) above the save to disk button?