smartavionics/Cura #149: "Python error in main script"

Snargola, 2022-01-08:
**Application version**
20211110

**Platform**
Windows 10

**Printer**
N/A

**Reproduction steps**
Started the program

**Screenshot(s)**

[image]

**Actual results**
Program crashes

**Expected results**
Program should start

smartavionics, 2022-01-08:
Hmm, don't know what the problem is there. How are you starting Cura?

Snargola, 2022-01-08:
Hello and thank you for your reply.

I have downloaded, and installed, the .exe file from this location:
https://github.com/smartavionics/Cura/releases/download/20211110/Cura-mb-master-win64-20211110.exe

I have accepted all of the defaults.
It created an icon on my Start menu that I have tried to start the program from. I have also tried to start it directly from File Explorer, with no luck. I have tried restarting my computer because, you know, Windows, with the same results.

The error points to a path/user that does not exist on my computer.

I'm going to attempt an uninstall/reinstall but I'm not holding my breath.

Thank you for your time and I look forward to any help you may be able to provide.

smartavionics, 2022-01-08:
> The error points to a path/user that does not exist on my computer.

Yes, that path is the original path of the file when the app was built. That's OK.

Do you get a cura.log file under app-data/roaming/cura/master ?

Snargola, 2022-01-08:
This is what I have:
[image]

smartavionics, 2022-01-08:
Hmm, no master folder. Must be crashing before cura.log is opened.

Do you have stderr.log ?

smartavionics, 2022-01-08:
That should be in AppData/Roaming/cura

smartavionics, 2022-01-08:
Yes, you do, I see it in the image above. What's inside that, please?

Snargola, 2022-01-08:
> Yes, you do, I see it in the image above. What's inside that, please?

You want the whole file? I'm assuming I can drag it into this window...?

Snargola, 2022-01-08:
[stderr.log](https://github.com/smartavionics/Cura/files/7833534/stderr.log)

Snargola, 2022-01-08:
For what it's worth, I can't run it as an Administrator, either.

smartavionics, 2022-01-08:
Thanks for the stderr.log file. Unfortunately, it only contains info from a previous run of 4.12.

So I don't know why my release is not working for you. Nobody else has reported a similar problem. Have you tried the next release, 20201119 ?

Snargola, 2022-01-08:
> Thanks for the stderr.log file. Unfortunately, it only contains info from a previous run of 4.12.
> 
> So I don't know why my release is not working for you. Nobody else has reported a similar problem. Have you tried the next release, 20201119 ?

I'll try it later today. Thank you for the assistance.

Snargola, 2022-01-08:
(sigh) Same problem with an older release :(

I am able to compile Marlin. How many orders of magnitude more complex is it to attempt to compile this?

Side note, and I don't think it has anything to do with the problem, here is this:
[image]

smartavionics, 2022-01-08:
Well, I don't know what's going wrong there. Do you have any other python based programs installed on your system? Perhaps there's some conflict with another program or DLLs? Sorry, I'm clueless when it comes to Windows as I never use it apart from when I build Cura releases.

Snargola, 2022-01-08:
I probably wouldn't know a Python based program if it bit me in the face :) Thank you for your assistance. I'm going to keep messing with it.

Snargola, 2022-01-09:
I installed Python, because, why not. Didn't help.

I found this, but really have no idea what they are talking about. Maybe you can geek it out?
https://github.com/Ultimaker/cura-build/issues/123

joe-ko, 2022-01-09:
i get the same errormessage here. only version 20210908 works but not the versions newer

smartavionics, 2022-01-09:
Well, there's not much difference between 20210908 and the later releases so hopefully you can use that. I will be making a new release soon after 4.13 is released.

Snargola, 2022-01-09:
> i get the same errormessage here. only version 20210908 works but not the versions newer

20210908, not working for me :(

notcraig, 2022-02-06:
Any news on this? Running in the same issue. 20210908 and newer.

Snargola, 2022-02-07:
After trying way too many things, I never got it work :(

notcraig, 2022-02-08:
> Any news on this? Running in the same issue. 20210908 and newer.

FWIW, this only happens on the Windows version. This isnt an issue with the Linux AppImage