smartavionics/Cura #150: Running CURA on UBUNTU 21.10 44 bits and RPI 4

jfle16, 2022-01-09:
Good morning,

I tried to run appimage Cura-mb-master-aarch64-20211119 on my raspberry pi 4 and UBUNTU 21.10 64 bits but without success.
I got this following error: 

`./Cura-mb-master-aarch64-20211119.AppImage: error while loading shared libraries: libz.so: cannot open shared object file: No such file or directory
`

I tried also the installation with Pi-APP but without success during launch.

Thank you in advance for your help!

Best regards,

smartavionics, 2022-01-09:
Try installing libz.so with sudo apt install zlib1g

jfle16, 2022-01-09:
Thank you for your prompt answer smartavionics!

Zlib1g is unfortunately already installed :+1: 
`sudo apt install zlib1g
[sudo] password for jerome: 
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances... Fait
Lecture des informations d'état... Fait      
zlib1g est déjà la version la plus récente (1:1.2.11.dfsg-2ubuntu7).
zlib1g passé en « installé manuellement ».
0 mis à jour, 0 nouvellement installés, 0 à enlever et 0 non mis à jour.`

smartavionics, 2022-01-09:
Hmm, well, in that case, I don't know what's going wrong. Sorry.

jfle16, 2022-01-09:
After installing zlib1g-dev it is running well ... Problem solved.

Thank you for your help!

smartavionics, 2022-01-09:
Ah, yes, it needs the -dev package as that contains the .so, sorry, my mistake, glad it's working now.