smartavionics/Cura #152: [20211110] execv error: Input/output error

Botspot, 2022-01-16:
**Application version**
20211110

**Platform**
Raspberry Pi OS 32-bit, using the default armhf kernel.

**Expected results**
The appimage would launch.

**Actual results**
The appimage did not launch. It exited with this error:
```
$ ~/Cura.AppImage
execv error: Input/output error
```
This used to work. The only change I can think of would be how I turned off the 64-bit kernel. I'm running the default 32-bit kernel.
Strace gives more detail:
```
$ strace ~/Cura.AppImage
strace ~/Cura.AppImage
execve("/home/pi/Cura.AppImage", ["/home/pi/Cura.AppImage"], 0xbed36130 /* 62 vars */) = 0
brk(NULL)                               = 0x424000
mmap2(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6f35000
access("/etc/ld.so.preload", R_OK)      = 0
openat(AT_FDCWD, "/etc/ld.so.preload", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=54, ...}) = 0
mmap2(NULL, 54, PROT_READ|PROT_WRITE, MAP_PRIVATE, 3, 0) = 0xb6f34000
close(3)                                = 0
readlink("/proc/self/exe", "/home/pi/Cura.AppImage", 4096) = 22
openat(AT_FDCWD, "/usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\254\3\0\0004\0\0\0"..., 512) = 512
fstat64(3, {st_mode=S_IFREG|0644, st_size=17708, ...}) = 0
mmap2(NULL, 81964, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6ef0000
mprotect(0xb6ef4000, 61440, PROT_NONE)  = 0
mmap2(0xb6f03000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x3000) = 0xb6f03000
close(3)                                = 0
munmap(0xb6f34000, 54)                  = 0
openat(AT_FDCWD, "/deps/lib/tls/v7l/neon/vfp/libdl.so.2", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/deps/lib/tls/v7l/neon/vfp", 0xbea174d0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/deps/lib/tls/v7l/neon/libdl.so.2", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/deps/lib/tls/v7l/neon", 0xbea174d0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/deps/lib/tls/v7l/vfp/libdl.so.2", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/deps/lib/tls/v7l/vfp", 0xbea174d0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/deps/lib/tls/v7l/libdl.so.2", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/deps/lib/tls/v7l", 0xbea174d0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/deps/lib/tls/neon/vfp/libdl.so.2", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/deps/lib/tls/neon/vfp", 0xbea174d0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/deps/lib/tls/neon/libdl.so.2", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/deps/lib/tls/neon", 0xbea174d0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/deps/lib/tls/vfp/libdl.so.2", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/deps/lib/tls/vfp", 0xbea174d0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/deps/lib/tls/libdl.so.2", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/deps/lib/tls", 0xbea174d0)     = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/deps/lib/v7l/neon/vfp/libdl.so.2", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/deps/lib/v7l/neon/vfp", 0xbea174d0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/deps/lib/v7l/neon/libdl.so.2", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/deps/lib/v7l/neon", 0xbea174d0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/deps/lib/v7l/vfp/libdl.so.2", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/deps/lib/v7l/vfp", 0xbea174d0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/deps/lib/v7l/libdl.so.2", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/deps/lib/v7l", 0xbea174d0)     = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/deps/lib/neon/vfp/libdl.so.2", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/deps/lib/neon/vfp", 0xbea174d0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/deps/lib/neon/libdl.so.2", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/deps/lib/neon", 0xbea174d0)    = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/deps/lib/vfp/libdl.so.2", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/deps/lib/vfp", 0xbea174d0)     = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/deps/lib/libdl.so.2", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/deps/lib", 0xbea174d0)         = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
[... 75 lines cut ...]
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=26951, tv_nsec=135719772}) = 0
mkdir("/tmp/.mount_Cura.AzA37NI", 0700) = 0
pipe([3, 4])                            = 0
clone(child_stack=NULL, flags=CLONE_CHILD_CLEARTID|CLONE_CHILD_SETTID|SIGCHLD, child_tidptr=0xb6f330a8) = 1225
close(4)                                = 0
read(3, "x", 1)                         = 1
--- SIGCHLD {si_signo=SIGCHLD, si_code=CLD_EXITED, si_pid=1225, si_uid=1000, si_status=0, si_utime=0, si_stime=0} ---
wait4(1225, NULL, 0, NULL)              = 1225
openat(AT_FDCWD, "/tmp/.mount_Cura.AzA37NI", O_RDONLY|O_LARGEFILE) = 4
dup2(4, 1023)                           = 1023
close(4)                                = 0
access("/home/pi/Cura.AppImage.home", W_OK) = -1 ENOENT (No such file or directory)
access("/home/pi/Cura.AppImage.config", W_OK) = -1 ENOENT (No such file or directory)
getcwd("/home/pi", 1024)                = 9
execve("/tmp/.mount_Cura.AzA37NI/AppRun", ["/home/pi/Cura.AppImage"], 0x4248d0 /* 66 vars */) = -1 EIO (Input/output error)
dup(2)                                  = 4
fcntl64(4, F_GETFL)                     = 0x80002 (flags O_RDWR|O_CLOEXEC)
fstat64(4, {st_mode=S_IFCHR|0620, st_rdev=makedev(0x88, 0x4), ...}) = 0
write(4, "execv error: Input/output error\n", 32execv error: Input/output error
) = 32
close(4)                                = 0
exit_group(127)                         = ?
+++ exited with 127 +++
```
This error doesn't occur on default Raspbian, so my system is ultimately causing the problem. But I'd rather learn what the problem is and fix it, rather than re-install the OS every week. :)

smartavionics, 2022-01-16:
The cura image is the 32 bit build and not the 64 bit build, isn't it?

Botspot, 2022-01-16:
> The cura image is the 32 bit build and not the 64 bit build, isn't it?

Yes.

Botspot, 2022-01-16:
Strange: I just downloaded 20211110 armhf appimage from your releases page and it launched fine. My app store uses the same url, but the file hashes turn out different.

closed 2022-01-16 by Botspot

Botspot, 2022-01-17:
I think it's safe to chalk this up to an incomplete download.