smartavionics/Cura #156: [emailed  version of curaengine that fixed Spiralize Combing] Slicing Fails when you add a raft

mattwn, 2022-01-30:
**Application version**
4.13.0 front end, emailed CuraEngine with Spiralize Combing fix

**Platform**
Windows 11

**Printer**
Ender 3

**Reproduction steps**
1. Add the Viking Horn model referenced in Cura https://github.com/Ultimaker/Cura/issues/11370 (project file with raft attached below)
2. Add a raft
3. Click Slice

**Screenshot(s)**
[image]

**Actual results**
Slicer Failed pop up generated. Backend generated error: 3221225477, more details in log below.

**Expected results**
Model sliced with raft and the new fix for Spiralized Combing

**Project file**
[CE3_VikingHorn-RaftIssue.zip](https://github.com/smartavionics/Cura/files/7966822/CE3_VikingHorn-RaftIssue.zip)

**Log file**
[cura.log](https://github.com/smartavionics/Cura/files/7966836/cura.log)

**Additional information**
This does not stop us from using your awesome Spiralize Combing fix, but I thought you would want to know about this bug. We do sometimes use rafts a built in stands for the vases, but again, not a deal breaker for us as brim can serve the same function.

smartavionics, 2022-01-30:
Ooops, that was dumb, sloppy coding and inadequate testing bites again. Anyway, please try this replacement...

https://github.com/smartavionics/Cura/releases/download/20220129/CuraEngine-for-mattwn-2.exe

smartavionics, 2022-01-30:
[image]

mattwn, 2022-01-31:
That version worked perfect. Raft works now. I've tried a bunch of other parameters, just to stress test it. I haven't found any other issues. 

I know you're you have your own branch and aren't interested a PR for this into mainline, but would it be easy to post another code diff with the raft fix in the original issue (in the Cura repo) so someone else can run with it? I would hate to see such great work not help out others.

Please consider accepting the payment link we sent to your email as a token of our gratitude. Thanks again! You have massively improved our workflow. We (and our customers) are very thankful.

You may close this issue as you see fit.

mattwn, 2022-01-31:
That sounds great. I see Burtoogle is already running with it a bit. Ghostkeeper said earlier (on a related issue before we identified the specific problem) that UM was open to the idea of a PR on this.

As I did further stress testing, I noticed that `Z Seam Alignment: Random` didn't randomize. This is not an issue for us at all so please don't feel the need to address it for our sake. I just wanted to report my findings.

Thanks again!

smartavionics, 2022-01-31:
The z seam setting is ignored when spiralizing, the position of the seam wanders about to ensure that the nozzle only ever moves in a forward direction as it crosses the boundary between layers. With the smoothing turned on, it should be very difficult to detect the seam on the spiralized layers that don't contain any support.

mattwn, 2023-06-08:
This was actually accepted into Cura main branch and combing now works correctly in version 5.2 forward. Thanks Smartavionics!

closed 2023-06-08 by mattwn