smartavionics/Cura #157: aarch64 appimage error for new 64Bit release of Raspbian

poevoid, 2022-02-05:
**Application version**
(The version of the application this issue occurs with.)
[20220129](https://github.com/smartavionics/Cura/releases/tag/20220129)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Raspberry Pi 4 running Raspberry Pi OS (64Bit)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
Made sure to  have libgles-dev installed, gave permissions to the file, executed it in terminal.

2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)
Terminall shows this message
'''pi@raspberrypi:~/Downloads $ ./Cura-mb-master-aarch64-20220129.AppImage 
Found libGLESv2
/tmp/.mount_Cura-mEuNu5k/usr/bin//cura.sh: line 36: message: command not found

<frozen zipimport>:259: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: minimal, offscreen, vnc, webgl, xcb.

/tmp/.mount_Cura-mEuNu5k/usr/bin//cura.sh: line 57:  1705 Aborted                 cura "$@"'''

**Expected results**
(What should happen after the above steps have been followed.)
Program should open and run
**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

smartavionics, 2022-02-05:
It could be missing some libraries. Please first mount the appimage by typing the following into a shell window:

Cura-mb-master-aarch64-20220129.AppImage --appimage-mount

It should print the name of the directory the appimage is mounted on. Then in another shell, run ldd on the libQt5XcbQpa module like this example (substituting in the correct /tmp.mount name, of course):

ldd /tmp/.mount_Cura-mN7hudB/usr/bin/qt/plugins/xcbglintegrations/libQt5XcbQpa.so.5

It should then print a bunch of stuff that looks like this:
```
	linux-vdso.so.1 (0x0000007fabe88000)
	libfontconfig.so.1 => /tmp/.mount_Cura-mN7hudB/usr/bin/qt/plugins/xcbglintegrations/libfontconfig.so.1 (0x0000007fabca2000)
	libfreetype.so.6 => /usr/lib/aarch64-linux-gnu/libfreetype.so.6 (0x0000007fabbc9000)
	libz.so.1 => /lib/aarch64-linux-gnu/libz.so.1 (0x0000007fabb9c000)
	libGLESv2.so.2 => /usr/lib/aarch64-linux-gnu/libGLESv2.so.2 (0x0000007fabb67000)
	libQt5Gui.so.5 => /usr/lib/aarch64-linux-gnu/libQt5Gui.so.5 (0x0000007fab5fc000)
	libQt5DBus.so.5 => /usr/lib/aarch64-linux-gnu/libQt5DBus.so.5 (0x0000007fab553000)
	libQt5Core.so.5 => /usr/lib/aarch64-linux-gnu/libQt5Core.so.5 (0x0000007fab041000)
	libpthread.so.0 => /lib/aarch64-linux-gnu/libpthread.so.0 (0x0000007fab012000)
	libX11-xcb.so.1 => /usr/lib/aarch64-linux-gnu/libX11-xcb.so.1 (0x0000007fab000000)
        ...
```

Please copy the output you get here. Thanks.

smartavionics, 2022-02-05:
It's quite possibly missing the xinerama library so it's worth doing...

```
sudo apt install libxcb-xinerama0
```

poevoid, 2022-02-05:
Xinerama library is already installed and up to date. I ran ldd on the module and got this:

`pi@raspberrypi:~ $ ldd /tmp/.mount_Cura-mVn5hpm/usr/bin/qt/plugins/xcbglintegrations/libQt5XcbQpa.so.5 
	linux-vdso.so.1 (0x0000007f7fbd0000)
	libfontconfig.so.1 => /tmp/.mount_Cura-mVn5hpm/usr/bin/qt/plugins/xcbglintegrations/libfontconfig.so.1 (0x0000007f7f9e8000)
	libfreetype.so.6 => /lib/aarch64-linux-gnu/libfreetype.so.6 (0x0000007f7f90f000)
	libz.so.1 => /lib/aarch64-linux-gnu/libz.so.1 (0x0000007f7f8e5000)
	libGLESv2.so.2 => /lib/aarch64-linux-gnu/libGLESv2.so.2 (0x0000007f7f8b0000)
	libQt5Gui.so.5 => /lib/aarch64-linux-gnu/libQt5Gui.so.5 (0x0000007f7f221000)
	libQt5DBus.so.5 => /lib/aarch64-linux-gnu/libQt5DBus.so.5 (0x0000007f7f186000)
	libQt5Core.so.5 => /lib/aarch64-linux-gnu/libQt5Core.so.5 (0x0000007f7ec1e000)
	libpthread.so.0 => /lib/aarch64-linux-gnu/libpthread.so.0 (0x0000007f7ebed000)
	libX11-xcb.so.1 => /lib/aarch64-linux-gnu/libX11-xcb.so.1 (0x0000007f7ebdb000)
	libxcb-icccm.so.4 => /lib/aarch64-linux-gnu/libxcb-icccm.so.4 (0x0000007f7ebc6000)
	libxcb-image.so.0 => /lib/aarch64-linux-gnu/libxcb-image.so.0 (0x0000007f7ebb1000)
	libxcb-shm.so.0 => /lib/aarch64-linux-gnu/libxcb-shm.so.0 (0x0000007f7eb9d000)
	libxcb-util.so.0 => not found
	libxcb-keysyms.so.1 => /lib/aarch64-linux-gnu/libxcb-keysyms.so.1 (0x0000007f7eb8a000)
	libxcb-randr.so.0 => /lib/aarch64-linux-gnu/libxcb-randr.so.0 (0x0000007f7eb6a000)
	libxcb-render-util.so.0 => /lib/aarch64-linux-gnu/libxcb-render-util.so.0 (0x0000007f7eb56000)
	libxcb-render.so.0 => /lib/aarch64-linux-gnu/libxcb-render.so.0 (0x0000007f7eb39000)
	libxcb-shape.so.0 => /lib/aarch64-linux-gnu/libxcb-shape.so.0 (0x0000007f7eb25000)
	libxcb-sync.so.1 => /lib/aarch64-linux-gnu/libxcb-sync.so.1 (0x0000007f7eb0e000)
	libxcb-xfixes.so.0 => /lib/aarch64-linux-gnu/libxcb-xfixes.so.0 (0x0000007f7eaf6000)
	libxcb-xinerama.so.0 => /lib/aarch64-linux-gnu/libxcb-xinerama.so.0 (0x0000007f7eae3000)
	libxcb-xkb.so.1 => /lib/aarch64-linux-gnu/libxcb-xkb.so.1 (0x0000007f7eab8000)
	libxcb.so.1 => /lib/aarch64-linux-gnu/libxcb.so.1 (0x0000007f7ea80000)
	libXext.so.6 => /lib/aarch64-linux-gnu/libXext.so.6 (0x0000007f7ea5c000)
	libX11.so.6 => /lib/aarch64-linux-gnu/libX11.so.6 (0x0000007f7e90f000)
	libxkbcommon-x11.so.0 => /lib/aarch64-linux-gnu/libxkbcommon-x11.so.0 (0x0000007f7e8f7000)
	libxkbcommon.so.0 => /lib/aarch64-linux-gnu/libxkbcommon.so.0 (0x0000007f7e8a6000)
	libdl.so.2 => /lib/aarch64-linux-gnu/libdl.so.2 (0x0000007f7e892000)
	libstdc++.so.6 => /lib/aarch64-linux-gnu/libstdc++.so.6 (0x0000007f7e6ba000)
	libm.so.6 => /lib/aarch64-linux-gnu/libm.so.6 (0x0000007f7e60f000)
	libgcc_s.so.1 => /lib/aarch64-linux-gnu/libgcc_s.so.1 (0x0000007f7e5eb000)
	libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000007f7e475000)
	libexpat.so.1 => /lib/aarch64-linux-gnu/libexpat.so.1 (0x0000007f7e43d000)
	libuuid.so.1 => /tmp/.mount_Cura-mVn5hpm/usr/bin/qt/plugins/xcbglintegrations/libuuid.so.1 (0x0000007f7e422000)
	/lib/ld-linux-aarch64.so.1 (0x0000007f7fba0000)
	libpng16.so.16 => /lib/aarch64-linux-gnu/libpng16.so.16 (0x0000007f7e3db000)
	libbrotlidec.so.1 => /lib/aarch64-linux-gnu/libbrotlidec.so.1 (0x0000007f7e3c0000)
	libGLdispatch.so.0 => /lib/aarch64-linux-gnu/libGLdispatch.so.0 (0x0000007f7e235000)
	libGL.so.1 => /lib/aarch64-linux-gnu/libGL.so.1 (0x0000007f7e13e000)
	libharfbuzz.so.0 => /lib/aarch64-linux-gnu/libharfbuzz.so.0 (0x0000007f7e04c000)
	libmd4c.so.0 => /lib/aarch64-linux-gnu/libmd4c.so.0 (0x0000007f7e02d000)
	libdbus-1.so.3 => /lib/aarch64-linux-gnu/libdbus-1.so.3 (0x0000007f7dfc8000)
	libdouble-conversion.so.3 => /lib/aarch64-linux-gnu/libdouble-conversion.so.3 (0x0000007f7dfa4000)
	libicui18n.so.67 => /lib/aarch64-linux-gnu/libicui18n.so.67 (0x0000007f7dc9a000)
	libicuuc.so.67 => /lib/aarch64-linux-gnu/libicuuc.so.67 (0x0000007f7daa0000)
	libpcre2-16.so.0 => /lib/aarch64-linux-gnu/libpcre2-16.so.0 (0x0000007f7da19000)
	libzstd.so.1 => /lib/aarch64-linux-gnu/libzstd.so.1 (0x0000007f7d94d000)
	libglib-2.0.so.0 => /lib/aarch64-linux-gnu/libglib-2.0.so.0 (0x0000007f7d80a000)
	libxcb-util.so.1 => /lib/aarch64-linux-gnu/libxcb-util.so.1 (0x0000007f7d7f4000)
	libXau.so.6 => /lib/aarch64-linux-gnu/libXau.so.6 (0x0000007f7d7de000)
	libXdmcp.so.6 => /lib/aarch64-linux-gnu/libXdmcp.so.6 (0x0000007f7d7c8000)
	libbrotlicommon.so.1 => /lib/aarch64-linux-gnu/libbrotlicommon.so.1 (0x0000007f7d797000)
	libGLX.so.0 => /lib/aarch64-linux-gnu/libGLX.so.0 (0x0000007f7d754000)
	libgraphite2.so.3 => /lib/aarch64-linux-gnu/libgraphite2.so.3 (0x0000007f7d722000)
	libsystemd.so.0 => /lib/aarch64-linux-gnu/libsystemd.so.0 (0x0000007f7d65a000)
	libicudata.so.67 => /lib/aarch64-linux-gnu/libicudata.so.67 (0x0000007f7bb33000)
	libpcre.so.3 => /lib/aarch64-linux-gnu/libpcre.so.3 (0x0000007f7bac1000)
	libbsd.so.0 => /lib/aarch64-linux-gnu/libbsd.so.0 (0x0000007f7ba9c000)
	librt.so.1 => /lib/aarch64-linux-gnu/librt.so.1 (0x0000007f7ba84000)
	liblzma.so.5 => /lib/aarch64-linux-gnu/liblzma.so.5 (0x0000007f7ba4c000)
	liblz4.so.1 => /lib/aarch64-linux-gnu/liblz4.so.1 (0x0000007f7ba1d000)
	libgcrypt.so.20 => /lib/aarch64-linux-gnu/libgcrypt.so.20 (0x0000007f7b94d000)
	libmd.so.0 => /lib/aarch64-linux-gnu/libmd.so.0 (0x0000007f7b931000)
	libgpg-error.so.0 => /lib/aarch64-linux-gnu/libgpg-error.so.0 (0x0000007f7b8fe000)
`

smartavionics, 2022-02-05:
>  libxcb-util.so.0 => not found

Hmm, the bullseye release uses libxcb-util.so.1 . A quick workaround could be to add a symbolic link, i.e.

cd /lib/aarch64-linux-gnu
sudo ln -s  libxcb-util.so.1  libxcb-util.so.0

poevoid, 2022-02-05:
That did it! I'm still getting this message in the terminal when I start it, but the program runs fine:

/tmp/.mount_Cura-mvg9mJE/usr/bin//cura.sh: line 36: message: command not found

<frozen zipimport>:259: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses

smartavionics, 2022-02-06:
Ignore that message not found message, it's a minor slip by myself which I have already fixed for the next release.

For future releases, I'm going to include libxcb-util.so.0 in the AppImage so this should not be a problem for anyone else.